cyoung/stratux #316: 8r1 - No Workie with WingX

ScraboTower, 2016-03-09:
1. Stratux version: 8R1
2. Stratux config:

    SDR
      [ ] single
      [X ] dual

    GPS
      [X ] yes
      [ ] no
      type:

    AHRS
      [X ] yes
      [ ] no

    power source: 

    usb cable:anker

3. EFB app and version: (e.g., WingX Pro7 8.6.2)  wingx

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:
Just updated using 8r1 image and there is no heartbeat to WingX but WingX shows it is connected to Stratux unit.  WebUi shows receivers etc are working

[image]

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

ScraboTower, 2016-03-09:
[image]

cyoung, 2016-03-09:
WingX version? I'll give it a try.

@HiltonG

ScraboTower, 2016-03-09:
8.6.3.1   - there is nothing even on the Stratux tab under "External Hardware" either.

AvSquirrel, 2016-03-09:
How did you upgrade to v0.8r1 -- .sh script, disk image, or pull/build?

I can't replicate on with 8.6.3.1 on iOS 9.2.1 and commit 796d0a263 made by `git checkout` and `make`.

ScraboTower, 2016-03-09:
Image

peepsnet, 2016-03-09:
Is this WingX on Apple?  I just bought WingX on android for $0.99 and activated the free trial. I cannot find any options for external hardware. So i returned the purchase.

ScraboTower, 2016-03-09:
You mean this !!!

[image]

ScraboTower, 2016-03-09:
You mean you didn't see this  ??

[image]

emuyshondt, 2016-03-09:
I get similar results. Tried putting the 0.8r1 image on an SD card and didn't even get the stratux network to show up to connect my iPad Air 2, iOS 9.2.1. Then downloaded 0.7b1 img, and that worked. Updated to 0.8r1 using the update-stratux-v0.8r1-796d0a2632.sh file. 

I could then see the stratux network but even after an iPad reboot, I got an error on my iPad that says: Unable to join the network "stratux" 

My iPhone was able to connect, though. I see the stratux heartbeat in the latest WingX Pro, and everything seems to be working.

After playing with the phone for a couple of minutes, tried the iPad again and this time it connected and seems to work as expected.

Then I tried Foreflight. I saw some traffic. When I tapped it, FF gave a lot of information:
UPS2782 20,000' MSL 15.4nm S
Hdg: 210º Speed 316kts (978ADS-B)

I was surprised that what looks like a jet in the flight levels was using 978 ADS-B. I thought most of those would be using 1090ES transponders.

Henry

PS: My Pi works differently than under 0.6r1. Back then, there's a red LED that always stayed on on the Pi. My Pi had been left powered on for several weeks on my office desk, and was still working with the red LED on before this latest update. After installing 0.8 r1 and 0.7b1 img files, the red LED comes on at first, then a green LED starts flashing next to it, like a disk access light. The red LED goes out, but the green LED flashes for a bit. After a few seconds, all LEDs go out. It was kind of comforting before to see that something was going on. At first I thought my Pi had died. It took a little while for the Stratux network to show up after the light went out.

skypuppy, 2016-03-10:
Isn't that a symptom of low power?

peepsnet, 2016-03-10:
The red light not staying on is indeed a symptom of low power!

@ScraboTower  YUP!!!  Sure did. and the android version is behind. there seems to be no support for external devices yet

peepsnet, 2016-03-10:
@emuyshondt Just redownloaded the img.zip
https://github.com/cyoung/stratux/releases/download/v0.8r1/stratux-v0.8r1-796d0a2632.img.zip

and it installed and booted with the stratux network working. all systems go for me

Try redownload and reimage

AvSquirrel, 2016-03-10:
Interesting. The v0.8r1 image worked fine in WingX for about a minute, then quit. I rebooted and it came up and stayed up.

[image]

@cyoung -- are there any scripts or other configuration items in the image that get set on first boot? It's been a very long time since I've done anything other than git and build, and I've lost track.

@emuyshondt 
> I was surprised that what looks like a jet in the flight levels was using 978 ADS-B. I thought most of those would be using 1090ES transponders.

Known issue. Stratux uses the same byte format as the Garmin GDL90, which was a 978-only device. ForeFlight assumes any traffic sent with this format is 978/UAT, even if Stratux saw it on the 1090 receiver. 

There are two workarounds: Go to the Stratux web UI (http://192.168.10.1 and 1) look on the traffic page, or 2) turn on "Traffic Markings" on the settings UI page.

Ergonomicmike, 2016-03-10:
Funny how the System Information in FF shows a second radio at 1090 but the app won't acknowledge that Traffic can come from 1090.

ssokol, 2016-03-10:
I have a v0.8b1 system that was working fine all day with only my computer
as a client. Late in the day I connected Droid EFB and while I was able to
get GPS, I did not see any of the traffic targets that were listed in the
Traffic tab. I connected my iPhone and ForeFlight and it almost immediately
died - the connection to the access point remained, but no data was
flowing. I tried to ssh in and it hung. I tried ping and it gave no
response until, suddenly, it spit out a flood of responses all with nearly
30 seconds of lag. Then back to no response.

I rebooted it and as soon as it came back up the tablet and iPhone
reconnected. Within a few seconds it was in the same state. I finally shut
down the tablet and the iPhone and after 2 - 3 minutes the ping backlog
cleared. At that point the web interface began responding again and I was
able to ssh in. Everything seemed normal. It appears that the connection /
connections from EFBs triggers some kind of problem.

ScraboTower, 2016-03-10:
Reimagined with 8r1 this morning.... Definately does not resume the heartbeat if WingX was suspended and resumed.

Restarted the Stratux but not WingX and WingX shows the wifi connection but no heartbeat.

ScraboTower, 2016-03-10:
Went back to my previous installed version (v0.7b3-d714a5139e)  and its rock solid

cyoung, 2016-03-14:
Thought at first it was the batched networking format - filling up the queue in sleep mode and bringing the app back doesn't cause it. Can't reproduce this.

@AvSquirrel - the first boot scripts are mostly things like setting an ssh key and such. Nothing stratux specific.

@ScraboTower - you didn't fill out the hardware profile in your original issue report. It looks like you're testing with an iPhone with at least iOS 7? I went back and edited your original issue report to include "WingX 8.6.3.1" so that this is easier to track.

@emuyshondt - are you using an iPhone as well?

ScraboTower, 2016-03-14:
@cyoung   Yes I was using iPhone 5s with iOS 9.2.1.   Never used 8r1 against the iPad but can if that helps.

cyoung, 2016-03-14:
I'm not sure if there are any relevant differences in iOS and/or WingX, it's worth a try if you're able to give v0.8r1 a shot on your iPad. I assume that everyone else is testing it with an iPad.

jzeevi, 2016-03-14:
Do you mean between Android running Wingx? Or an iOS device running Wingx?

 

I would suspect that any differences seen between working and not working versions of Stratux (whatever flavor) eliminate the OS (Android or iOS).

ScraboTower, 2016-03-14:
So updated to 8r1 using the .SH file this time, last time I used the image refresh (twice!!) 
90 mins later both iPad and iPhone are not showing the issue I reported. Even after switching wifi off and then back on again after a few mins, heartbeat resumed, same as resuming from having the app in the background.

cyoung, 2016-03-21:
@ScraboTower - I think this bug may have not been WingX specific and https://github.com/cyoung/stratux/commit/b2eb382c1a27c7bae9774d712972701920a5ea64 may have fixed it.

Could you give v0.8r2 a try?

emuyshondt, 2016-03-21:
I tried v0.8r2 today and I still saw an issue. I've updated my cellular iPar Air 2 to iOS 9.3, but I don't think the OS version has anything to do with this.

I still get a problem using the img file vs. updating 0.7b1 with the .sh file. 

My symptoms are this:

I loaded a new image 0.8r2 image onto my SD card. When I inserted it into my RPi, it came up, and I could see the Stratux network. However, when I tried to join with my iPad, I got an error that said: Unable to join the network "stratux". I tried rebooting the iPad, disabling WiFi on the iPad and then enabling it after a few minutes, but I could not connect. I also tried power cycling the RPi that is connected to a wall wart.

I then reimaged the SD card with version 0.7b1. With that image, I was able to connect to the stratux network. However, I wasn't able to update with the .sh file stored in my Box account. I went all the way through selecting a file in a folder, but immediately got an error that said: File not selected, and the UI froze. I couldn't tap on the OK button. I tried killing the browser session. When I restart it , I get the same problem. BTW, v0.8r2 now selects the file as expected. I didn't try actually installing it over itself, though.

I then went to my Mac and loaded the .sh update from there. Everything worked as expected. After reboot, I was able to join stratux with my iPad, and everything seems to be running well. I don't have enough technical skills to diagnose the connection issue.

One small nit: When I select the stratux network on my iPad, the WiFi connection symbol keeps cycling as when the iPad is trying to connect to a network. I know I'm connected because I get the UI, and I can see data from stratux, but the icon on the Mac (latest iOS) never stops moving. I guess it is a similar issue as the connected icon not showing on the iPad.

Ergonomicmike, 2016-03-22:
In the middle of this thread https://www.reddit.com/r/stratux/comments/49josg/any_error_updating_to_v08r1/
someone wrote about the hoops he had to jump thru to get iOS to update via .sh and drop box. He says it's due to iOS paradigm.

cyoung, 2016-03-23:
> One small nit: When I select the stratux network on my iPad, the WiFi connection symbol keeps cycling as when the iPad is trying to connect to a network. I know I'm connected because I get the UI, [...]

Expected behavior.

cyoung, 2016-04-27:
@emuyshondt @ScraboTower - have the issues persisted?

ScraboTower, 2016-04-27:
@cyoung  - it has been fine with me since reinstall of 8R2

emuyshondt, 2016-04-27:
I was able to get the file installed from Dropbox and Box after making a copy of the file available offline, per the Reddit thread.

peepsnet, 2016-04-29:
If you did not make the file avaliable offline then it will try to pull it from the Internet but you are connected to stratux not the internet

emuyshondt, 2016-04-29:
I was connected to the Internet via the cellular connection of my iPad but the file wouldn't download on its own. I had to make it available offline before it would work. I was able to navigate the directory structure to point to the file, so there were communications going on. It was not at the last step, where the actual file should transfer, that the process broke down.

peepsnet, 2016-04-29:
Thanks for the report!

cyoung, 2016-05-16:
Looks to be fixed.

closed 2016-05-16 by cyoung