cyoung/stratux #317: Additional Sensors and status indicators

dbhorvath, 2016-03-10:
My first GitHub so I apologize if this is not the proper method.

I would like to see the ability to add more sensors -- like CO sensor (analog or digital) and possibly Pulseox (analog likely). 

Status LED also -- "Active" (radios "up" maybe), GPS solution, CO sensor connected/alert.

Currently running 0.6r1, 2 SDR, RY GPS/AHRS with ifly GPS Android.

224XS, 2016-03-11:
Are you aware of any available consumer grade CO or SpO2 sensors that provide wifi or bluetooth or even an output connector?  I for one would prefer to keep the tangle of wire in my cockpit to a minimum. A pulseOx on a fingertip with no wire seems a better solution likewise a CO chem-disk on the panel is simple and effective (if you look at it)

dbhorvath, 2016-03-12:
This is the CO sensor I ordered and will be experimenting with -- analog and digital indications.

http://www.amazon.com/gp/product/B00E5XVLGA?psc=1&redirect=true&ref_=oh_aui_detailpage_o07_s00 

The problem with the CO dots is that they expire silently and require you to look at them (as opposed to a flashing LED and buzzer/flag via servo on my Stratux.

I ordered a sample PulseOx to play with also.  

- David

peepsnet, 2016-03-12:
I believe this kind of application is now getting into a very dangerous legal area. 

Its one thing to have a tool that aids to VFR(see and avoid) flight and IFR "help". All these tools are in addition to the FAA rules for flying.

A CO detector now is a different animal.

Could you imagine the NTSB finding a Stratux in the cockpit of an A/C where the pilot suffered from CO poisoning and crashed. 

Its one thing to have the ADS-B EFB tool its another to have this be a "primary" means of flight safety.  I don't know legally where this goes. With the admission:

> The problem with the CO dots is that they expire silently and require you to look at them (as opposed to a flashing LED and buzzer/flag via servo on my Stratux.

You are saying this will be the primary means of CO protection. 

Could you imagine the scope of this project if VFR was defined by ONLY what was picked up by this type of traffic detection device. We would never accept the legal implications. 

Just my $0.02

JohnOCFII, 2016-03-12:
There a number of portable CO detectors that have been available for use in GA for years.  While I don't personally feel the need to have this integrated with Stratux, I don't think it is of any different level of concern than the other portable products on the market.

http://www.aircraftspruce.com/menus/ps/carbonmonoxidedetectors.html

skypuppy, 2016-03-12:
IMHO, the number of sensors we have in our plane/car/whathaveyou is no 
one's business but our won.  Except, of course, for the required 
minimums.  I'm all for the extra safety if the person wants it and can 
deal with it.
I don't think there are legal limits as to having more sensors than are 
required for safe flight ops.

Skypuppy

cyoung, 2016-03-12:
@dbhorvath - Thanks for trying them out, and I hope you're not put off by the "legal scare". I'd appreciate any info you might have from testing this, if you want you can email it to me. It seems like a great sensor to integrate but I was discouraged by the "48 hours" pre-heat time: https://www.sparkfun.com/datasheets/Sensors/Biometric/MQ-7.pdf. Seems a little excessive, but who knows. If you develop some way to test the sensor that would be great.

peepsnet, 2016-03-13:
@cyoung Maybe it would be worth it to look into having the user be required to accept some kind of terms and conditions via the webui before activation any features like these. If they were added. 

At lease a warning to accept before activating these features.

skypuppy, 2016-03-13:
Please, not yet another lawyer screen.

dbhorvath, 2016-03-13:
Rather than including specific sensors in the build, and having to dealing with the possible liability (which I thought of), I'm suggesting a standard approach to handling sensors. Then it would be up to a third party or user to add the sensors -- and decide what would trigger an alert.

A framework also helps to reduce overlapping and conflicting add-on software that people might come up with.

I'm going to be playing with that sensor @cyoung Those sensors can be tested with regular smoke or spraying CO from a can ( https://jet.com/product/HOME-SAFEGUARD-HOCO2-Carbon-Monoxide-Detector-Tester-Aerosol/bff6929df6cd4042a5cdd3749c9ad33b ).

cyoung, 2016-03-13:
Good idea. When you try the MQ-7, would you be able to report results on the warmup time?

224XS, 2016-03-13:
Dave,
I think it's fine to experiment - Thats what we all do here. But, I am going to put on my other hat (that of
a medical school professor and AME) and caution you about using something like the rattle can full of CO to do anything more than do a very coarse test of sensitivity for your tinkering.  They don't state what the concentration is in the can, but a reasonable detection threshold in a cockpit should be perhaps 100 parts per million, a concentration in my view which would give a pilot time to react before ill effects start.  You are simply not going to get that without an expensive environmental chamber for testing. Be safe!

dbhorvath, 2016-03-13:
@cyoung I plan on setting up a series of tests. I work in IT/Data Analytics -- I'll figure something out :-)

@224XS That makes total sense. I want something better than a chemical dot that fails silently; another battery driven device also isn't good (dead battery syndrome). Even without onboard power, I've gotten 7 hours of battery life with my Stratux -- fan, 2 SDR, GPS/AHRS chip. With the MQ7 it appears you can set the trigger level.

cyoung, 2016-03-13:
You could probably set up an airtight vessel and compare against a household carbon monoxide detector that shows PPM ($45). Not to say that either is guaranteed to be accurate, but it would be an interesting comparison.

Also, a plot of sensor response of the MQ-7 against time (from cold startup) with fixed concentration of CO (leaving it sealed with whatever amount of CO you expose it to) would be helpful to figure out the warmup needs.

If you want to get quantitative, maybe use a CO generator kit? http://www.sigmaaldrich.com/technical-documents/articles/technology-spotlights/cogen-coware/faq.html. ~$160-200.

224XS, 2016-03-13:
The sensitivity range from the spec sheet at SparkFun, the MQ-7 has a detection range down to 20 ppm.
That's at STP with 21% O2.  At altitude things change - will be interesting to test out.
Another interesting consideration is that the MQ-7 has a minimum power consumption of 350 mW for heating the element. Say an extra 100 mA power drain for the battery.

skypuppy, 2016-03-13:
You can kind of cheat on the enviro chamber by using some hot glue and 3 
or 4 tarps.  It won't last but it doesn't need to.  If you wanna be real 
fancy, sew in some clear visqueen to see stuff and a couple of latex 
gloves on arm-reaches.  The arms will definitely be the hardest work.  :)

skypuppy, 2016-03-13:
You can rent some analytical gear to do your initial calibrations for 
the home-made stuff, and from there, you're good to go!
personally, i'd rather have the RPi handling the issues and watch/record 
the ppm over time, with an [analog]-looking meter that had a yellow zone 
and then a red zone, with flashing led's of different color for each zone.

Danger, danger, Will Robinson.

skypuppy, 2016-03-13:
Which brings to mind having an O2 sensor in the same kit and kill two 
birds with one stone.  So to speak.

dbhorvath, 2016-03-14:
A bit more detailed manual: https://cdn.sparkfun.com/datasheets/Sensors/Biometric/MQ-7%20Ver1.3%20-%20Manual.pdf

@skypuppy As I understand it, the % O2 in the atmosphere is fairly steady (at altitudes we're talking about). The *effective* O2 as the body can absorb it goes down though...  http://www.higherpeak.com/altitudechart.html

Unfortunately, the manual doesn't mention altitude though...

Hence my idea that the user would set the indicator levels -- they can calibrate as they like. Personally, I'd like it to trigger at fairly low concentrations no matter what; so if it gets less sensitive with altitude or time since last use or humidity or ... I'd still be OK. 70-100ppm would likely be my warning setting. I'll have to try running in real life (in the car, in the home/office, at the bbq, car exhaust, etc. 150-200 is danger https://www.cpsc.gov/en/Safety-Education/Safety-Education-Centers/Carbon-Monoxide-Information-Center/Carbon-Monoxide-Questions-and-Answers-/

I doubt it would be necessary to run the sensor constantly in the plane. 20% duty cycle?

cyoung, 2016-03-14:
@dbhorvath - https://www.grainger.com/product/HOME-SAFEGUARD-Carbon-Monoxide-Detector-Tester-6TWH5
http://complyplus.grainger.com/grainger/msds.asp?sheetid=3827769

Looks like that product is a mixter of CO and Argon and the CO concentration is 2800ppm.

RobertGary1, 2016-03-14:
I use this. It makes a loud alarm noise that you can easily hear in the cockpit. It also shows you a real time digit display of CO concentration. 

Kidde KN-COPP-B-LPM Battery-Operated Carbon Monoxide Alarm with Digital Display https://www.amazon.com/dp/B004Y6V5CI/ref=cm_sw_r_cp_awd_OxT5wb4CJ5YT4

I'm not sure I'd spent much time looking at my Stratux. In fact I put mine behind the panel. 

-Robert

224XS, 2016-03-14:
Hi Robert,
I think the idea is to look at your iPad or android display, not the rasp-pi

224XS, 2016-03-14:
@skypuppy:  re O2 sensor

Again donning my MD hat, I am going to flat out state that having SpO2 (hemoglobin Oxygen saturation from a fingertip pulse ox probe) is far, far more valuable that simply having the partial pressure of cabin oxygen. The SpO2 is a reflection of the bioavailability of oxygen to your mitochondria and takes into account your lung and breathing quality. It is also basically self-calibrating.  The problem, as I mentioned above, is that, so far as I know there is no consumer level SpO2 module or finger probe that has any other than a visual interface to tie into a rasp-pi system. And yes, the presence of CO will completely screw a two wavelength pulse-ox - thats why an independent CO detector is required.

skypuppy, 2016-03-14:
I was advocating two different sensors, one for O2 and one for CO.
And yes, the finger probe :) is much more accurate, having an O2 sensor 
talking to the Pi costs very little more and gives one just that more 
help in the air, equivalent to the utility of a CO sensor.

cyoung, 2016-03-14:
That plus why bother - unless your plane is capable of transporting you to another planet, what does an O2 sensor tell you that barometric pressure doesn't? I can also imagine that the biggest consumer of O2 aboard (the engine) would let you know also.

skypuppy, 2016-03-14:
In that vein, why bother having a stratux?
Because it helps safety and it is great FUN!

Further, even if you know baro, you'd still have to look up the 
*expected* O2 concentrations and sometimes they don't agree.  Who wants 
to bother looking something up when you can have the actual data right 
there for peanuts?

Skypuppy

skypuppy, 2016-03-14:
Also, it needn't be connected to the stratux RPi2/3; it could even be on 
a separate Beaglebone Black.  :)  We can have more than one.

224XS, 2016-03-14:
I’m with Chris here.  Unless you are high altitude physiologist, the atmospheric pO2 is pretty useless because
unless you are also monitoring the conditions of lungs and respiration, i.e., ETCO2,, the pO2 doesn’t tell you how well
you are absorbing it.

There is a far higher price to be paid than the cost of the sensor and the code, it is the price of your
brain’s processing power that you waste when confronted with meaningless but attractive and distracting data.
That is especially true when the readings start to indicate you are nearing the limits of usual compensation.
SpO2 is exactly meaningful and easy to interpret: below 80% you are impaired whether you feel it or not, below
50% you are highly impaired, below 30% you will be dead shortly.  It is very difficult to assign such limits to pO2
because every body is different in their degree of athletic compensatory ability and the pathology in their lungs.
Further, the SpO2 tells you if your supplemental nasal/mask O2 is working, information the pO2 in cabin will not.

But then again, some people just want toys and blinking lights. 

As for bars, the fO2 stays constant well above your engine’s ability to climb.

cyoung, 2016-03-21:
@dbhorvath - any testing done yet?

dbhorvath, 2016-04-18:
I haven't forgotten, just have had a few issues tying up my time.

skypuppy, 2016-04-18:
Is there a finger O2 concentration sensor that we can tie into an SOC?  :)

dbhorvath, 2016-04-19:
I tried to grab one they used on me recently but I lost track of it. It was disposable/tape on.

224XS, 2016-04-19:
Believe it or not, you could actually build an SpO2  (that’s Hb saturation, Not concentration) monitor out of a RaspPi, but an arduino would be much better because you need two digital output lines to drive the two LEDs, and a moderate resolution ADC for the photocell. The algorithm was invented by an acquaintance of mine at Stanford about 1980 and has since been published widely. It’s become a pretty routine first year bio-engineering student project. When you build it yourself, then you could put in a serial out to your home-brew EFIS.

cyoung, 2016-12-01:
@dbhorvath - if you have further progress, please re-open. Still interested to see any findings from testing a CO sensor.

closed 2016-12-01 by cyoung

cyoung, 2017-01-18:
https://www.seeedstudio.com/MAXREFDES117%23%3A-HEART-RATE-AND-PULSE-OXIMETRY-MONITOR-p-2762.html