cyoung/stratux #318: v0.8r1 Service Just stops. No BLOD

peepsnet, 2016-03-10:
1. Stratux version: 0.8r1
2. Stratux config:

    SDR
      [ ] single
      [ X ] dual

    GPS
      [ X ] yes
      [ ] no
      type:

    AHRS
      [ X ] yes
      [ ] no

    power source: 2.1A Wall Plug

    usb cable: portapow 20awg

3. EFB 
N/A

4. Description of your issue:

Service just stops. 
[stratux (1).log.txt](https://github.com/cyoung/stratux/files/166476/stratux.1.log.txt)

This happened a few times. After the last I started a fresh log file. This is the log attached. just over 30min of time then the service stopped. 

SSH still works and my webmin interface still works.

This is a new issue for me since v0.8r1

What other system logs can I look at for any errors??

Axtel4, 2016-03-10:
The next time it happens, try `service stratux status` to see what it says.  If it says, "[FAIL] stratux is not running ... failed!".  Try `service stratux start` to see if it restarts.

peepsnet, 2016-03-10:
It says not running

but restarts just fine

peepsnet, 2016-03-10:
Dump1090 is still running

Axtel4, 2016-03-10:
> It says not running

I saw that yesterday on v0.7b3 right before I updated to v0.8r1, but I haven't see it since updating.  I don't know why Stratux just stopped, but I was able to restart it without rebooting Stratux.

AvSquirrel, 2016-03-10:
For posterity, fd936e7e1b is the current commit on my experimental branch.

@peepsnet - can you just start `./gen_gdl90` from the command line and let it run until it crashes again? I'd like to capture a stack trace to see exactly what's causing it to crash.

Not that it's necessarily related, I was getting some intermittent out-of-bounds array indexes a few commits back due to unsafe array handling in my pseudoattitude code. I thought I got that fixed (haven't seen a crash in about a week) but I'd like to make sure you're not seeing that.

Ergonomicmike, 2016-03-10:
@AvSquirrel I looked at the commit. The lines in the config in wheezy-stage-4

```
        echo "arm_freq=900" >>boot/config.txt
  	echo "sdram_freq=450" >>boot/config.txt
  	echo "core_freq=450" >>boot/config.txt
```

seem unnecessary, since we're running Powersave governor now. Yes?

AvSquirrel, 2016-03-10:
@Ergonomicmike - Dunno. Chris put those in about six weeks ago (https://github.com/cyoung/stratux/commit/5aca8adea2b4f8c6fe7baf312088bafa152a73be). Must have come along for the ride at some point when I synced against master.

Ergonomicmike, 2016-03-10:
@peepsnet I've had my Stratux running at home for 24 hrs now on v0.8r1 w/o problems. Trying to isolates the differences, what brand of GPS? And did you write to the SDR's? (@cyoung This might be useful info in the Issue Template, since there's a lot of extra (new) code that handles dedicated SDR's.) I have stock SDR's.

peepsnet, 2016-03-10:
SDRs: 2 X Nano2+
(Serials: stx:978:5 / stx:0:0)
RY835AI
2 X Edimax

I added an app called wicd-curses, a few hours ago, to experiment with controlling my wlan1 . I uninstalled this and restarted the stratux service. I am now waiting for a failure.

Ergonomicmike, 2016-03-10:
Oh, yeah, I forgot that you've two Edimax's going. So w/ those and your added app, your system is definitely not normative.

peepsnet, 2016-03-10:
I have webmin
http://www.webmin.com/deb.html

htop
http://www.webhostingbuzz.com/wiki/install-htop-linux/

ntp
https://wiki.debian.org/NTP

and lighttpd w/php

but I have had them installed since v0.6

Ergonomicmike, 2016-03-10:
So Chris should probably add another item to the template: non-standard software. (Although were it me, I would say it voids your warranty.)

peepsnet, 2016-03-10:
So I am going to chalk this up to the wifi manager tool thing!!  I am not seeing a failure. I will leave it to run overnight

Ergonomicmike, 2016-03-10:
I've thought of that too. I'm always typing the same thing into the template, I'm going to save it as a text file next time.

peepsnet, 2016-03-10:
We should set the example... lol

skypuppy, 2016-03-10:
Valgrind is your friend.

peepsnet, 2016-03-10:
9hours up time! We can close it

peepsnet, 2016-03-10:
@AvSquirrel 
Ok so the error is with gen_gdl90

`root@raspberrypi:~# /usr/bin/gen_gdl90
2016/03/10 15:37:16 Stratux v0.8r1 (fd936e7e1b71429abb4e853f8a7116078a3da718) starting.
2016/03/10 15:37:16 Entered uatReader() ...
2016/03/10 15:37:16 read in settings.
2016/03/10 15:37:16 lease expired for 192.168.10.18 (2016/02/16 03:22:36) - skipping.
2016/03/10 15:37:16 lease expired for 192.168.10.19 (2016/02/27 00:28:02) - skipping.
2016/03/10 15:37:16 lease expired for 192.168.10.10 (2016/03/08 21:52:32) - skipping.
2016/03/10 15:37:16 client connected: 192.168.10.12:4000 (android-a694dd93f3d60a75).
2016/03/10 15:37:16 client connected: 192.168.10.12:4001 (android-a694dd93f3d60a75).
2016/03/10 15:37:16 client connected: 192.168.10.12:4002 (android-a694dd93f3d60a75).
2016/03/10 15:37:16 client connected: 192.168.10.12:43211 (android-a694dd93f3d60a75).
2016/03/10 15:37:16 client connected: 192.168.10.13:4000 (android-495004c3ce8ba7d3).
2016/03/10 15:37:16 client connected: 192.168.10.13:4001 (android-495004c3ce8ba7d3).
2016/03/10 15:37:16 client connected: 192.168.10.13:4002 (android-495004c3ce8ba7d3).
2016/03/10 15:37:16 client connected: 192.168.10.13:43211 (android-495004c3ce8ba7d3).
2016/03/10 15:37:17 ===== UAT Device Name  : Generic RTL2832U OEM =====
2016/03/10 15:37:17 ===== UAT Device Serial: stx:978:8=====

Kernel driver is active, or device is claimed by second instance of librtlsdr.
In the first case, please either detach or blacklist the kernel module
(dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

usb_claim_interface error -6
2016/03/10 15:37:17     UAT Open Failed...
2016/03/10 15:37:17 UATDev.sdrConfig() failed: resource busy
2016/03/10 15:37:17 ===== UAT Device Name  : Generic RTL2832U OEM =====
2016/03/10 15:37:17 ===== UAT Device Serial: stx:0:0=====

Kernel driver is active, or device is claimed by second instance of librtlsdr.
In the first case, please either detach or blacklist the kernel module
(dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

usb_claim_interface error -6
2016/03/10 15:37:17     UAT Open Failed...
2016/03/10 15:37:17 UATDev.sdrConfig() failed: resource busy
2016/03/10 15:37:20 Configuring GPS
2016/03/10 15:37:20 Using /dev/ttyAMA0 for GPS
2016/03/10 15:37:20 Finished writing u-blox GPS config to /dev/ttyAMA0. Opening port to test connection.

Initializing IMU .2016/03/10 15:37:21 GPS dectected: NMEA messages seen.
2016/03/10 15:37:21 GPS detected: u-blox NMEA position message seen.
.2016/03/10 15:37:21 Average sendable queue is 0 messages. Changing queue timer to 0.100000 seconds
........ done

Default accelcal.txt not found
Default magcal.txt not found
2016/03/10 15:37:21 NMEA messages last minute. Valid: 9. Position: 9. Invalid: 0.
data_ready() - data not ready.
2016/03/10 15:37:24 NMEA messages last minute. Valid: 42. Position: 36. Invalid: 0.
2016/03/10 15:37:26 Average sendable queue is 0 messages. Changing queue timer to 0.100000 seconds
dmp_read_fifo() failed
2016/03/10 15:37:28 NMEA messages last minute. Valid: 94. Position: 80. Invalid: 0.
2016/03/10 15:37:28 gpsSerialReader() scanner loop iteration i=100
dmp_read_fifo() failed
2016/03/10 15:37:31 Average sendable queue is 0 messages. Changing queue timer to 0.100000 seconds
2016/03/10 15:37:32 NMEA messages last minute. Valid: 146. Position: 124. Invalid: 0.
dmp_read_fifo() failed
2016/03/10 15:37:36 Average sendable queue is 0 messages. Changing queue timer to 0.100000 seconds
2016/03/10 15:37:36 NMEA messages last minute. Valid: 197. Position: 167. Invalid: 0.
2016/03/10 15:37:36 gpsSerialReader() scanner loop iteration i=200
dmp_read_fifo() failed
2016/03/10 15:37:40 NMEA messages last minute. Valid: 249. Position: 211. Invalid: 0.
2016/03/10 15:37:41 Average sendable queue is 0 messages. Changing queue timer to 0.100000 seconds
2016/03/10 15:37:44 gpsSerialReader() scanner loop iteration i=300
dmp_read_fifo() failed
2016/03/10 15:37:44 NMEA messages last minute. Valid: 302. Position: 256. Invalid: 0.
2016/03/10 15:37:46 Average sendable queue is 0 messages. Changing queue timer to 0.100000 seconds
[... 349 lines cut ...]
        /root/go/src/net/http/server.go:1862 +0x190
net/http.(*conn).serve(0x10caa240)
        /root/go/src/net/http/server.go:1361 +0xbfc
created by net/http.(*Server).Serve
        /root/go/src/net/http/server.go:1910 +0x360

goroutine 14 [select]:
_/root/stratux-dev/mpu6050.(*MPU6050).Run.func1(0x107d7f10)
        /root/stratux-dev/mpu6050/mpu6050.go:137 +0x12c
created by _/root/stratux-dev/mpu6050.(*MPU6050).Run
        /root/stratux-dev/mpu6050/mpu6050.go:148 +0x44

goroutine 15 [runnable]:
main.attitudeReaderSender()
        /root/stratux-dev/main/ry835ai.go:1702 +0x6c
created by main.initAHRS
        /root/stratux-dev/main/ry835ai.go:1775 +0x1cc

goroutine 16 [chan receive]:
main.tempAndPressureReader()
        /root/stratux-dev/main/ry835ai.go:1605 +0x78
created by main.initAHRS
        /root/stratux-dev/main/ry835ai.go:1776 +0x1e0
root@raspberrypi:~#
`

peepsnet, 2016-03-10:
Let me know what u need

peepsnet, 2016-03-10:
The real issue is probably with the Avsquirrel Fork
Opened an issue in his Fork
https://github.com/AvSquirrel/stratux/issues/15

@cyoung: Feel free to close here

cyoung, 2016-03-10:
https://github.com/AvSquirrel/stratux/tree/fd936e7e1b71429abb4e853f8a7116078a3da718
https://github.com/AvSquirrel/stratux/blob/fd936e7e1b71429abb4e853f8a7116078a3da718/main/ry835ai.go#L948

closed 2016-03-10 by cyoung

AvSquirrel, 2016-03-10:
Thanks. Very helpful. This appears to be related to my development branch only, so I'll tag as 'externalbug' and continue tracking at avsquirrel/stratux#16