cyoung/stratux #323: Is it normal to have two dump1090 processes active?

jpoirier, 2016-03-11:
1. Stratux version: tip @ 60e240d185b
2. Stratux config:

    SDR
      [ ] single
      [X ] dual

    GPS
      [ ] yes
      [X ] no
      type:

    AHRS
      [ ] yes
      [ X] no

    power source: AC adapter

    usb cable: NA

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2) NA

    EFB hardware: (e.g., iPad Mini 2) NA

4. Description of your issue:

At bootup I have two anonymous dongles inserted but only 978 enabled, go to the webui and enable 1090. Via htop I can see two dump1090 processes spawned and active. If I disable 1090 via the web ui only one dump1090 process is killed.

jpoirier, 2016-03-11:
It seems dump1090 does run two processes on startup, but top only shows one whereas htop shows two. 

If I kill gen_gdl90 and run dump1090 via command line I see two dump1090 processes in htop, the problem is that calling cmd.Process.Kill() in gen_gdl90 only seems to kill one of the processes.

jpoirier, 2016-03-11:
The fix is to probably create a new process group and use syscall.Kill with the Process.Pid. I'll send a PR with the changes.

cyoung, 2016-03-11:
htop shows threads, make sure you verify that this is not just a child of the main process.

peepsnet, 2016-03-11:
I believe there is a "tree view" in htop. Chris and I looked at this a few weeks ago

skypuppy, 2016-03-11:
<F5>

kdknigga, 2016-03-11:
Indeed there is.  You can also color processes and threads differently.

jpoirier, 2016-03-12:
fixed by pull request #326

jpoirier, 2016-03-14:
There's actually only ever one parent process, and it's children, so I think this can be closed.

closed 2017-04-19 by cyoung