cyoung/stratux #324: web ui out of synch with dongle config in gen_gdl90

jpoirier, 2016-03-11:
1. Stratux version:
2. Stratux config:

    SDR
      [ ] single
      [X ] dual

    GPS
      [ ] yes
      [ X] no
      type:

    AHRS
      [ ] yes
      [ X] no

    power source: AC

    usb cable: NA

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2) NA

    EFB hardware: (e.g., iPad Mini 2) NA

4. Description of your issue:
Observations done via shell stdout, ie kill gen_gdl90 then run again via in the shell window and observe stdout.

Start with both 978 and 1090 enabled in the webui then disable 1090 bia the webui, observe via the shell's stdout that both 978 and 1090 are disabled (that's the expected behavior) but 978 does not get re-configured even though 978 is still enabled in the web ui. 

In the process of debugging...pull request #322 may or may not be relevant.

Note, I'm running stratux on Jessie and I have a web browser opened locally (on the desktop) with the web ui page loaded.

peepsnet, 2016-03-11:
> Start with both 978 and 1090 enabled in the webui then disable 1090 bia the webui, observe via the shell's stdout that both 978 and 1090 are disabled (that's the expected behavior)

Why would both 978 and 1090 disable if only 1090 was switched off??

skypuppy, 2016-03-11:
@Joseph, may I get a copy of your SD card image, please?

Skypuppy

jpoirier, 2016-03-11:
On Fri, Mar 11, 2016 at 3:47 PM, skypuppy <notifications@github.com> wrote:

> @Joseph, may I get a copy of your SD card image, please?
>
> Skypuppy
>

Just image an SD card with Jessie then clone
https://github.com/jpoirier/stratux-setup and follow the steps in the
readme

skypuppy, 2016-03-12:
EXCELLENT script there, joseph!

A few unimportant warnings here and there but is running now like a 
champ, with both SDR's active and .8r1.  I'm surprised I haven't seen 
any airplanes yet, though.

Thanks again,
Skypuppy
(you saved me an unknown large amount of time)

jpoirier, 2016-03-12:
fixed by pull request #326

jpoirier, 2016-03-14:
Al least the issue I was seeing seemed to be a side effect of the hanging 1090 read process; it went away when the read process started exiting properly, but no doubt there are other issues  due to the use of globals with no synchronized read/writes.

closed 2018-01-17 by cyoung