cyoung/stratux #325: DHCP lease expires after GPS sets time

cyoung, 2016-03-11:
1. Stratux version: v0.8b1
2. Stratux config:

    SDR
      [ ] single
      [X] dual

    GPS
      [X] yes
      [ ] no
      type:

    AHRS
      [ ] yes
      [X] no

    power source: EasyAcc 10Ah

    usb cable: y-cable

3. EFB app and version: FF 7.5.2

    EFB platform: iOS 9.2

    EFB hardware: iPad Mini 2

4. Description of your issue:

If DHCP lease is obtained before system clock is set and the system clock is 3.3 days or more behind actual time, then the client that connected can remain connected for some time without renewing the DHCP lease but be ignored by gen_gdl90.

Ergonomicmike, 2016-03-12:
Boy, talk about corner cases. How did you find that one?

cyoung, 2016-03-13:
A system image that was 4-5 days old (never booted), a new GPS (takes longer to get first lock), and connecting right after boot.

cyoung, 2016-03-20:
I've found that this really isn't an edge case in testing. Going to remove this feature as it adds unnecessary complexity. The devices count in the web interface will again be overstated - we'll have to let the DHCP server age out leases as it did before.

commit b2eb382c (2016-03-20): Remove DHCP lease expiry code.

Fixes #325.
closed 2016-03-20 by cyoung