cyoung/stratux #327: radios don't appear to be working

skypuppy, 2016-03-12:
1. Stratux version: 0.8r1
2. Stratux config:

    SDR
      [ ] single
      [x ] dual

    GPS
      [ x] yes
      [ ] no
      type: usb SIRF and serial mtk3339

    AHRS
      [ z] yes
      [ ] no

    power source:(mega) powered usb hub

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)  iFlyGPS 9.4 I( think)

    EFB platform: (e.g., iOS 9.2)  Android Tab S 10.2 os 5.something

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

Had jessie-based 0.8r1 up for several hours now and get zero reception from either rtlsdr.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

skypuppy, 2016-03-12:
Update: Further testing with laptop and a Linux PC makes me thing that BOTH radios have died!  They were working just fine 2 weeks ago while running stratux.  Dang.  I need to do more testing but I may need to close this one.

skypuppy, 2016-03-12:
This really has me baffled.  It seems unlikely that *both* radios would go out simultaneously.  Can you guys recommend some software to test them via desktop pc Linux Debian jessie?

skypuppy, 2016-03-13:
No one knows how to test these rtlsdr radios?  Pretty please?
Thanks.

mtfgeek, 2016-03-13:
GNU Radio works for me...

jpoirier, 2016-03-13:
> No one knows how to test these rtlsdr radios? Pretty please?

apt-get install gqrx

skypuppy, 2016-03-14:
Thanks, Joseph.
Both radios are confirmed working, at least in the normal FM band.

So that makes me wonder if something is wrong on the stratux side.
Stratux sees both of them, but no traffic for days now, when there used 
to be lots.

dbhorvath, 2016-03-14:
This may seem like a silly question but have you confirmed actual ADS-B traffic in your area? Have you checked on something like planefinder.net * (filter for ADS-B only)? Would those planes be "visible" to your antenna (i.e., not behind a building)?

* I happen to provide the local feed for planefinder.net (they provided the receiver and a really nice antenna -- SBS-1 several years ago). There are other providers...

skypuppy, 2016-03-14:
UAT traffic is not very common at my location but there have always been 
some.  1090 traffic is quite a lot.  I verified both radios are working, 
at least with FM stations.  However, I have to turn the gain up to 35.

Have there been recent changes in gain or other controls in stratux for 
the rtlsdr radios?  I see no user-configurable options there. And I 
can't keep up with all the changes that occur in the spurs off the 
main-line code.

I have 4 radios of 3 different types coming but it will be 2-4 weeks 
before they arrive.  Still, these two worked just fine for months, and 
still seem to be working with every app except stratux.

Thanks,
Skypuppy

Ergonomicmike, 2016-03-14:
Since you're running non-standard Stratux (w/ jessie, right?), all bets are off. I suggest trying another SD card w/ stock v0.8r1 image (although Jeff had to .sh up from v0.7b1 to get to v0.8r1 properly). Or write your existing image to a file and then overwrite w/ 0.8r1. Better yet, go back to the last image that was working for you. That should confirm that you SDR's are fine. Then you can focus on why your build of jessie is giving you problems.

skypuppy, 2016-03-15:
It all worked fine (except the radios) until today's "apt-get upgrade."

Btw, I tried a couple of prior images and the radios did not work on 
them, either, with older (smaller) SD cards.  But they did work before.  
One was 0.3something, the other was 0.4r4 which used to be the "gold 
standard."  4r4 was on wheezy.

How about a test, please?  Do a "apt-get update, then an upgrade" 
*without* accepting the newest upgrade.  You should see something about 
the boot partition and maybe that gives us a better clue than my 
non-eidetic memory.

I'm very curious why stratux won't even run now.  Also, since secure 
shells are still allowed in, what toggle is needed for the browsers to 
be allowed in, too?  It's been so long since I worked on that stuff...

Thanks,
Skypuppy

jpoirier, 2016-03-15:
@skypuppy 

I just did a Jessie update and upgrade (accepted all parts), rebooted, and all seems to be working fine, i.e. stratux is up and running with no problems noted.

skypuppy, 2016-03-15:
Well, phooey.  You think I should run the stratux-setup.sh script again 
and accept everything as good?

Did you see the 2 or 3 entries about the boot partition?

jpoirier, 2016-03-15:
Yes, the boot stuff was included in the upgrade. 

Remove all the go* and stratux* stuff from your root folder then clone the stratux-setup repo again and run the script. 

Note, the current stratux-setup.sh script install the latest stratux release where previous version install tip. If you want tip comment out the "git checkout v0.8r1" part in the stratux-setup.sh file.

skypuppy, 2016-03-15:
I don't know that phrase, "install tip."  What does that mean?

Thanks.

jpoirier, 2016-03-15:
On Mon, Mar 14, 2016 at 8:44 PM, skypuppy <notifications@github.com> wrote:

> I don't know that phrase, "install tip." What does that mean?
>

The head of the master branch, i.e. the most update code in the repo.

closed 2016-03-15 by skypuppy

Ergonomicmike, 2016-03-17:
@skypuppy FWIW, my SDR's "stopped working" when I tried the jessietest image. I found that if I unplugged my GPS, that the SDR's came alive.

skypuppy, 2016-03-17:
Dang, with all this data, I should have built a table so we could keep 
up with it.