cyoung/stratux #328: Is gpsd practical for stratux?

skypuppy, 2016-03-12:
1. Stratux version:
2. Stratux config:

    SDR
      [ ] single
      [x ] dual

    GPS
      [x ] yes
      [ ] no
      type:

    AHRS
      [x ] yes
      [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:
I am unclear as to how to use gpsd (as an API?) but it bears thinking about.  If the code needed in stratux is not too onerous, it could make life easier for everyone with different GPS chipsets.  All the drudge work is handled by gpsd instead of stratux programmers.  Opinions?
If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

skypuppy, 2016-03-12:
It might be especially attractive for those who like json.  :)

skypuppy, 2016-03-18:
After playing with gpsd for a couple of days, I've made no progress but that may be to my limitations.  The goal was to allow backend software to handle the administrivia for all gps devices that stratux owners may choose and relieve the burden from the stratux programmers: write once, work with many.  My efforts were on the host PC (linux debian jessie amd64) and not the Pi yet, so the Pi will be next to try.

cyoung, 2016-03-20:
Could be.

http://www.catb.org/gpsd/hardware.html

Looks like NEO-5Q has been well tested and works well with it.

BU-353-S4 should work fine, as well. The comments on that one were surprising considering the good results that are being reported:
>This receiver does not support PPS timing output. Much like the BU-353 except for using a SiRFStar IV chip. Has poor sensitivity and takes a lot longer to cold-start than the vendor claim 45 of 45 seconds. Gary Miller rates this device "DO NOT EVER BUY ONE!", observing "The big selling point of the SiRF IV is the lower power needs. Looks like they made that happen at the expense of sensitivity and timeliness of fixes." Also, has the SiRF-IV tendency to freeze when switched from binary to NMEA mode (power-cycling it unjams it).

Ergonomicmike, 2016-03-24:
That is surprising. I can't speak to PPS timing. But as for everything else, that's completely opposite to my experience with the BU-353. I never saw it jam changing from binary to NMEA. Perhaps these guys had a bad run, Chinese knock-offs or earlier firmware?

cyoung, 2016-04-27:
It may be. Thanks for the suggestion. Going to defer this until either 1) someone wants to do some in-depth testing and report on it, or 2) it becomes a higher priority for some other reason.

closed 2016-04-27 by cyoung

skypuppy, 2016-04-28:
I started another thread about it, maybe in reddit.  With some valuable hints from Adafruit, gpsd works just fine in the Pi3.  Then it becomes a choice of which interface method to use.  dbus is the only one of the three methods that did not work in my tests but that may be due dbus just not being turned on in the main stratux releases.  Shared memory is faster anyway, and requires a smaller amount of code testing for the developer.  However, that bit if testing is *critical* to avoid memory leaks.