cyoung/stratux #329: GPS debug messages or GPS problem?

dcecchino, 2016-03-13:
1. Stratux version:Stratux v0.8r1
2. Stratux config:

    SDR
      [ ] single
      [ X] dual

    GPS
      [ X] yes
      [ ] no
      type: ry835ai

    AHRS
      [X ] yes
      [ ] no

    power source: Anker PowerCore External Battery 20100mAh

    usb cable: supplied with Anker 

3. EFB app and version: (e.g., WingX Pro7 8.6.2)  WingX Pro 7 (8.6.3.1)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2) IPAD 3 

4. Description of your issue: In the logs, I see many invalid checksum, checksum failed, and 
reading standard input: bufio.Scanner: token too long messages. Not sure if this is debug output, or letting me know that my soldering is messed up. The GPS and AHRS is working through the UI, and I receive SATs. Please see my stratux.log attachment.
[stratux.txt](https://github.com/cyoung/stratux/files/170783/stratux.txt)

Thanks for your support and development! 

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

AvSquirrel, 2016-03-13:
Lots of bad checksums in that log. Two likely causes:

1. Noisy / bad connection to the GPS. This will cause random bits / characters to appear in the GPS NMEA messages.
2. Multiple instances of gen_gdl90 are running, taking turns unloading the serial FIFO. This will sporadically cause blocks of 8 or 16 bytes to drop from the GPS messages.

If logging is turned on, you could pull one of the GPS logs and post it here.

dcecchino, 2016-03-13:
Thanks for the quick response! I've tar'd up the stratux.log and the stratux log directory. I will take a look at my connections in the morning, please see the gz file. 
[stratux.tar.gz](https://github.com/cyoung/stratux/files/170795/stratux.tar.gz)

AvSquirrel, 2016-03-13:
Very useful. 

The receiver's telling us that it's being sent a whole series of messages it doesn't understand, and that its transmit buffer is occasionally overflowing. It also shows that you're not getting a signal lock.

Both of these are likely due to a short in your wiring -- I dealt with the same problem on my initial setup. First troubleshooting step should be to double check your soldering -- make sure you're not bridging the TX and RX wires to each other or to ground.

````
32015523582,$GNTXT,01,01,01,NMEA unknown msg*46
32048855978,$GNTXT,01,01,01,NMEA unknown msg*46
32090516863,$GNTXT,01,01,01,NMEA unknown msg*46
32123838113,$GNTXT,01,01,00,txbuf alloc*61
32157126655,$GNTXT,01,01,01,NMEA unknown msg*46
32198769155,$GNTXT,01,01,01,NMEA unknown msg*46
32240405509,$GNTXT,01,01,01,NMEA unknown msg*46
````

dcecchino, 2016-03-13:
Dude, you're awesome!

That's exactly what it was, the TX and RX were touching together. Cleaned it up. I also noticed there was a steady red light before, now it is flashing for the GPS and it picked up the SATs right away. Attached logs. 
[stratux_03132016.tar.gz](https://github.com/cyoung/stratux/files/170846/stratux_03132016.tar.gz)

skypuppy, 2016-03-13:
from elinux.org, re the red led flashing:

steady ON when Pi receives some power on original B and A, but since the 
B+ (except for the zero) the PWR LED is also a power good indicator that 
goes off when power drops below 4.65V

http://elinux.org/R-Pi_Troubleshooting

So you have a power supply problem.  Might be the USB cable that feeds 
power to the RPi2.

dcecchino, 2016-03-13:
Thanks, but just refering the the ry835ai gps. No
problems with the pi

closed 2016-03-13 by dcecchino

Ergonomicmike, 2016-03-13:
@AvSquirrel Do multiple instances of gen_gdl90 cause sporadic blocks to drop from GPS messages over USB too? iFly's AutoTaxi feature has been popping up on the ground lately, and when I look at the iFly logs, it says that GPS was lost & then found, even tho the signal indicator is solid. (AutoTaxi pops up on GPS acquisition when on the ground.)

cyoung, 2016-03-13:
@dcecchino - if you have further problems, go with the only recommended hardware for GPS: Vk-172. See the website for details.

Team - another great reason not to recommend or focus on supporting breakout boards.

dcecchino, 2016-03-13:
Thank you, yes in case there were more issues I did purchase the vk-172!

megadip, 2020-06-28:
> Very useful.
>
> The receiver's telling us that it's being sent a whole series of messages it doesn't understand, and that its transmit buffer is occasionally overflowing. It also shows that you're not [...]
I am having this same problem. 
The thing is the terminals are only shorted when connected to my pi.
Is this normal?
Tx, Rx, and GND all have continuity, only when connected though.
Sorry to bring back an old thread, I just really want to understand what I am doing wrong.

rhole, 2020-06-28:
Could you have Tx/Rx cross connected?

Tx connects to Rx at the other end

Rx connects to Tx at the other end

 

 

Rick Hole

megadip, 2020-06-28:
I do not believe so, I soldered on a socket that came with the board I
ordered. It's is the ozzmaker gps board without SIM card.
They should be in the correct location.

megadip, 2020-06-28:
I figured it out! Haha I'm an idiot, had serial port configured incorrectly!
Thanks for trying to help!

dcecchino, 2020-06-28:
My issue was with the soldering just make sure nothing is bleeding over.

rhole, 2020-06-29:
nah, that's so easily done and you found it yourself. congrats