cyoung/stratux #330: RY835AI AHRS stopped working - troubleshooting help?

JohnOCFII, 2016-03-13:
1. Stratux version: v0.8r1 (also tried images for .6 and .7)
2. Stratux config:

    SDR
      [ ] single
      [ X] dual

    GPS
      [ X] yes
      [ ] no
      type:

    AHRS
      [ X] yes
      [ ] no

    power source: wall wart and Anker (same behavior each)

    usb cable: thick mono price cable 
[stratux.log.zip](https://github.com/cyoung/stratux/files/171281/stratux.log.zip)

3. EFB app and version: (WebUI)

    EFB platform: (iOS 9.2.1)

    EFB hardware: (iPad Air 2)

4. Description of your issue:

As I prepared to put my components in a 3D printed case, I've tested as I've made changes.  After coating the rear side of the RY835AI in liquid electrical tape, I re-connected the wires to the Raspberry Pi, and the AHRS is no longer working.  
- The Status screen will show the green check box for AHRS for about 6 seconds, then show the red X for about 2 seconds.  This cycle repeats.  
- On the GPS/AHRS page, Heading, Pitch, Roll, and P-Alt all show 0, and do not move at all when the RY835AI is moved.
- I enabled log, and the following message shows up multiple times: 
2016/03/13 23:05:36 readBMP180(): input/output error

This was working just before I applied the coating.  I did not coat the sensors on the front side of the board.  GPS is working fine.  I re-confirmed continuity for all connections, and confirmed that there was no short between the SCL1/SDA1 wires.

I re-tried the last three images (which had worked successfully for me).  I am assuming this is a hardware issue on my end, but wanted to see if there were any ideas on what to try next.

Thanks,

John

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.  (Logs attached)

Nokomis449, 2016-03-14:
I had similar green check/red X blinking behavior once after disassembly/reassembly.  Thought I had fried the RY, but my 4th check of the wiring finally revealed that I had transposed two of them, possibly the send/recv.  Your eyes will see what your brain tells it to see.

JohnOCFII, 2016-03-14:
> I had similar green check/red X blinking behavior once after disassembly/reassembly. Thought I had fried the RY, but my 4th check of the wiring finally revealed that I had transposed two of them, possibly the send/recv. Your eyes will see what your brain tells it to see.

Certainly possible.  On the RY, there are only two wires involved (as power and ground are good from the GPS side.  I have swapped them around, and the behavior does not change.  I'm hoping a developer can tell me if the above-mentioned log message [ **readBMP180(): input/output error** ] provides any additional clues.

Thanks,

John

AvSquirrel, 2016-03-14:
First, make sure the modules are being detected with i2cdetect. As su:

````
apt-get update
apt-get install i2ctools
i2cdetect -y 1
````

The gyro has hardware address 0x68, and the pressure sensor is 0x77. If both are connected and talking to the bus, i2cdetect will show this:
````
      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
 00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
 10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
 60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
 70: -- -- -- -- -- -- -- 77        
````

bjax, 2016-03-14:
Looks like that should be

  apt-get install i2c-tools

JohnOCFII, 2016-03-15:
> The gyro has hardware address 0x68, and the pressure sensor is 0x77. If both are connected and talking to the bus, i2cdetect will show this:

Thanks - I'll check it out this weekend when I'm back in town.

JohnOCFII, 2016-03-19:
I get nothing.  I can pull the wires and re-solder, in case there is something I can't see and the multi-meter can't detect.  Of course, that just doesn't make sense, as I tested it after I soldered everything in place.  The only change between that test, and the failure situation was covering the back side of the RY board with liquid electrical tape.

pi@raspberrypi ~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@raspberrypi ~ $

skypuppy, 2016-03-19:
Your Pi is definitely not seeing the AHRS unit at all, with i2cdetect 
output like that.  Has to be 1) electrical, or 2) i2c comms not enabled 
in the kernel.

JohnOCFII, 2016-03-19:
> Your Pi is definitely not seeing the AHRS unit at all, with i2cdetect
> output like that.  Has to be 1) electrical, or 2) i2c comms not enabled
> in the kernel.

It was electrical.  I pulled off the liquid electrical tape, and there was a whisker of a wire shorting on the SCL1/SDA1 wires.  Not sure why the continuity check didn't catch it.

cyoung, 2016-03-19:
Glad you got it!

closed 2016-03-19 by cyoung