cyoung/stratux #332: ForeFlight 7.6 incompatible with GDL90 AHRS output

AvSquirrel, 2016-03-15:
1. Stratux version: v0.8r1 (AvSquirrel@f7c9b7e)
2. Stratux config:

    SDR
      [ ] single
      [X] dual

    GPS
      [X] yes
      [ ] no
      type: U-blox

    AHRS
      [X] yes
      [ ] no

    power source: 2A wall-wart

    usb cable: Attached

3. EFB app and version: ForeFlight 7.6

    EFB platform: iOS 9.2.1

    EFB hardware: iPad Air 2

4. Description of your issue: 

ForeFlight 7.6 is actively incompatible with GDL90-formatted AHRS messages. Enabling AHRS in the web UI will completely disable the "FreeFlight" connection to Stratux, causing weather, traffic, and ownship to disappear. Disabling AHRS restores the behavior seen on ForeFlight 7.5.2.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

224XS, 2016-03-15:
Sadly, I don't think this is an external bug (implying that, once notified, someone would be interested in fixing it) but rather a cold corporate calculation by ForeFlight probably in response to pressure by Appareo.
Perhaps a useful strategy for mitigation might be to flood the ForeFlight booth at SnF and Oshkosh, maybe even badgering Tyson (who is otherwise a seemingly nice guy) and other honchos with requests.

skypuppy, 2016-03-15:
Or, vote with your wallets and not fund Foreflight anymore.

Nokomis449, 2016-03-15:
For those who want AHRS via Stratux, I'd like to invite you over to the dark side - iFly.  Actually I should call it the Bright side because the guys at Adventure Pilot are extremely responsive to pilot requests.  Their latest versions explicitly support Stratux and as a beta tester for them I have personally spent hours and sent logs to their programmers specifically for their Stratux AHRS development (i was using the RY).  Their app runs on iDevices, Android, and their dedicated hardware.  I have it running on my Android phone, Android tablet, and their iFly 740 hardware which is incredibly sunlight readable.  They have a 30 day free trial of the full-blown app in the App Store and Google Play.  They don't play "upcharge" games like others; you buy the VFR version or the IFR version and all the candy comes with each.  For the icing on the cake, check out their features and then compare their subscription price to the "industry leader" that wants to lock you into their preferred hardware.

braddward, 2016-03-15:
I'll give iFly a try.  I started with ForeFlight because that's what my CFI used.  I don't really like their UI though.

I like the Garmin Pilot UI but sadly it only works with their hardware (I have their GPS).  

FlyQ seems OK - but their UI on the maps seems to freeze at points while it must be recalculating which is annoying.  I like how they overlay wind direction and speed on the airport diagrams and can overlay those on the maps.

skypuppy, 2016-03-15:
I'll second iFlyGPS as a good product.  They don't respond positively to 
all pilot requests but they do for a lot of them.  I just paid for 
another year (yeah, unfortunately it is a subscription-based business 
model and they are VERY closed-mouth about their methods and algorithms, 
which makes it difficult to trust their software completely, but 
compared to the paid and free market, it is worth it, IMO.)

Skypuppy

pp4life, 2016-03-15:
> Or, it may be as simple as finding out through a polite discussion what they want/need on their end to support compatibility, then implement that.

Hmm, that's a good idea. Maybe I'll send them an email. I have a Foreflight subscription until next November so I have some time to decide. Hopefully there will be a clear decision made if they are going to never allow Stratux AHRS by then.

Nokomis449, 2016-03-15:
For those that weren't around or don't remember, here's a similar discussion from last December:
https://github.com/cyoung/stratux/issues/138

JohnOCFII, 2016-03-15:
> For those that weren't around or don't remember, here's a similar discussion from last December:
> #138

Which, begs the question -- if it was already disabled as of 7.5.2, why did ForeFlight take this additional step?    I liked to see the AHRS, at least in the WebUI, if not in ForeFlight.

ssokol, 2016-03-15:
I don't think this is purely the AHRS GDL-90 messages. I'm not getting the FreeFlight device on my iPad in FF (works fine in other apps). Guessing it's our heartbeat/status messages - FF detects them and shuts us out. May need a FF compatibility mode that mimics the FreeFlight exactly.

ssokol, 2016-03-15:
Oh, oddly, the same build works fine on my iPhone. Both are 1991.

jzeevi, 2016-03-15:
I'm successful with any FF version with the AHRS turned off with the latest release. 

Josef
Sent from a phone with a tiny screen

braddward, 2016-03-15:
I disabled AHRS and ForeFlight seems to be working again.

Thanks

AvSquirrel, 2016-03-15:
@ssokol -- Testing on a stock image of v0.8r1 and an iPad, the FreeFlight indicator consistently takes about 20 seconds to come back after disabling AHRS.

(That's after dealing with first-boot weirdness, as noted in #316)

AvSquirrel, 2016-03-16:
@JohnOCFII 

> I liked to see the AHRS, at least in the WebUI, if not in ForeFlight.

Already accounted for in my development branch as of last night.
- One slider will enable the hardware and cause AHRS output to the UI / http 'getSituation' interface
- One slider will disable/ enable GDL90 AHRS output for all connected clients
- One slider will disable ALL GDL90 output, and enable ForeFlight port 49002 GPS, traffic, and AHRS.

cyoung, 2016-03-16:
Yes, ForeFlight doesn't use AHRS over GDL90 on any platform afaik - especially not the format that ours sends.

So this is an active attempt by ForeFlight to fingerprint Stratux output and block it.

pp4life, 2016-03-16:
I wrote Foreflight an email asking about Stratux. I basically said if it comes time to renew my subscription in Nov and AHRS is working in Stratux, the benefits of Foreflight over any other EFB does not justify the cost of having to purchase a Stratus. So I'd be switching if that was the case. I even commented I'd be willing to pay extra $/year to have Stratux work with Foreflight instead of having Logbook or something else I don't use.

In any case, this was their response
> Hi Nick,
> 
> Thanks for letting us know your feelings about Stratux. I will make sure to share your message with the right people.
> 
> If you've been following the changes to ForeFlight Mobile in the past year, you'll know we have invested significant developer and QA resources to add support for the Dynon Skyview avionics system, the FreeFlight RANGR ADS-B receiver, the Garmin Flight Stream 210 (and associated avionics and ADS-B receivers), and most recently the Avidyne IFD540/440. We're committed to continuing to add support for other devices, but I can't share any specifics about "which" or "when."
> 
> We appreciate your passion for aviation and experimentation, and we hope we will earn your renewal business later this year. Stay tuned to http://blog.foreflight.com for news.
> 
> Blue skies,
> 
> -Josh
> 
> Josh Berman
> Pilot Support Team
> ForeFlight, LLC.

cyoung, 2016-03-16:
I wouldn't hold my breath. I've spoken with Tyson on this and the answer was no. That they're training their support team to BS people out of money is fairly sad.

JohnOCFII, 2016-03-16:
While ForeFlight is supporting more than just the Appareo Stratus, it appears Stratus is still the only portable system they support.

https://www.foreflight.com/connect/

I do wonder if Stratux had a different name, if there would be less push-back for support (or at least being officially ignored) by ForeFlight.  From looking at their web site, the Appareo connection is still pretty strong, and I'm sure Tyson doesn't want to damage that business relationship.

pp4life, 2016-03-16:
> I wouldn't hold my breath

I'm breathing freely and basically have the "non renewal" stance if Stratux is shutout when renewal time comes around in Nov.

cyoung, 2016-03-16:
>I do wonder if Stratux had a different name, if there would be less push-back for support (or at least being officially ignored) by ForeFlight. From looking at their web site, the Appareo connection is still pretty strong, and I'm sure Tyson doesn't want to damage that business relationship.

Far be it from me to guess what they want and try to do a dance to appease them. If they want something then they can speak up, otherwise we'll look at other solutions like adding FlightStream 210 emulation and keep on truckin'.

bkwny, 2016-03-16:
If a particular software vendor wants to block or ignore your hardware, he can always find a way to do it.  For this reason I suggest you drop ForeFlight interoperation as an objective and continue to work constructively with the EFB vendors who want to exploit your hardware.

ssokol, 2016-03-16:
@bkwny - Actually, this is one of the few situations where they can't "always find a way to do it." They've committed to supporting a number of certified, installed devices - FreeFlight RANGR, FlightStream 210, Avidyne IFDX40. Those systems are already in the field - installed in certified aircraft. There is no practical means for them to alter those systems to include a "secret handshake" or some form of encryption. They also use a public protocol: GDL90. We just have to do a true emulation and we're set.

I will be sending Chris a check for quite a bit - the $10/unit donation from FlightBox. Perhaps he could use part of that to buy a FreeFlight RANGR FDL-978-RX (the $2k receiver) or RANGR Lite FDL-978-XVRL (the $3600 transceiver). Another $999 would get a FlightStream 210 (though I'm guessing that might be encrypted, and will certainly have a handshake).

Thoughts?

bkwny, 2016-03-16:
Sure.  All ForeFlight LLC has to do is decide that they will no longer exploit hardware that doesn't cooperate on a ForeFlight handshake.  The vendors of the systems you mentioned will be provided the handshake details under nondisclosure agreement.  The vendors will in short order release firmware updates that implement the handshake.  In my view, the bottom line is this:  why try to play with somebody who clearly wants not to play with you?

cyoung, 2016-03-16:
We don't pay with "ForeFlight LLC", we do this for the users.

cyoung, 2016-03-16:
@AvSquirrel - for your ForeFlight feature, check out this:

https://github.com/cyoung/stratux/blob/f47a224c1d0685f5d3a53ae230d435545d16e94b/main/network.go#L231-L241

Can be done/detected automatically. I've just verified that FF 7.6 sends these packets just the same as before.

ggoodknight, 2016-03-17:
ForeFlight interoperability has been announced by Avidyne
http://www.avidyne.com/news/press.asp?release=329

jzeevi, 2016-03-17:
Oh yay.

 

http://www.avidyne.com/news/press.asp?release=329

AvSquirrel, 2016-03-18:
@JohnOCFII 
> I liked to see the AHRS, at least in the WebUI, if not in ForeFlight.

A control for the AHRS GDL90 output is included in #340, along with a mode that will output ONLY the `XGPS`/`XATT`/`XTRAFFIC` ForeFlight flight simulator format:
[image]

@cyoung 
> keep on truckin'.

++. We do what we can, as long as we can. 

> Can be done/detected automatically. I've just verified that FF 7.6 sends these packets just the same as before.

Cool. Automatic control of the output format -- especially on a per-port basis -- would be a great feature. I'll have to take a look at that.

MSGTFE, 2016-03-18:
I am not sure if this is common knowledge or even useful for this problem, I was troubleshooting an unrelated issue with them and they instructed me to accomplish this.  I am not a software guy so I was not sure if there was any useful/relevant information in the logs this created.  I tried reading them before I sent them the email but I was not sure what I was looking at.  It creates an email with a few attached logs that you could send to anyone.  

Open ForeFlight Mobile, then tap More > Settings, and tap the word "Settings" in the top menu bar 3 times in quick succession. This will create a Diagnostic email to us (NOTE: It can take up to 30 seconds to create the email, depending on your iPad).

tweihs, 2016-03-18:
> So this is an active attempt by ForeFlight to fingerprint Stratux output and block it.

@cyoung's conjecture is not factual (or constructive in building any sort of relationship). ForeFlight is not actively working to block Stratux, and this issue isn't a ForeFlight bug. With 7.6, we learned from a beta tester after Apple had approved the update that Stratux devices running a newer software build were affected by a code mod we made that rejects unsupported message types (older Stratux builds were not affected). 

We don't officially support Stratux, so we don't test it in our labs or regression suite. As for supporting Stratux, when @cyoung asked me if ForeFlight would support Stratux, this was my response:

> The short answer is no, as "support" has deep roots. Among other issues, pilots use these devices to avoid weather and traffic, so control of quality in what we support is paramount. As I mentioned previously though, love seeing the innovation.

Reading on, @cyoung wrote:

> That they're training their support team to BS people out of money is fairly sad.

I don't think that is accurate or fair. Josh was was explaining to @pp4life that we've been investing in connecting to other devices and listed the initiatives. 

Every device ForeFlight supports is an up front and (not insignificant) ongoing investment. We have an avionics / accessories lab, we regression test all the hardware we support, we flight test it, we have engineers dedicated to triaging issues. Stratux isn't one device - it is many different devices (or I predict will be) and different versions (as we saw with 7.6). The idea that we have to support all these different combinations and versions, triage issues, etc. isn't as fun as working on next generation (post-ADSB) tech.

In my opinion, ForeFlight's been responsive, courteous, and professional with @cyoung, in addition to being supportive in general of the innovation.

skypuppy, 2016-03-18:
@tweihs, well there is "official," publicly-announced support and there 
is "unofficial" support.  It *appears* to the ever-growing stratux 
community (most of which are pilots) that FF is blocking stratux by 
design.  FF doesn't have to spend any money to unofficially "support" 
stratux and it's community.  It might even boost your sales if you 
"unofficially support" this effort.
Gdl90 seems to be ubiquitous in the ADS-B world for now.
I applaud pilots who do not purchase your products until this issue is 
resolved to the satisfaction of all, especially pilots who fly 
experimental aircraft.  That is, until the FAA changes policy (which 
seems imminent) about what can "legally" be used in certificated aircraft.

David Merchant
PP-ASEL of both certificated and experimental aircraft

cyoung, 2016-03-18:
Well then, we can make it easy and just be straight.

>@cyoung's conjecture is not factual

Then what DID happen? Rejecting invalid GDL90 messages? When we stop sending the AHRS GDL90 messages everything is fine. We still send @HiltonG's status message and it works fine also - so is that message valid/processed by ForeFlight?

>(or constructive in building any sort of relationship)

We have nothing but double-speak to go from, there is no relationship. If you want to support Stratux, go ahead and ask one of the many EFBs that officially support Stratux and you'll find that I'm not that hard to deal with and Stratux is easy to support. If you're asking for more than that, well, I do this for free and thus do not have the time nor disposition to kiss ass.

>>That they're training their support team to BS people out of money is fairly sad.

>I don't think that is accurate or fair. Josh was was explaining to @pp4life that we've been investing in connecting to other devices and listed the initiatives. 

If the answer is no then there isn't any reason to create an artificial gray area. "We're committed to continuing to add support for other devices (...)". Well, Stratux is not one of them.

egid, 2016-03-18:
>Perhaps the Stratux community can come up with a security paradigm that works for the industry.

@tweihs I understand that one of your primary concerns around official (or even _unofficial_) support for any experimental ADS-B unit is liability. It sounds like a secondary concern is an inability to ensure data quality from those units.

What, if anything, can be done to address your concerns?

1. Would you be willing to accept a GDL-90 stream that contains _only_ wx/traffic/gps data, excluding AHRS entirely? This would exclude the most immediately life-threatening and liability-inducing data, were it incorrect.
2. It's additional work on ForeFlight's end, but what about firing a legal warning if the Stratux heartbeat is detected, reminding pilots that they're using experimental hardware? I mean, from a legal perspective ForeFlight is not approved for primary navigation in the first place, but that seems like it could fill some gaps.
3. I'm not sure how coming up with a 'security paradigm' solves your problems, but if it would - could you provide more information on how that would help ForeFlight trust experimental units? Are we talking some sort of handshake to make sure it's the 'right' unit?
4. If you don't intend to ever verbally state support for Stratux or any other experimental ADS-B receiver, and you do in fact intend to passively or actively prevent ForeFlight from receiving data, could you not just tell us?

I've been using ForeFlight for almost 10 years, and I love the product. I'm frankly tired of people suggesting that we all jump ship at the drop of a hat because of issues with Stratux. That said, I understand their frustration. A lot of what your support folks have said can be read multiple ways, and unless they're following policy or coming up with their own, I think ForeFlight and its customers would benefit from clearer language.

HiltonG, 2016-03-18:
"If you don't intend to ever verbally state support for Stratux or any other experimental ADS-B receiver, and you do in fact intend to passively or actively prevent ForeFlight from receiving data, could you not just tell us?"

Tyson has already publicly stated this - no need to discuss or debate this point; i.e. if FF detects Stratux, it deliberately stops listening.  (As always, feel free to correct me if I'm wrong, but it's on the forums for all to read).

Hilton

egid, 2016-03-18:
@HiltonG I guess I can simplify my request down, then, although I'd love for you to be wrong ;).

@tweihs: What – if anything – can we do to satisfy ForeFlight (or you personally) that Stratux ADS-B and GPS data is accurate enough to gain tacit or explicit support? Would it be unreasonable to ask for some criteria that could be aimed at and met?

jzeevi, 2016-03-18:
As of is moment, FF supports Stratux unless AHRS messages are received.

egid, 2016-03-18:
@jzeevi That's not how ForeFlight sees it. I think they're using 'support' as a legal term, as in 'We support and guarantee the information from [device x]'.

jzeevi, 2016-03-18:
Has anyone gotten that definition from them? What I meant was they support it in that it works. 

I'm puzzled how they can guarantee anything since the thing isn't approved for primary flight. Why are we quibbling about this?

I'd sure like to know what (from them) what it would take. 

Josef
Sent from a phone with a tiny screen

egid, 2016-03-18:
@jzeevi [Tyson's comment in this thread reiterates that](https://github.com/cyoung/stratux/issues/332#issuecomment-198527879) - Tyson is CEO of ForeFlight, if that context was missing. :)

jzeevi, 2016-03-18:
I see. I know who Tyson is, but didn't recognize the handle. 

Still, I don't see active "we're not supporting it". 

I really wish I could get someone from FF to talk to me about the requirements they have for being on the official list. :-)

In fact, there's several things I'd love to discuss with Tyson or Adam. 

Josef

egid, 2016-03-19:
@jzeevi here's what I'm talking about:

>We don't officially support Stratux, so we don't test it in our labs or regression suite. As for supporting Stratux, when @cyoung asked me if ForeFlight would support Stratux, this was my response:

>>The short answer is no, as "support" has deep roots. Among other issues, pilots use these devices to avoid weather and traffic, so control of quality in what we support is paramount. As I mentioned previously though, love seeing the innovation.

jzeevi, 2016-03-19:
I get it. I'm just wanting to open the conversation. If they ever do actively deny stratux, I'll consider my options then. 

Josef
Sent from a phone with a tiny screen

skypuppy, 2016-03-19:
Give them a way to make profit from it.  Yeah, like we all want to do THAT.

bsdcat, 2016-03-19:
Hi, um, ForeFlight developer here- the one that implemented the block, actually. I did some basic testing on my own time to see if Stratux would be effected, and it wasn't. (The version I had handy had not yet introduced this message.) Oops.

Then we got the report from a beta user after the app had been approved, and I looked into it a bit more: it was apparent that with a simple configuration change, everything worked exactly like it had before. So we shipped, because the problem could be addressed without code changes or rebooting the release process (which costs money in addition to the delay). Which led to the appearance that we were trying to block Stratux, but were so incompetent that there was an easy workaround. Ouch.

To answer @egid's questions...

1. This is the current state of affairs. Disable AHRS in the Stratux web interface, and it behaves exactly the same as before.
2. Stratux is itself so easily modified, and other experimental hardware not so easily identified, that I don't think this does anything meaningful to convey experimental hardware usage to a pilot. If you're using a Stratux, I expect you to be aware of that fact.
3. That's above my pay grade.
4. We can't always talk about everything we're doing. It's a business. But our goal with this change was not to block Stratux and we don't intend to block Stratux.

cyoung, 2016-03-19:
@bsdcat - is 0x4C45 the message that you refer to?
https://github.com/cyoung/stratux/blob/master/main/ry835ai.go#L1065-L1104
This is the one that seems to be causing the issue and was added in Sept 14, 2015/v0.3b2.
You maybe didn't have AHRS enabled in your test setup, either that or you're using an extremely old version.

If you have test cases that you can share then we could put something together to run tests.

bsdcat, 2016-03-19:
Hmmm... my mistake. I misread the diff for e15c389 and thought that message was introduced Feb 15. My characterization was wrong- I guess it was strictly to my AHRS setup. Like I said, it was quick and dirty ad hoc testing.

egid, 2016-03-19:
@bsdcat thanks for the responses. That clarifies a lot.

I've been operating on the assumption that ForeFlight isn't trying to actively block anything, but there's enough confusion and enough folks in here that don't seem to trust you guys that it's worth trying to get more specific answers about this stuff.

Thanks for doing some rough testing of stratux :) I think many of us would be willing to widen your beta coverage if that's something that could help.

braddward, 2016-03-19:
@bsdcat - thank you!  Glad to hear it wasn't intentional.

AvSquirrel, 2016-03-19:
@bsdcat -- thanks for the explanation!

Considering the speed at which this project is moving and the number of possible hardware iterations that builders can assemble, I think it's natural that you're going to see a lot more corner cases than you would with one of your typical hardware partners' projects. I'll echo @egid's offer to widen the beta coverage. More eyes on the project from both sides makes this less likely to happen.

@tweihs -- perhaps a better way to phrase what I'd personally like to see at this point in the Stratux lifecycle is "tolerance", rather than "support".

"Support" is all of the validation / qualification that you described. It is paying your engineers, developers, and customer service staff to picking up the phone, enter tickets, and research answers to paying customers. Support is what I expect when the AHRS heading from that brand new Garmin GTX345 displays a red "X" on the synthetic vision display, and causes the display to go sideways on takeoff. (That's *not* a theoretical example. I submitted a ticket about an hour ago.) It's all the things that are part and parcel of delivering and guaranteeing a high-quality product.

"Tolerance" costs you little or nothing, and generates goodwill. It is allowing us figure out how to interface with your product -- or better yet, agreeing on an interface so neither dev team accidentally breaks anything. It is opening up the betas so *we* can validate that our code doesn't break when you improve yours, so we can spread around the validation work.

jzeevi, 2016-03-19:
Thanks for the input, @bdscat. I do have a question - since all of the non-certified HW (Stratux, stratus, or any other portable device) is advisory, why (in the discussion) does Stratux seem to get pointed out?

And, as a pilot, I agree and would extend your statement - it's all advisory and the pilot had darned well better understand that.

bsdcat, 2016-03-21:
@AvSquirrel There's not much I can say or commit to.

@jzeevi There are plenty of reasons. One is that since development of Stratux is public, bugs that crop up in development are much more visible. A big one, too, is that regardless of certification there is still a more tightly controlled review, validation, and QA process with e.g. Stratus, which sets our minds more at ease. I'm not sure I can discuss some of the other reasons, but it isn't just trying to push Stratus on everyone.

cyoung, 2016-03-21:
Can we expect the change (whatever it was) to be rolled back/repaired or do we continue telling users to turn off AHRS if they expect it to connect to ForeFlight?

cyoung, 2016-03-21:
@tweihs, @bsdcat - Here's what we'll do. We will add back in the code that listens for "*-ffm-udp" broadcast messages that ForeFlight sends. When this is detected, a warning will be issued on the status page as below. AHRS will be disabled on the unit at this point.

In addition, we'll revise our documentation to let users know that we do what we can on our end (Stratux) to make things work but to expect issues when using ForeFlight as their EFB.

This will set clearer expectations with users and seems like the fairest way that we can deal with the situation from our end.

[image]

bsdcat, 2016-03-21:
@cyoung Yes, plan is to fix it.

Regarding the warning, I understand your perspective, and that seems reasonable.

cyoung, 2016-03-21:
>@cyoung Yes, plan is to fix it.

Excellent. With the above changes we will consider this issue closed, but if you'd like to revisit this after making changes on ForeFlight's end then please still comment.

closed 2016-03-21 by cyoung

bsdcat, 2016-07-14:
Follow up: we have released ForeFlight Mobile 7.7, and this bug is fixed. I apologize that this ever came up, and I'm sorry it took so long to get a new release out.

bsdcat, 2016-07-14:
To test this, you'll need to be connected to a RY835ai with AHRS, and run a build of Stratux with line 500 of main/network.go commented out (the line from 65537db that disables AHRS when FFM is detected), then enable AHRS. AHRS messages from the Stratux will not affect the SV display, but neither will they trigger any change in ForeFlight Mobile behavior.

You could also somewhat 'fake' test it by by adding `makeAHRSGDL90Report()` after `makeOwnshipGeometricAltitudeReport()` on line 573 of main/gen_gdl90.go - that should cause Stratux to send AHRS messages, regardless of whether you have a RY835ai connected, have meaningful pitch and roll data, or have enabled / disabled AHRS.

cyoung, 2016-07-14:
Thanks for the update!

AvSquirrel, 2016-07-15:
@bsdcat - thanks!

Stratux release v1.0r1 (945c8bf6) removed the 'FFM' message detection, and standard GDL90 messages are now working correctly in FF v7.7 (traffic and ownship are displayed) with AHRS active and Stratux sending port 4000 GDL90 AHRS messages to the iPad client.

ScraboTower, 2016-07-17:
Sorry I am late to the party on this one, not being a FF user.

So how does this link back to Tysons comments on AOPA site , did they try a block of AHRS and that block caused more issues ?

[1 more reply]

commit 65537dbd (2016-03-21): Issue a warning and disable features when FF broadcast messages are received.

Fixes #332.