cyoung/stratux #333: FlyQ 2.2.1 altitude 65,200 feet too high

yasbean, 2016-03-15:
1. Stratux version:v0.8r1
2. Stratux config:

    SDR
      [ ] single
      [ x ] dual

    GPS
      [x ] yes
      [ ] no
      type:RY835AI

    AHRS
      [ x] yes
      [ ] no

    power source:RAVPower

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) FlyQ 2.2.1

    EFB platform: (e.g., iOS 9.2) iOS 9.2.1

    EFB hardware: (e.g., iPad Mini 2) iPAD Air 2

4. Description of your issue:

FlyQ is displaying altitude about 65,200 feet too high (i.e. sitting at home at about 24 ft MSL, it says I am at 65,236 MSL).  UI GPS status has it more accurate at about 60ft +- 110ft.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

skypuppy, 2016-03-15:
Are you saying this is a cheap way to get high?

yasbean, 2016-03-15:
I was surprised to see traffic at -391 (k) and climbing!

AvSquirrel, 2016-03-16:
I'm going to take a wild stab at the air, and guess that your local altimeter was about 30.25" this afternoon, and that the web UI was showing a pressure altitude of -300'.

What's you're seeing is consistent with FlyQ using an unsigned 16-bit integer to store pressure altitude. Those have valid ranges of 0-65535, and dealing with values outside the that range is like dealing with a 5-digit odometer: negative values will roll over over backwards, and larger values rollover forwards.

Stratux must have sent a pressure altitude that decoded to -300. FlyQ would have read that wrote it to some internal variable, where it rolled over to 65236'.

I'd recommend sending a quick note to FlyQ support to tell them what you saw.

skypuppy, 2016-03-16:
i guess no one at FlyQ fly's in Death Valley and doesn't expect anyone 
else to.  Or Dead Sea,or ...

bkwny, 2016-03-18:
I reported this to FlyQ efb support on Mar 15.  The next day they wrote back and said they would look into it.

yasbean, 2016-03-18:
Yup, Steve asked me to provide a FlyQ log, so they should know what is
going on.  I did not check the altimeter setting that day, but the UI was
showing me at apx. 60' +- 110', so I guess FlyQ is averaging it out to be
-MSL.

cyoung, 2016-03-21:
Update?

bkwny, 2016-03-22:
I've heard nuttin'.

yasbean, 2016-03-22:
No news on my side, either.

Curtis Smith

cyoung, 2016-03-23:
I put this into `heartBeatSender()`:

```
mySituation.Lat = 44.00
mySituation.Lng = -85.00
mySituation.HeightAboveEllipsoid = 500
mySituation.LastFixLocalTime = stratuxClock.Time
mySituation.Alt = 2000
mySituation.NACp = 11
mySituation.Pressure_alt = -100
mySituation.lastTempPressTime = stratuxClock.Time
```

Altitude shows as 65,436. So they interpret the Ownship altitude as MSL alt instead of pressure alt, and don't account for it potentially being negative - just as @AvSquirrel first thought.

I've sent in a report to them with more details, so we'll keep waiting.

yasbean, 2016-04-14:
I just received a notice from Seattle Avionics: "This ticket is over 30 days old and has been closed."  I have not had a chance lately to see if they have fixed the error.  As they closed their ticket, I will close this, unless someone else has something to add.

cyoung, 2016-04-15:
It's not fixed yet. I've been in contact with them and have tested their latest beta - but a fix isn't out yet.

cyoung, 2016-12-01:
Has been fixed.

closed 2016-12-01 by cyoung