cyoung/stratux #336: No new updates at http://updates.stratux.me/builds/

Ergonomicmike, 2016-03-16:
1. Stratux version: v0.8r1
2. Stratux config: N/A

    SDR
      [ ] single
      [ ] dual

    GPS
      [ ] yes
      [ ] no
      type:

    AHRS
      [ ] yes
      [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

As a beta tester, am looking to experiment with the latest commits. (26 as of this writing.) I used to do git pulls; make; make install. Then we went to the automatic build model for betas. No new .sh's since March 10. Should I go back to git pulls for these later commits?

Ergonomicmike, 2016-03-16:
NeverMind. New one just popped up.

closed 2016-03-16 by Ergonomicmike

cyoung, 2016-03-16:
The builds are made on a Raspberry Pi that I keep hooked up at home. It's not quite dedicated for this purpose just yet but will be set up soon so that there are no interruptions. Back online now.

Ergonomicmike, 2016-03-16:
@cyoung I've got an old Pi B+ that I'll never use. Want me to snail it to you? (You'll have to get your own UPS tho. <grinning>)

cyoung, 2016-03-16:
Thanks for the offer - I'm working on getting the Pi3 up and running and that will be my main testing platform. The old Pi2 will be the builds machine. I've got my Pi B+ running as a PiAware feeder, the "free" flightaware enterprise subscription that results has proven useful a few times for the project.