cyoung/stratux #342: Ry835ai gps gpio "disconnected"

JEgasmanpa, 2016-03-20:
1. Stratux version:
2. Stratux config:

    SDR
      [ ] single
      [ x] dual

    GPS
      [ x] yes
      [ ] no
      type:

    AHRS
      [ x] yes
      [ ] no

    power source: adafruit powerboost 1000 with 3.7v LiPo battery

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) yes

    EFB platform: (e.g., iOS 9.2)yes

    EFB hardware: (e.g., iPad Mini 2) yes

4. Description of your issue:

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

JEgasmanpa, 2016-03-20:
After upgrade to v8r1 GPS (ry835ai) web ui "disconnected" AHRS ok.  USB gps works ok.

cyoung, 2016-03-21:
Looks like you've fixed it? Re-open if not.

closed 2016-03-21 by cyoung