cyoung/stratux #343: Not getting the PPM values

kalustian, 2016-03-22:
I was trying to run some commands in mi Pi2 to get the PPM information. I tried to follow instructions on the stratux page and after doing the 3rd command I got below results - what I did wrong ?

NOTE - I am running stratux-v0.8r2-db130aab76.img

pi@raspberrypi ~ $ sudo su -
root@raspberrypi:~# pkill screen
root@raspberrypi:~# kal -g 48 -s GSM850
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM

Kernel driver is active, or device is claimed by second instance of librtlsdr.
In the first case, please either detach or blacklist the kernel module
(dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

usb_claim_interface error -6
Failed to open rtlsdr device #0.

Thanks-

skypuppy, 2016-03-22:
is stratux running while you try this?

kalustian, 2016-03-22:
yes, stratux running and connected via ethernet cable so I can SSH

Ergonomicmike, 2016-03-22:
service stratux stop  ??? (Just hacking.)

kalustian, 2016-03-22:
after used the following command: sudo service stratux stop the scanning command works, but it does not returned any channel#

here is the results after 3-4 min:

pi@raspberrypi ~ $ sudo su -
root@raspberrypi:~# sudo service stratux stop
[ ok ] Stopping Stratux ADS-B Receiver: stratux.
root@raspberrypi:~# kal -g 48 -s GSM850
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 270833.002142 Hz
[R82XX] PLL not locked!
Setting gain: 48.0 dB
kal: Scanning for GSM-850 base stations.
GSM-850:
root@raspberrypi:~# 

Looks likes it did not return any channel#

kalustian, 2016-03-22:
Note...I used kal -g 48 -s GSM900 instead and it returns the following channel:

chan: 119 (958.8MHz + 13.753kHz)	power: 92899.08

QUESTION - should I used GSM900 instead GSM850 for the PPM correction ??

Thanks-

Axtel4, 2016-03-22:
You can use GSM900 for GSM850.

skypuppy, 2016-03-22:
especially if you have a cordless phone nearby.  :)

kalustian, 2016-03-22:
OK, I will try EGSM, DCS or PCS  and test drive - thanks lot

jthawkes, 2016-03-23:
Is the ppm info worth worrying about? A few PPM at 1090Mhz with 2Mhz bandwidth is probably not going to make any difference.  Spend the money on an 820T2 model.

cyoung, 2016-03-23:
>chan: 119 (958.8MHz + 13.753kHz) power: 92899.08

Looks like you picked up a channel that you can kal off of. Now do `kal -g 48 -c 119`.

closed 2016-03-23 by cyoung