cyoung/stratux #345: Throwing out another idea

skypuppy, 2016-03-22:
1. Stratux version:  0.8r2
2. Stratux config:

    SDR
      [ ] single
      [ x] dual

    GPS
      [ x] yes
      [ ] no
      type: MTK3339 on serial, sirf on usb

    AHRS
      [ x] yes
      [ ] no

    power source: all wart

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)  iFlyGPS 9.4

    EFB platform: (e.g., iOS 9.2): Android 5.x

    EFB hardware: (e.g., iPad Mini 2): Samsung Galaxy Tab S 10.1

4. Description of your issue:
The RPi-3 brings up an interesting possibility for an addition or change: using Bluteooth (LE) instead of wifi.  Does all EFB hardware have BT?  The main advantage is it consumes much less power than wifi.  Another advantage is for clubs/groups that fly with stratux because BT already has the capability for unique connections along with lower power radio so that fewer planes will see the same signal during close-in flying.  Wifi can be used in that same config by turning on WPA or something but the signal collision overhead is still incurred.
Thoughts?  A BT dongle is just as cheap as a wifi dongle.

ScraboTower, 2016-03-22:
But isn't Apple for example, very tight about the BT standards used to connect to their devices ?

skypuppy, 2016-03-22:
I don't know. 

 /rant on/ I am *really* sick of these companies locking in their products trying to create "vertical markets."  Garmin, Apple (in many areas), and so on just make me angry and I vote with my wallet.  Let them "sit on the shelf and rot."
/rant off/

I do know that my chosen EFB, iFly, does not have BT capability at this time.  However, lot's of them do.

Nokomis449, 2016-03-23:
Skypuppy I'm sure you're talking about iFly's dedicated hardware (720, 740) which are wifi only, but the iFly app on Android and iDevices can use BT recievers as well as wifi receivers.  Just wanted to clarify that for others.

peepsnet, 2016-03-23:
BLE is not avalible on all devices. Bluetooth is but the low energy is not.

Wifi is though

Ergonomicmike, 2016-03-23:
On iFly: I was wondering whether iFly could use BT. So I tried transferring a Flight Plan from one tablet to the other paired via BT. No joy.

skypuppy, 2016-03-23:
I remember iFly via tablets, etc., as not being bt capable.

Nokomis449, 2016-03-23:
The BT-only Dual XGPS is an option under iFly's "Connected Devices".  Someone on Amazon gave this review:

http://www.amazon.com/review/R1Q45F8B1TL55A

Nokomis449, 2016-03-23:
Re: ***On iFly: I was wondering whether iFly could use BT. So I tried transferring a Flight Plan from one tablet to the other paired via BT. No joy.***

Yeah I didn't mean to imply all networking functions will work on iFly via BT; only that Android and iDevices could receive ADS-B via BT devices.  I was specifically referring to skypuppy's BT/vertical market rant (with which I agree).  I wasn't thinking about flight plan sharing via BT.

mhrussell1, 2016-04-26:
I just bought the Pi3 and GPS receiver for use with Avare and CloudAhoy.  Unfortunately, I then found that CloudAhoy will only work via BT, so I am motivated to add a BT interface to stratux.  I write S/W for Linux systems for a living but I have not done any projects via Git so I am hesitant to commit taking on this task.  Also, I hope this is not an improper way to jump into an open source project.

skypuppy, 2016-04-26:
Just create another branch and have at it!  :)

cyoung, 2016-08-24:
Unable to add Bluetooth support. Not enough EFB-side support to be worth the time.

closed 2016-08-24 by cyoung

adambertsch, 2017-02-20:
Any chance we could try this for Avare w/ external I/O plugin and bluetooth on the pi3 as a first step?  I can work on this, but don't want to put in the effort if the pull request is unlikely to be merged.

cyoung, 2017-02-20:
Yes, if you want to add this it will be merged so long as everything works well.

cyoung, 2017-02-20:
#developers on Slack (see link on website) if you need help getting started.