cyoung/stratux #346: Display non-position mode-s traffic in efb

cubaneight, 2016-03-23:
Hi all,
 
just set up my stratux-receiver (1090), works great!
 
I fly out of Austria (Europe). Contrary to the US, we don’t have TIS-B or any comparable ATC-service available. So we can only rely on transponder signals (and flarm).
I just installed the latest  config-file (https://github.com/cyoung/stratux/files/164529/update-stratux-v0.8r1-fd936e7e1b.sh.zip) by  AvSquirrel. Now I can also see transponder-signals without location information in the statux-traffic-page.
 
Is it possible (at least as an option) to display such traffic in the respective EFB-program, eg ForeFlight? I know, there is no position information for that traffic. But we have altitude information and signal strength. With that, we could display the altitude difference to the own aircraft and at least an estimated distance-range where the traffic might be. Some commercial systems already use this technique, eg: [http://www.air-avionics.com/air/index.php/produkte/kollisionsvermeidung/trx-2000-57mm-kollisionswarnsystem](url)
 
This might finally look as follows (the green circle showing non-position-traffic):
[http://www.air-avionics.com/air/images/trxtraffic.png](url)
 
Have you got an idea how to implement that? Would be a great advantage for pilots outside the US.
 
Thank you!

Regards,
Dieter

224XS, 2016-03-23:
This is not a Stratux issue, but rather an EFB app issue. They do what they choose to with the data provided.

Targets without location data can be displayed on the Stratux web UI.  Signal strength as a means to calculate distance is a pretty poor method, especially when there is no control over antenna location or orientation. Just because someone sells such a product doesn't mean it actually works.

cyoung, 2016-03-23:
> Signal strength as a means to calculate distance is a pretty poor method

Is that certainly the case? My intuition says this as well but every time it comes up there seem to be a number of people that believe that it's a great idea.

I think this is what Zaon does/did? How well does that product work?

cyoung, 2016-03-23:
@Axtel4 - what is your opinion on this?

224XS, 2016-03-23:
Zaon had a carefully tuned array of antennae inside its case and was always on the glare shield ( that was the only way to see the display).
Zion is also out of business.  Again, the market does not control the entry of product or its quality, but only its longevity!

peepsnet, 2016-03-23:
How does this fit into the gld90 format. Can you send traffic that has no position into to stream?

Ergonomicmike, 2016-03-23:
We had a Zaon MRX and it was fantastic. I didn't care so much about how accurate the distance read out was. I cared that it alerted me to traffic near me. (5 miles or less.) And then told me the nearest aircraft's offset (from Mode C) and whether that traffic was climbing, descending or level, relative to me. I could make an evasive maneuver based on that. (Especially if the distance was decreasing.) It was like a poor man's TCAS.

Yes, an air transport would appear to be closer because their xponders are more powerful. But then, they move faster too. So no harm there. 

Yes, Zaon is out of business. I don't know why. Maybe ADS-B? Bad economy? Unfortunately, there was a part in their box that seemed to fail at the same time for all users. Prior to the Stratux, I was trying to buy another MRX but couldn't find any working units for sale.

Instead, we bought, and still use, a Monroy ATD-300. But it doesn't work nearly as well as the MRX did. The ONLY offsetting factor with the Monroy is that he offers a $300 phased antenna which can give azimuth info on the strongest target. (But since I wasn't happy with his product and his unwillingness to improve it, I didn't buy his antenna. So I can't report on how well (or not) that feature works.)

224XS, 2016-03-23:
Mike!

My point was that Zaon controlled the antenna and the location of the box on the glaresheild. No such luck with either point
in the Stratux world.

Ergonomicmike, 2016-03-23:
Actually, I used my own antenna for the MRX. So the odds are that I goofed up their distance predicting algorithm. I didn't care. It alerted me to traffic nearby (be that 3 or 6 miles) and that was fine with me. So if the Stratux and an EFB simply reported the altitude of the strongest signal/traffic near me (above some cuttoff threshold) absent distance, saying "Traffic nearby, your altitude and level," that would be better than nothing. If it wanted to display a metric for increasing or decreasing signal strength, so that I could see if the traffic was approaching, that would be nice too.

cubaneight, 2016-03-23:
@Ergonomicmike : correct, that's exactly what I think would be a great feature: alert me that there is traffic in my area, and xx freet above or below me. The change in signal strenght could trigger a symbol or a colour (eg green for weakening and red for increasing signal strenght).

The GDL-90 Protocol description says "A target with no valid position has Latitude, Longitude, and NIC all set to zero." That might mean that it is possible to have no-position-targets in the GDL90-protocol and it's up to the EFB-system how that is displayed.

Maybe it is possible to display such in formation directly in the stratux traffic page? That would avoid the problem that some EFB-Displays are unable to display it.

Regards,
Dieter

skypuppy, 2016-03-23:
Sounds like a recipe for a full-time second display/tablet.

Ergonomicmike, 2016-03-23:
@cubaneight There is already a commit (and I'm running it now) that shows non-ADS-B trargets on the Traffic page. https://github.com/cyoung/stratux/files/183349/update-stratux-v0.8r2-ec7d793dd1.sh.zip 
It presents the raw data we want, although it doesn't organize targets as to who's the worst case threat. Although it gives the raw info, I doubt it that it would be practical to fly with it the way it is now. Skypuppy might be right - perhaps you could code up a Special Traffic Display page for the web UI for the Stratux with a worst case threat algorithm?

skypuppy, 2016-03-23:
Not that that is a bad thing.

Axtel4, 2016-03-25:
@cyoung,
As @224XS points out, using signal strength to determine target distance has a number of limitations.  To be effective to for Stratux to determine the distance to a target a number of parameters will need to be controlled to provide consistent results.  These include, but are not limited to; calibrating the received signal strength to a known source, control the type of antenna used with Stratux, and controlling the placement of Stratux antennas within the aircraft.  

tl;dr
Background:
It takes more “effort” to determine the distance to a target based on signal strength than using RADAR distance.  In simple terms a traditional RADAR system pings (paints) a target with a radio signal.  The RADAR system knowing when it sent the signal starts a timer and waits for the return echo.  When it receives the echo it determines the elapsed time for the round trip for the signal.  Using the RADAR distance formula of 12.36 microseconds (uS) per nautical mile for a round trip the RADAR system calculates the distance to the target.  For example, the elapsed time to ping a target and receive the echo is 61.8 microseconds.  Dividing 61.8 uS by 12.36 uS per / nm round trip we find the target is 5 nm away.

Active Traffic Systems:
The Secondary Surveillance RADAR (SSR) and Active traffic systems utilize the same principle to determine the distance to a target.  When an active system (SSR, TCAS or TAS) interrogates a target it times the interval from the start of the interrogation to reception of the the reply.  Through the gift of defined specs, the turnaround time for the interrogation / reply cycle through the transponder is defined.

Pathloss Formula:
The distance to a target can also be determined by using the Free Space Path Loss (FPSL) formula.  The Path Loss formula states the free space path loss between two points is:

FPSL (dB) = 20Log(d)+20Log(f)+K
Where:
d = distance
f = frequency
K = constant that depends on units used for d and f

If d is measured in kilometers and f in MHz, the formula becomes:
FSPL (dB) = 20Log(d) + 20Log(f)+32.45
or,
If d is measured in nautical miles and f in MHz, the formula becomes:
FSPL (dB) = 20Log(d) + 20Log(f)+37.8

If the path loss is known, the distance between the two points becomes:
Distance (km) = 10^((FSPL-20Log(f)-32.45)/20)
or,
Distance (statue miles) = 10^((FSPL-20Log(f)-37.8)/20)

Now that we know we can determine the distance to a target via Path Loss, what will it take to accomplish this with Stratux?

How far away is that target?
To determine the distance to a target with path loss we need to know the path loss and to determine that we need to determine a few items about our system:

What is the Effective Radiated Power (ERP) of the subject’s transponder?
What are the Stratux receiver characteristics (i.e. sensitivity)?
What is the frequency of interest?

Frequency of Interest:
The frequency of interest is stratight forward as it will be either 978 MHz or 1090 MHz.  It is used as one of the variables in the Path Loss formula.

Transponder ERP:
This value is a know entity and are defined in the transponder TSO and DO (MOPS) documents.  The ERP will vary by the class and type of transponder (i.e. GA transponder v. Air Transport transponder).  The ERP takes into account the output of the transponder at the antenna jack, the loss of the coax cable between the transponder and the antenna, and the gain (or loss) of the transponder antenna.  The ERP can be expressed in Watts or dBm, or dBW.  Knowing the ERP of the transponder is the first step in determining the path loss.

Receiver sensitivity:
The second item needed to determine the path loss is the characteristic of the Stratux receiver and in particular the detected received signal strength for a given input stimulus.  In a typical transponder receiver the detector will have a voltage level for a given RF input.  For example, the receiver will be calibrated to have a detected level of 3 volts at a -60 dBm RF input level.

Example target:
Our target 1090 MHz transponder has an ERP of +52 dBm (158 Watts).  The received signal level at our receiver is -60 dBm.  The path loss from the target transponder is +52 dBm - (-60 dbm) = 112 dB.

Given the path loss formula the distance the target is:
Distance (nautical miles) = 10^((FSPL - 20Log(f) - 37.8) / 20)
Distance = 10^((112 - 20Log(1090 MHz) - 37.8) / 20)
Distance = 10^((112 - 20*(3.037) - 37.8) / 20)
Distance = 10^((112 - 60.75 - 37.8) / 20)
Distance = 10^((13.45) / 20)
Distance = 10^(0.7325)
Distance = 4.7 nautical miles

jpoirier, 2016-03-25:
Never seen the nautical mile constant used before
d = 10 * (112 - 60.75 - 32.45) / 20 
d = 10 * 18.8 / 20 
9.4 km = 5.8 sm = 5.04 nm

Ergonomicmike, 2016-03-25:
'Course, as Axtel4 said, we have no idea how many volts an SDR produces in response to so many dBm. And, with all the different antennas everyone uses, that's another variable that affects overall sensitivity.

If Stratux were going to implement this feature, would Stratux send distance data or signal strength data to an EFB? If the former, then perhaps there could be a calibrate function, where a user taps a button when an aircraft is at one mile?

cyoung, 2016-03-25:
So my ADS-B In antenna is on the belly of the aircraft at a fixed distance from a Mode C transponder anenna. Can't I use the output from the transponder antenna and known distance to establish the readings from the SDR? Then what further unknowns remain?

Ergonomicmike, 2016-03-25:
In theory, yes. But, given the nature of logarithmic signal strength - ignoring that you'll probably be in saturation in the SDR - I doubt that you could calibrate to anything meaningful with a signal that close/strong. (BTW, shows the need for either the Stratux or the EFB to not display a really strong signal as a target - because it's probably you.)

peepsnet, 2016-03-25:
Since we have many targets at known distances(all the other ada-b traffic that does report position) could.you not continually refine/calibrate the measurements?

Ergonomicmike, 2016-03-25:
That's a clever idea. Actually, you would just need to find one that's, say, 3 miles away, note the signal strength, and use that as a baseline to trigger alerts. Could update that value every now and then whenever there's another hit at 3 miles.

224XS, 2016-03-25:
If the antenna is not outside then the signal strength would be wildly different depending on the target position angle relative to the nose of the your aircraft.
Does the signal arrive through plexiglas or through aluminum skin, aluminized Dacron or perhaps carbon fiber?
Not everyone is Montgomery Scott who casually breaks the laws of physics everyday as a pass-time.

Ergonomicmike, 2016-03-25:
It's true that even with the antenna for the Monroy mounted on the side of our front windshield of our fiberglass plane (which is kind of a best case aircraft), I notice that it will not pick up aircraft who are blocked by the engine or the metal prop. (That would be true even if the antenna were mounted externally.) And it has a hard time seeing metal aircraft below us, whose transponder antennas are on their bellies, shielded from us. Still, like my eyeballs, it sees some aircraft. And seeing some is better than seeing none.

peepsnet, 2016-03-25:
> If the antenna is not outside then the signal strength would be wildly different depending on the target position angle relative to the nose of the your aircraft.

You could create a spherical map based on the relative altitude difference, azimuth, distance of position/altitude reporting targets.

224XS, 2016-03-26:
That is certainly possible, given enough 978 traffic (which doe not yet exist).
How are you going to be sure the Stratux antennae are in the same position day after day, glue them to the glare shield?

It seems clear that there are some very enthusiastic support for this feature. I just wish it was possible to make it work
to a degree of reliability commensurate with the effort involved.

Axtel4, 2016-03-26:
This is an analysis of a 1090ES target that explores using Path Loss to determine distance.  

[1090 Target Path Loss Analysis.pdf](https://github.com/cyoung/stratux/files/190333/1090.Target.Path.Loss.Analysis.pdf)

peepsnet, 2016-03-26:
I feel the feature is not accurate enough to be useful. I was just offering a suggestion... No worries though!!

cubaneight, 2016-03-26:
Hi all,

I think it is not most important to show such traffic with the exact distance from the ownship. Being a CFII and DE, my experience is that spotting a traffic target visually is much easier when you know that there is traffic at all. The range itself is not that important (albeit useful). What would be really nice is a direction information of such traffic, but that seems not possible to me, as long as we do not talk about multiple antenna installations.

What about the following idea (workaraound), based on data we have already available:

Since most EFBs will not support targets without a known position, why not display a non-position target as follows:

[image]

That means: 
- take the ownship-positon.
- display the non-position-target 4times: eg 2nm north, south, east and west of the ownship.
- that would be the clue for the user that it is one tsrget without exact postion, but with the displayed altitude difference.

I know it might be a little bit rough. But it might also be a lot of help. What do you think about that? It should be an optional setting, of course.

Regards,
Dieter

peepsnet, 2016-03-26:
@cubaneight 
The issue is still how far the traffic is!  We do not have a "valid" way to calculate it!

skypuppy, 2016-03-27:
I wonder if the variance produced by antenna location is more or less 
than the widely varying output of transponders, now and old, and older.  :)

Are we at the point  of diminishing returns when there are *so* many 
variables?

But, hey, as a fun exercise, I'm all behind that!

Skypuppy

Axtel4, 2016-03-27:
I think it leans more toward antenna location.  Per the transponder specs the delta between the lowest Mode A/C transponder power to the current Mode S transponder power is 8 dB.  Antenna location can exceed 20 dB in signal variation.

cubaneight, 2016-03-27:
This might be a reason just to use the signal strenght as an indicator to find out which target is the closest, and which is next-closest (without the necessity to calculate its exact distance).

If the efb could warn us of the, let's say three, closest targets, depending on the signal strenght, (and shows their relative altitude), that would be very helpful. 

In the efb, the closest target (highest signal strenght) could (by definition) be displayed as eg 2 miles away, the next 4 miles and the next 6 miles. (I know, those are not exact distances, but it would still give a good indication).

Regards,
Dieter

224XS, 2016-03-27:
Dieter,
Several people have pointed out that a single whip antenna inside an airplane made of conventional material can not separate signal strength change due to distance from those due to changes in absorption varying with angle.

The simple minded dIstance = 1/distance is likely to be wrong most of the time. Copious bad data is not better than less, but accurate trustworthy data.

peepsnet, 2016-03-27:
Its not as easy as saying the loudest signal is the closest. It is still not possible to tell distance just by a lone signal. It would take calibrated and tested equiptment with no change to the installation once calibrated. 

One could use the position of known traffic to build a bubble of reception around the aircraft. This would take time to developed as you fly. Say 100 aircraft reporting all around your plane over time. So initially there would be no way to give valid position based on signal strength for some time into the flight. 

Then you would need to repeat the process for every flight. So its not a good way to do it.

224XS, 2016-03-27:
Of Course that was distance = 1/db^2

damn autocorrect!

> On Mar 27, 2016, at 10:36 AM, Ira Rampil <ira.rampil@gmail.com> wrote:
> 
> Dieter,
> Several people have pointed out that a single whip antenna inside an airplane made of conventional material can not separate signal strength change due to distance from those due to changes in absorption varying with angle.
> 
> The simple minded dIstance = 1/distance is likely to be wrong most of the time. Copious bad data is not better than less, but accurate trustworthy data.
> 
> Sent from my iPhone

peepsnet, 2016-03-27:
Not to mention one mile away could be 5280 fret above you...

skypuppy, 2016-03-27:
The stratux, in that vein of thought, could record those and compare 
based only on those aircraft who broadcast their gps position and 
altitudes.  It *could* learn over time and build internal statistics 
tables.  The user would never know it's happening.
However, you still couldn't trust your life to it, but as an INDICATOR 
only, it might have some merit.

Skypuppy

Ergonomicmike, 2016-03-28:
Although it's true that traffic one mile away might be 5280 feet above, the signal strength will probably show worse than a mile away due to antenna patterns. 

A vertical antenna has a doughnut shaped pattern, that drops to zero above and below. Since both the transmitting and receiving have this pattern, there will be a doubling of attenuation when an aircraft is overhead. (Presumes S&L flight.)

FWIW, I don't think the Zaon did any geometry correction for signal strength. IIRC, it had a max altitude filter of two or three thousand feet. So even it it picked up a target 5000 ft above, it didn't display it.

cubaneight, 2016-05-15:
Hi all!

I'm trying to implement a simple traffic map now, based on a html 2D-canvas.
For testing purposes, I try to modify the traffic.html page and add a simple basic html-camvas. 

Does anybody know why I alwas just see the border of the canvas (with its given style, eg a grey css-border), but never any content in the canvas.

Even the simplest
```
<canvas id="aqq" width="400" height="100" style="border:1px solid #d3d3d3;">
Your browser does not support the HTML5 canvas tag.</canvas>

<script>
var c = document.getElementById("aqq");
var ctx = c.getContext("2d");
ctx.font = "30px Arial";
ctx.fillText("Hello World",10,50);
</script>
```

doesn't work. Is there any canvas definition in the underlying .js pages that might interfere?

Thank you!

Regards
Dieter

skypuppy, 2016-05-15:
This is pulling way, way back, but in Mode C, doesn't the transponder send altitude data, normalized to STP above 18,000 and actual below that?  We could grab and interpret the xponder outputs of other planes.  Mode S (airlines and some GA) has even better data?

rhole, 2016-05-17:
Mode C transmits pressure altitude at all times. Ground station converts that to true altitude. So you would get the "29.92" altitude only and have to compute from there

Rick Hole

skypuppy, 2016-05-17:
If we can interpret that with stratux, that would give us a (geometric) plane so we'd 1) see non-ADS-B aircraft and know whether they're above, below, or same altitude as us, and 2)  let us know that someone is there.  Now, if they're like me and can't afford an xponder then we still have to rely on the Mark 1 eyeball.

Axtel4, 2016-05-17:
> If we can interpret that with stratux, that would give us a (geometric) plane so we'd 1) see non-ADS-B aircraft and know whether they're above, below, or same altitude as us, and 2) let us know that [...]

But, wouldn't that also be a source of distraction?  Consider if this is implemented as suggested and Stratux identifies a target at +100 ft, but with no range or bearing information.  That target is 5 miles in trail and not a threat.   How much time would be devoted to searching for a non-threat target instead of flying the aircraft?  Also, how would that target be identified and displayed to the flight crew so they would have a reasonable idea on where to search for the target?

skypuppy, 2016-05-17:
Being able to answer that is way above my abilities.  Been thinking on it, too.  And it's hard to know whether knowing one or more are out there is beneficial or not.  If ADS-B Out is fully implemented, it is a non-problem except in uncontrolled airspaces (which is by far the most airspace in the US.)  If only the <censored> FAA would give the info it has in real-time to us from their existing ADS-B towers...

cyoung, 2016-12-01:
@cubaneight - the traffic websocket (ws://192.168.10.1/traffic) now reports signal strength which you might be able to use. If you come up with a clever way to use it or some sort of display, it would be great to hear back on it. Closing this for now, but feel free to respond here in the future.

closed 2016-12-01 by cyoung

dns-007, 2018-03-04:
Hello. 
I have the same problems as Dieter, flying in Germany. We have no TIS-B and not a lot small Aircrafts are sending their GPS position. Inspired by this Thread and some discussions in a microglider forum this Webradar prototype was developed : https://github.com/dns-007/stratux-webradar
It is using GPS and Altitude and Signal strength to alert you. It is not using an EFB, but is a standalone Webpage. Making the changes also to the EFB is a challange to the App developers and will not happen that quick i asume.
 The Signal strength is not used as an absolute value, but the change of the signal strangth indigates, if a target is approaching. 
Comments and Questions on this are welcome!