cyoung/stratux #353: Stratux compatability is listed as a feature on the first page of the iFly GPS website

Nokomis449, 2016-03-25:
It may have been there a while; I've just now noticed it.

[image]

cyoung, 2016-03-26:
Nice!

Not an issue, though :)

closed 2016-03-26 by cyoung

Nokomis449, 2016-03-26:
Yeah, I'm new to github and didn't know how to make a comment here otherwise.  IS there a way?  Or should I just have put it on reddit?  My intent was to give kudo's to the Stratux developers on their expanding visibility.

peepsnet, 2016-03-26:
Reddit

skypuppy, 2016-03-27:
Good idea, peepsnet.

With that in mind, I've not seen it written down anywhere but this seems 
to be the way we're doing things:
1. reddit is used for bug reports, general comments, both yay and nay, 
regarding all things stratux, while
2. in here is the place for bug reports and their discussions, and all 
things needed by the software developers and contributors but not the 
general user.

Or am I way off base here?

Thanks,
Skypuppy

peepsnet, 2016-03-27:
Reddit is used for ppl who are not technical or aware or github. It seems to be the place where more everybody can measly get to and understand. 

Reddit is where a non-tech person calls attention to an issue a d then either they find their way here or one of the tech minded guys gets it here.