cyoung/stratux #354: ForeFlight text update count vs. actual received is too low

cyoung, 2016-03-26:
1. Stratux version: v0.8r2 / 8de91a55f0
2. Stratux config:

    SDR
      [X] single
      [ ] dual

    GPS
      [ ] yes
      [X] no
      type:

    AHRS
      [ ] yes
      [X] no

    power source: ship power (via UBEC).

    usb cable: none. direct wired to GPIO.

3. EFB app and version: FF v7.6 (1991)

    EFB platform: iOS 9.2.1

    EFB hardware: iPad Mini 2

4. Description of your issue:

UAT log attached. FF was open 75% of the time. UAT log shows 13,535 METAR updates. The devices page shows 1,163 for "Text Report Count". The average text reports per frame containing a text report in this UAT log is about 5, so even if this count were "frames with text reports" it is off by a factor of two.

Need to investigate if the "Text Report Count" is a meaningful metric, and if so investigate if FF is receiving all of the data that is being sent to it.

[image]

[0117-uat.log.gz](https://github.com/cyoung/stratux/files/190614/0117-uat.log.gz)

bsdcat, 2016-03-28:
"Text Report Count" is a count of unique, current text updates (METARs and TAFs). That log has 13,535 METARs and 2,054 TAFs but quite a few duplicates. I picked one example for comparison that was duplicated 23 times over the course of the log - that's a report with identical location (KMKG), time (261655Z), and text (VRB03KT 10SM FEW030 08/M02 A3021 RMK AO2 SLP238 T00831017).

cyoung, 2016-03-28:
Here are the counts for METAR/TAF/SPECI de-duped by field, PIREP de-duped by timestamp, and WINDS de-duped by location+timestamp. Looks like it checks.

Thanks @bsdcat, this saves us some time investigating.

```
$ zcat ~/Desktop/temp/0117-uat.log.gz | cut -d, -f2 | grep -v '^$' | ./extract_metar  | grep 'METAR' | cut -d' ' -f2 | sort | uniq | wc -l
     685
$ zcat ~/Desktop/temp/0117-uat.log.gz | cut -d, -f2 | grep -v '^$' | ./extract_metar  | grep 'TAF' | cut -d' ' -f2 | sort | uniq | wc -l
     193
$ zcat ~/Desktop/temp/0117-uat.log.gz | cut -d, -f2 | grep -v '^$' | ./extract_metar  | grep 'SPECI' | cut -d' ' -f2 | sort | uniq | wc -l
      18
$ zcat ~/Desktop/temp/0117-uat.log.gz | cut -d, -f2 | grep -v '^$' | ./extract_metar  | grep 'PIREP' | cut -d' ' -f3  | sort | uniq | wc -l
      33
$ zcat ~/Desktop/temp/0117-uat.log.gz | cut -d, -f2 | grep -v '^$' | ./extract_metar  | grep 'WINDS' | cut -d' ' -f2,3  | sort | uniq | wc -l
     285
```
285+33+18+193+685 = 1,214

closed 2016-03-28 by cyoung