cyoung/stratux #355: Bug in stratux or iFly?

skypuppy, 2016-03-27:
1. Stratux version:
2. Stratux config:

    SDR
      [ ] single
      [x ] dual

    GPS
      [ x] yes
      [ ] no
      type:

    AHRS
      [ x] yes
      [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)  IFlyGPS 9.4

    EFB platform: (e.g., iOS 9.2)  Android 5.2

    EFB hardware: (e.g., iPad Mini 2)  Galaxy Tab S 10.2

4. Description of your issue:
iFly randomly drop the AHRS usage.  The AHRS shows up fine in the stratux webui, showing orientation and indicating it's on and being used.  However, Sometimes in iFly, iFly sees and uses the AHRS and sometimes it doesn't -- during the same EFB session.  Now, it is more off than on.  All the other ADS-B functions continue to work so I know comms are there.  It's just the AHRS that is flaky in iFly.  Ideas?

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

closed 2016-03-27 by skypuppy

cyoung, 2016-03-27:
What was the fix/finding?

skypuppy, 2016-03-27:
Somehow, there appeared a switch for AHRS output on the webui.  It was 
probably AvSquirrel doing a workaround against the <censored> 
ForeFlight.  Very, very strange that it would stop after running for a 
couple hours, with no changes.