cyoung/stratux #356: Backup or guard radio

welbo97, 2016-03-27:
1. Stratux version:0.8r2
2. Stratux config:

    SDR
      [ ] single
      [x ] dual

    GPS
     X [ ] yes
      [ ] no
      type:vk-172

    AHRS
      [ ] yes
      [ x] no

    power source:Canakit 2.5A 

    usb cable:n/a

3. EFB app and version: (e.g., WingX Pro7 8.6.2)FF 7.6

    EFB platform: (e.g., iOS 9.2) iOS 9.2.1

    EFB hardware: (e.g., iPad Mini 2) iPad Air 1

4. Description of your issue:

These receivers are so useful. Here are a few ideas for additional features. Some pretty far fetched.
1. I would like to be able to turn off one of them and set a VHF air band frequency to monitor. For example, in the event of a comm radio failure, be able to set and listen to the controller frequency. Or set and continuously monitor 121.5 as suggested in the AIM. Either with a third receiver or by sacrificing one of the traffic receivers. Send it out of the pi audio jack, or even through the Bluetooth link.
2. Show me what my transponder is transmitting. Code (squawk) and altitude. Any Mode S data too.
3. Audio server with stored music, or RSS feeds previously downloaded?
4. I set up a pi to act as a foreflight proxy so multiple tablets can download charts from it without having to go back to the Internet. How about using Stratux as a master chart server for those with low tablet memory, or for guests on the plane who want to download a chart in the air. 
5. I even saw some of the Ham guys showing how to use the rtlsdr to receive reflected radio signals, calculate the Doppler shift and have a makeshift radar receiver. How about a passive onboard radar receiver looking at reflected primary ATC radar! 

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

skypuppy, 2016-03-27:
I recommend using another Pi for those kinds of functions.  They're 
cheap.  :)  A Pi3 has the cpu room, certainly, but there is that feature 
creep gremlin...  I also have ideas aas to how to use them, like glide 
slope visualization, and so on.  I'd use a different system apart from 
the base stratux, though.  Just my thoughts.
Another possibility is to use a 3rd antenna IF you have the USB ports 
available on the same stratux box.

224XS, 2016-03-28:
Build up another RaspPi for a VHF receiver. Receiving clear, unbroken AM in the airband is a full time SDR job (about 30-70% time for CPU). There are several open source packages to do that. You can also do VOR, localizer and glideslope, if you are ambitious. 

Build another Rasp Pi for OSMC so you can play music, or run Music on your iPad while running ForeFlight

How often would anyone use a squitter sniffer since most modern transponders display squawk and mode C.

Are you using #4 (the chart proxy server) to circumvent the FF subscription limitation? All my charts download at home over a fiber link. Talk about mission creep!

welbo97, 2016-04-20:
The FF chart cache is a project I worked with some of the FF people on at their request. It does not circumvent your subscription so it would not be of much use in an unconnected environment unless you did. My flight department found it useful while doing updates at our hangar with only LTE connectivity for 6 or more ipads at a time. I set it up at my house just to save the internet usage since we are limited now (and to tinker with). The downloads come down at the limit of the 802.11 link after the first unit does it's update. Pretty nice. It would work for any software that uses standard http requests for updates (not https). 

I don't really like the thought of having a bunch of pi's scattered about the airplane. The pi 3 looks like it is able to handle more than just stratux so maximizing its usage seems logical. A backup receiver would be a very desirable feature over and above any other products out there and fairly easy to implement (I've done it from the command line while in flight). Think about how many more units you will sell with this feature!

224XS, 2016-04-20:
Depending on mission, I have loaded between 5 - 8 gig of chart data on my iPad or phone.  Its a considerable amount of data for a constrained RPi environment.

Also, who did you have in mind with your last comment about selling more units?
This is an open source project.

closed 2019-01-17 by Helno