cyoung/stratux #358: Utilize MsgLog to populate weather for previously-unconnected clients.

cyoung, 2016-04-01:
All FIS-B messages are saved in `MsgLog` for a maximum of 1 minute.

Would it be worthwhile to:

1. Parse incoming weather reports. De-dupe METARs and such by field name so that we're not using a lot of memory storing duplicate/outdated data. This would also be useful for the network queues.
2. Save radar frames for a "long time" (1 hour?) since they are sent so infrequently.
3. Populate each new DHCP lease/client that has connected with the above data.

This boils down to the question: how many people use a device that you connect for the first time while in the air? Maybe it's a good feature if you really want to have a fail-safe device to view data like a smart phone that you rarely connect to stratux.

peepsnet, 2016-04-01:
Yes!!  I hate loosing all that data.

Would it keep after a reset/restart/power cycle??  If there is an issue and I need to reboot the statux then it would be nice to "inject" all the data. 

Would it be feasible to only purge the data for an hour or until new data is received?

This means I could save battery on the tablets by letting the statux run and gather data like a ADS-B buffer and deliver it when I connect. 

I would like metars, notams, tafs, and maybe pireps or other weather info.  Basically anything non-aircraft from the towers...

Ergonomicmike, 2016-04-01:
AFAIK, iFly stores wx info locally. (And reports how old wx info is.) I have no problem with the Stratux storing it too as long as it doesn't cause a problem if Wi-Fi is lost on the tablet and reconnected in flight.

cyoung, 2016-04-01:
>This means I could save battery on the tablets by letting the statux run and gather data like a ADS-B buffer and deliver it when I connect.

It does this already. Only difference from the above is that you must first connect to stratux to make it aware of your device for it to start buffering data for that device.

peepsnet, 2016-04-24:
Picking up from [Reddit](https://www.reddit.com/r/stratux/comments/4g8um9/does_stratux_cache_weather/)

So if I connect my EFB to the Stratux and then disconnect it(reboot) and connect again I will get all the FIS-B data since the disconnect??

How long will Stratux "backfill" FIS-B Data??

cyoung, 2016-04-24:
>So if I connect my EFB to the Stratux and then disconnect it(reboot tablet) and connect again I will get all the FIS-B data since the disconnect??

Yep.

>How long will Stratux "backfill" FIS-B Data??

It just creates a buffer when it detects that your EFB is out of context or the tablet is asleep. It saves up to 25,000 FIS-B messages in memory and will replay them when your tablet is active again.