cyoung/stratux #362: Change web root.

peepsnet, 2016-04-02:
Since Apache on Jessie uses /var/www/html as a root
and /var/www/ is the root for all other web servers

Maybe we should look at another location to keep the files clean and unaffected by other systems/apps

Choices:
/var/stratuxweb/
/var/stratux/www/  (my choice)
/var/www/stratux/

skypuppy, 2016-04-02:
If we wanted to stay somewhat standard, /var/www/stratux would work, too.

peepsnet, 2016-04-02:
>  /var/www/stratux would work, too.

But this would create a folder in those with apache with /var/www as their root

/var/stratux/www
This allows for us to not interfere with another apps that a user would install...

cyoung, 2016-04-12:
...but we're not using Apache?

peepsnet, 2016-04-12:
I thought we were using the GO built in browser?

cyoung, 2016-04-12:
Webserver, yep. Doesn't matter to me the web directory, what would be a good reason to change it?

peepsnet, 2016-04-12:
to protect the files we use from being overwritten. to prevent someone having an issue after tinkering around with their pi and installing apache2 and some web app and it writes a bunch of files to the /var/www/ folder.

Just a way to separate stratux files from anything messing with them because we do not "lock down" the FS...

cyoung, 2016-04-27:
Will likely change it in the future as the need arises, other applications installed on Stratux devices, etc. Not a priority now.

closed 2016-04-27 by cyoung

pasniak, 2016-06-06:
Hi, just installed Foxlor (a web admin toolkit for my wife to schedule some automation stuff) and it runs on apache2; knowing Stratux listens on 80 I tried to redirect to 8080 then 8080 but to no avail; I only get Stratux web gui. Why is that?

pasniak, 2016-06-06:
Never mind, I figured it out.