cyoung/stratux #364: Audible Traffic Alerts?

glwilkins, 2016-04-03:
Will we ever see audible traffic alerts in any new release?

peepsnet, 2016-04-03:
That is what the EFB does...

ScraboTower, 2016-04-03:
Not all of them

skypuppy, 2016-04-03:
iFlyGPS does.  You can turn the sound and/or visual alarms on or off.

JohnOCFII, 2016-04-03:
> Will we ever see audible traffic alerts in any new release?

From my understanding of what the Web interface design goals for Stratux are -- probably not.  The Web interface is mainly for setup and confirmation of functionality of the Stratux components.  Features such as audible traffic alerts are better suited for one of the EFB targets, such as iFly, WingX, ForeFlight, etc.

glwilkins, 2016-04-03:
I haven't had my stratux in the air yet to test this out

Will it work via Bluetooth to for flight?
Does it use voice comands like:
"Traffic at 10 o'clock low"?

Thanks for all the comments.

cyoung, 2016-04-03:
Stratux itself doesn't output any audio. What app are you using it with? ForeFlight does give traffic alerts when it sees a target (I think) within 2nm.

AMNegron, 2016-04-04:
FltPlan Go also has that functionality, but I have not tested it

peepsnet, 2016-04-04:
lol... WORKS!!

cyoung, 2016-04-12:
@glwilkins - the app you're using with stratux likely already does this.

closed 2016-04-12 by cyoung