cyoung/stratux #371: v0.8r2 suddenly failing to connect

Ergonomicmike, 2016-04-06:
1. Stratux version: v 0.8r2 + AvSquirrel special sh 3dd1
2. Stratux config:

    SDR
      [ ] single
      [X ] dual

    GPS
      [X ] yes
      [ ] no
      type: BU-353

    AHRS
      [ ] yes
      [X] no

    power source: Anker E5

    usb cable: Anker cable

3. EFB app and version: (iFly v5.9.26b)

    EFB platform: (Android KK)

    EFB hardware: (two Android tablets)

4. Description of your issue:

Been flying with v0.8r2 for a couple weeks now. About ten trouble free hours. And then, suddenly, Saturday, on a return flight home (where it had been working trouble free on the way out), it suddenly began failing. At first it wouldn't send GPS info to iFly, although traffic info was coming thru.

Yesterday (Tuesday) it would connect and then disconnect after about five minutes. I would have to do power down resets in flight. Eventually, it wouldn't connect at all. Strong signal on Ch 1. But no heartbeat and not able to connect to the web UI. Couldn't see messages on Avare's I/O Plug-in. 

I've since reimaged the SD card and Stratux is working again. (On the ground. Haven't flown with it yet.)

Here's a link to a truncated stratux.log file, where I only included yesterday's troublesome flight.

[stratux.v0.8r2.avsquirrel.3dd1.log.txt](https://github.com/cyoung/stratux/files/207229/stratux.v0.8r2.avsquirrel.3dd1.log.txt)

cyoung, 2016-04-06:
Did it give you the "No Internet" warning?

peepsnet, 2016-04-06:
are you using a version that is using sqlite?

Ergonomicmike, 2016-04-06:
@cyoung No, to the question about a "no Internet" warning. (I forgot that I had added that one line to the .conf file on the old image to fake out devices. I had reportred that freeze to iFly and IIRC, they fixed a bug on their end.)

@peepsnet No, to the question about sqlite. (I saw your issue reporting a disconnect. I've been hanging back on the betas because v0.8r2 has been working so well and the latest changes seem to have more to do with logging than performance.)

cyoung, 2016-04-12:
I did have some overheating issues with the Pi placed in the side pocket on my plane on the way down to SnF last week.

Note that your CPU temps don't stabilize completely and the peaks were near the end of the first 34 minute run. Not sure if related.

Fixing the -99.00ºC CPU temp issue now.

Ergonomicmike, 2016-04-12:
I flew two 2-hour legs yesterday. (Coolish OAT's, FWIW, although I've had the Stratux running in the hot AZ sun at home with no problems.) Stayed connected for the flights - ONCE I was able to connect. I think that yesterday's problem is an iFly issue, since they've been changing stuff with how they parse FIS-B. I was able to connect in the sense that I got GPS and I got 1090. But weather wasn't coming thru. It took rebooting the Stratux in flight to get iFly to play nice. The difference between the Stratux for yesterday's flight and the previous disconnecting flights was a re-image of the SD card and Logging disabled. (I dunno - maybe the partition was full last time from logging running for two weeks?)

Ergonomicmike, 2016-04-18:
Closing this. Flew many hours recently with 4964.sh and iFly v9.6.4b. No problems.

closed 2016-04-18 by Ergonomicmike