cyoung/stratux #373: VK-172 GPS altitude report sign error

welbo97, 2016-04-07:
1. Stratux version:8r2
2. Stratux config:

    SDR
      [ ] single
      [ x] dual

    GPS
      [ x] yes
      [ ] no
      type:vk-172 usb

    AHRS
      [ ] yes
      [ x] no

    power source:RAV 16,200 mAh

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) FF 7.6

    EFB platform: (e.g., iOS 9.2) iOS 9.3.1

    EFB hardware: (e.g., iPad Mini 2) IPad Air 1

4. Description of your issue:

aircraft altitude fl450, GPS (freeflight) altitude reports -24,499'. Traffic at presumed fl400 reports 65,000' above. 
0026-GPS-log.gz link to gdrive: https://drive.google.com/open?id=0B85PHVGbC7SQMHdHbUJHS3ZXWDQ

FF logging on as well. Let me know what other logs you may need.

AvSquirrel, 2016-04-07:
ForeFlight and Stratux have an overflow bug for geometric altitudes above 32767'.

#375 fixes it on our end; and is confirmed to give correct altitudes in Avare. However, correctly formatted GDL90 0x0B messages still cause the overflow in ForeFlight 7.6.1. Suggest contacting ForeFlight support (team@foreflight.com) to review and fix.

welbo97, 2016-04-07:
Are the NMEA or GDL 90 messages from the flight where the overflow happened cached somewhere I could get it and feed it back to FF to simulate the situation again? Or do you guys have a file prepared for such a purpose? It will be a while before I am able to take a plane above 35k again and I do not have a GPS simulator.

cyoung, 2016-04-12:
@bsdcat

bsdcat, 2016-04-12:
Thanks for the @ mention, I see the issue.

cyoung, 2016-04-12:
Excellent, we'll close it here now - if you need testing then post back.

closed 2016-04-12 by cyoung

bsdcat, 2016-07-14:
Addressed in ForeFlight Mobile 7.7.