cyoung/stratux #374: Interference in a Learjet

welbo97, 2016-04-07:
1. Stratux version:8r2
2. Stratux config:

    SDR
      [ ] single
      [x ] dual nano's

    GPS
      X[ ] yes
      [ ] no
      type:vk-172

    AHRS
      [ ] yes
      [x ] not yet

    power source:RAV 16,400

    usb cable:

    Makexyz printed helno case with dual bulkhead mcx-sma connectors and rubber single pole antennas.

3. EFB app and version: (e.g., WingX Pro7 8.6.2)FF 7.6

    EFB platform: (e.g., iOS 9.2)iOS 9.3.1

    EFB hardware: (e.g., iPad Mini 2)iPad Air 1

4. Description of your issue:

I have had issues with multiple receivers in the cockpit of the Lear-stratus 1, onboard iPad GPS, iridium sat phones, dual GPS receivers. Signal strength on all are very weak and even unusable in some cases. The Stratux GPS receiver and the 978 uat reception are very poor. Move the device to the rear of the plane near a window and it works much better. I attempted to use rtl_power and one of the sdr's to map spurious transmissions but failed due to my I experience. While not directly a Stratux issue, any suggestions would be appreciated. Stratux operated much better in a Cessna 340.

skypuppy, 2016-04-07:
External antennas?

welbo97, 2016-04-07:
I won't have the option to install anything on the aircraft. I have had some success with locating the box in the baggage compartment with the antennas suction cupped to the window. That has aesthetic problems and limits your rf view to half the sky and ground. 
I saw someone say shielding the box makes for better reception because of ground plane so I'll try that. What I would really like is some direction to using the sdr's to make a heat map while switching accessories on and off to try and isolate the noisy device in the airplane, thereby solving a lot of problems at once. I know that is out of the scope of this forum though. Point me to the right place to go and I'll quit bugging you(I'll quit this topic here either way). But I will report on the performance with the box shielded.

224XS, 2016-04-08:
I don't know why you think this is most likely an interference problem or a shielding issue. Most people are running Stratux just fine w/o shielding.  I think sky puppy is correct. You just have a big iron plane with lots of metal between you and the target/tower sources. Remember, all the 978 transmitters are below you and if the stratux antenna doesn't have a clear view of ground beneath you, you will be SOL

welbo97, 2016-04-08:
Several of my points indicate interference: 1. Problems with multiple devices operating in the ~1GHz range, 2. Problem improves the further away from the cockpit you get(where most of the electronic equipment is located), 3. GPS as well as ground stations are affected. 
I get your point but you know the plane is made of mostly non-ferrous material, right? J/k. 
I have been flying it for a few years and this particular plane has this problem, not other identical models with identical equipment. So it doesn't seem like much of a leap to believe interference is an issue. I'm trying not to jump to a premature conclusion but I've seen this issue before (lots) in the time I was an avionics repair tech. In fact, I would even wager it is a comm or the ELT since they are the usual culprit. We used to have a problem with the carrier generation tube in the old ELT's reflecting resonant frequencies in the GPS range from the comm radios.  You would see your GPS lose its lock every time you transmitted. A simple notch filter on the ELT fixed it every time. Also, for some reason, the face of the comm radio was a usual source of spurious interference. 
I think I could prove it too if I could figure out how to use the rtl_power app correctly. I tried looking at 977-979MHz and 1089-1091MHz but the results boiled down to an intermittent ~40db evenly across the whole scan range. Maybe I should broaden the range, make larger bins? I just don't know enough about how these sdr's and the rtl utilities work. Exacerbating the problem is the fact that it is a 2.5hr airplane that doesn't fly very often. With no internet in this plane, I have to have a better idea what I'm doing going into it.
Interesting side note- kal found a EGSM 900 station in one of its scans over CO at 45,000'. Very rare (if ever) to get a phone signal from that high.

skypuppy, 2016-04-08:
There are several utilities that one can use that makes a Pi with rtlsdr 
radios an EMF chaser.  Use it like a probe and move it around the 
cockpit looking for the highest readings.  Set it at 1 Ghz with a 2 Mhz 
bandwidth might be a place to start for moving around mode. GNURadio is 
a good candidate for the software, but there are others.  Just turn off 
stratux "service stratux stop"  and run the other program  In fact, some 
Android tablets can run an EMF scanner program but they don't work well 
in the frequencies we need.  Come to think of it, begin the scan with 
the stratux in the center of the cockpit and just look at where the freq 
spikes are first.  i would suggest trying both types of tests first with 
panel on, engine off, then with engine on, panel off, then both on.
HTH.

Ergonomicmike, 2016-04-08:
Whatever the problem in this instant case, I don't see how this is a "code" issue for thr github. (That is, how could a change in software fix this?) Perhaps reddit is the better forum for soliciting ideas for a solution?

224XS, 2016-04-08:
Excellent points! Nice to have some data.
Let me clarify some of mine.
Much of the TSO'd devices will be EMI/RFI shielded for certification though. Blocking RF is easily done with aluminum, that effect doesn't require any iron. In fact, you will find that the best Faraday cages are made with copper mesh, not ferrous.

Any digital device is going to generate RF hash, the question is just how many db outside the case.

The loss of gps lock you had is interesting, even bizarre, but not a fair comparison. Terrestrial level GPS signal strength is probably around 30 db less than what you might hope to receive on 978 UAT.

I hope you find an offending emitter though, because I wouldn't want to pay to stick an external UAT antenna on a Lear.

bradanlane, 2016-04-13:
The wifi is the only real "transmission" from Stratux. Is it possible to "turn down" the wifi power? This test would keep all of the wifi code running but allow some control over the power output.

Shielding will help with any oscillator harmonics generated from the RPi and SDRs.

When you move the Stratux to the back of the airplane, you are ostensibly doing both of the above.

_this is mostly just 'thinking outloud'_

Ergonomicmike, 2016-04-13:
There was an earlier "issue" here about how to lower the power on the Edimax. https://github.com/cyoung/stratux/issues/141 From what I read, it can't be done. The Edimax is either full on or full off. It can be put to sleep, but it cannot go to half power. 

Now, if a compatible Wi-Fi adapter (e.g. TP-Link) has provisions for reducing power, that might be interesting to consider from a power savings/interference viewpoint. And what about the RPi3's built-in Wi-Fi? Can it be throttled?

bradanlane, 2016-04-13:
@welbo97 - if you have a short USB "extension" cable, you could extend the WiFi dongle away from the Stratux and then wrap it in tinfoil. This should tell you if the interference is the WiFi or not.

It won't make Stratux very useful but it is a simple test.

krustyjonez, 2016-04-15:
What model Lear?  Heated windshields / side windows?

welbo97, 2016-04-20:
@krustyjonez Lear 31A, 2000 model mainly (I fly several and have only had the problem in this one). 
Heated windshields, but not turned on, no side windows. 

Since this one airplane is the only one that has issues, I do not believe the problem is in the stratux box or with the wifi signal. I believe it is actually something in the airplane creating the interference. To complicate the problem, it does not have the problem on the ground or low altitudes (<25K'). 

I do like having the wifi power turned down all the way on the stock stratus. Surely it at least saves some battery. No use blasting full wifi power while everyone interested is confined to a 15' radius.

Ergonomicmike, 2016-04-21:
I still don't see how this is a code issue, but the obvious question is: What system turns on/arms itself in a Lear 31A at or about 25,000 feet? Some kind of yaw / stall controller? Etc. Can you pull breakers in flight when the interference is present to isolate the system?

cyoung, 2016-06-13:
@welbo97 - Try `rtl_power -f 100M:1600M:1M -g 48 data.csv`, it will run until you kill it (ctrl+c). Open data.csv in Excel and see what it shows. If you're able to share, I'd be curious to take a look at it.

cyoung, 2016-12-01:
@welbo97  - there are a few good suggestions here, please comment back if you have further findings.

closed 2016-12-01 by cyoung

welbo97, 2016-12-01:
I am unfortunately not associated with that airplane any more. I have used the Stratux in the Hawker 800 series I have been flying (similar altitudes and speeds) with great success. 

Sent from my mobile device.