cyoung/stratux #378: Edimax driver doesn't load

admiral1701, 2016-04-08:
4. Description of your issue:  I downloaded and ran the latest build from official stratux webpage.  As far as I can tell, the driver or the configure file does not load the wifi adapter.  I have the edimax ew-7811un.  

Dump1090 seems to wiz right along, I just can't connect the iPad to the pi network.  I put the edimax in my laptop and it worked wonderfully.

I'm a computer guy just not a Linux guy.  This has been a steep learning curve.

Any help is appreciated.

peepsnet, 2016-04-08:
So what type pi? And all that other stuff...

admiral1701, 2016-04-09:
Pi2 b+

admiral1701, 2016-04-09:
Dude.  Thanks again.. Read the response on the Reddit forum

cyoung, 2016-04-12:
Ah, the Pi2B+ again...

closed 2016-04-12 by cyoung