cyoung/stratux #389: continuous reboots from webui

skypuppy, 2016-04-13:
1. Stratux version:  0.8r2  acee114964
2. Stratux config:

    SDR
      [ ] single
      [ x] dual

    GPS
      [ x] yes
      [ ] no
      type: mtk3339

    AHRS
      [x ] yes
      [ ] no

    power source:  120vac wall wart

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) na

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:
Using a Winblows 7 laptop and Firefox to watch the webui.   Wanted to reboot, so hit the reboot switch on the webui and the RPi3 (and -2) continuously reboots until I shut down the laptop browser page.
severity: critical
Does anyone else get this result?  It happened before with the RPi2 as well.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

bradanlane, 2016-04-13:
when you click "reboot" did you get the "are you sure" message?
once you say you are sure, did you see the text, "Stratux is rebooting. You may need to reconnect WiFi one it reboots." ?

skypuppy, 2016-04-13:
bradanlane, I really don't remember.  The text you list sounds like just 
a reminder and not a checklist item.  Either way, it is weird that the 
Pi would continuously reboot based upon a message sent more than boot 
ago.  Must be something in the boot shell script that does not get 
over-written or replaced at the next boot.  That's just a guess as I 
have not seen that particular bit of source code.

(note, if the ... shows up, it is because bradanlane's post did not show 
up on github by the time I replied.)

Skypuppy

bradanlane, 2016-04-13:
I've been able to reproduce it. the problem is occurring from the browser. since the browser has not received a reply, it keeps trying the last URL. Since that URL is "http://ip_address/reboot" it reboots over and over. I even powered off my Stratux but left my browser open and as soon as Stratux was up and running, it rebooted.

skypuppy, 2016-04-13:
Can we add a reply for boot request back to the webui browser page?

bradanlane, 2016-04-13:
I'm not that fluent with the oddities of the GO webserver. I'm working on a test to see how to clean up the active browser page. Ideally, I can make the browser go to the "status" page and wait for Stratux to finish rebooting.

skypuppy, 2016-04-13:
An alternative is to remove the "reboot" option from the webui, but that 
would interfere with other options, like the need to reboot in flight 
until we get the other problems all sorted out.  On the ground, we can 
reboot manually until a) the go web is fixed, or b) we drop the go web, 
or c) we get all the other workarounds figured out.  :)

Still, one would expect an ack from the Pi to the webserver before 
reboot was allowed, simply as a matter of routine protocol.  Or maybe 
the webserver IS expecting an ack and not getting it from the Pi?

Still trying to fix things,
Skypuppy

bradanlane, 2016-04-13:
the most likely "better" solution is for the reboot and shutdown to be AJAX web service calls rather than HTTP POST operations.

braddward, 2016-04-13:
Maybe we need to have the reboot function launch another thread that will
sleep for a second or two and then do the reboot.  This would give time to
allow the response to get back to the browser.

Perhaps a simpler solution might be to add a .then method on the $http.post
function and force a flush - or send a config option with a very short
timeout - not sure if either of those would work.

If I could reproduce it that would make it easier to test solutions...

skypuppy, 2016-04-13:
If I set Wireshark to snoop it, that would not help if the Pi is not flushing the ack (if there even IS one) before the actual reboot occurs.

(And I wonder why these comments are not showing up on github?)  Replying via email leaves the previous text added.  Aside: before I knew the threads were also occurring on github, I was annoyed at some of the posts because there was no context.  :)  Now, I tend to reply only in github (if I can see the post.)
<sigh>  NOW the posts are showing up on github.

Skypuppy

braddward, 2016-04-13:
My guess is that the RPI 3 is just faster so the reboot is happening quicker.  @bradanlane is likely correct in that making this an AJAX call will probably resolve it.

Guess I need to order an Pi3!

Ergonomicmike, 2016-04-13:
We had this problem a while ago from the web UI
(https://www.reddit.com/r/stratux/comments/3ylhyl/video_of_possible_bootloop_problem). It got fixed and I haven't had a problem with reboot loops via my Android tablet. What changed?

bradanlane, 2016-04-13:
I do not know if anything changed or this is a manifestation of a different browser (I can reproduce with Chrome on my Linux Dev system).

bradanlane, 2016-04-13:
it looks like I have a fix.

following the "are you sure" message from a reboot, the WebUI now redirects to the status page and waits for the Stratus to complete its reboot process.

I'll create a branch, commit the code, and submit a PR.

closed 2016-04-13 by cyoung

rhole, 2016-04-13:
Might I suggest the use shut down his web client when this oddity occurs?

skypuppy, 2016-04-13:
It happens with -2's as well.

bradanlane, 2016-04-13:
Yes. I did my debugging, development, and verification on my RPi2.

skypuppy, 2016-04-13:
Yes, Rick, that is what I did to stop the rebooting.  However, a newbie, 
up in the air ahd having unidentified problems with his stratux, will 
resort to the reboot (we're all trained the Micro$loth way!) and then 
his stratux may not come back at all -- and he won't know what to do.  
In-flight failure is not going to be fun under any circumstances, much 
less when the stuff has hit the fan.  And then, of course, is when stuff 
happens to fragile computer gear and interconnects.  We *must* engineer 
as much reliability as we can into a flight instrument, certificated or not.

Skypuppy

commit c327c022 (2016-04-13): Merge pull request #391 from cyoung/reboot_repeat_fix

trigger reboot as a webservice and redirect webui to status page. Fixes #389.