cyoung/stratux #392: gps issue (maybe) and 978 issue (maybe)

skypuppy, 2016-04-15:
1. Stratux version: 0.8r2
2. Stratux config:

    SDR
      [ ] single
      [x ] dual

    GPS
      [x ] yes
      [ ] no
      type:  MTK3339 and USB Sirf III

    AHRS
      [x ] yes
      [ ] no

    power source: wallwart

    usb cable: unk

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue(s):

First, the 978 potential problem.  I used all combinations of RPi2, RPi3, R802T, R802T2, original antennas, latest dmurray high-gain antennas, nano2, original DVBT, and a couple others.  All config tests were run for multiple days at a time.  With 0.4r4 I would get 978 hits every day.  After 0.4r4, I get no 978 hits at all, zero, for several weeks now.  It is possible that the traffic has changed but to get them daily before and none now?  Something might be wrong.  I admit I've not been able to drive up to the ADS-B (cell) tower to perform the ultimate check but I'm still concerned.
Power issues are all good.

Secondly, not being able to upgrade to AvSquirrel's MTK3339 branch and keeping 0.8r2, after 3 days of running with no GPS results (it has eventually worked,) I simply plugged in the USB Sirf GPS without unplugging the serial-based MTK unit.  Stratux found the Sirf within minutes and was seeing GPS  sats.  Now, on the webui, the GPS/AHRS page shows the proper GPS info, but on the "status" page, it shows "gps disconnected."  More importantly, iFly is not getting GPS info from the stratux.  I don't know how to fix this.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

224XS, 2016-04-16:
I get zero 978 traffic on the ground 30 miles from KATL where there are three ADSB ground stations. Drive up I-75 next to KATL and from the top of hills, Stratux starts seeing 978 FIS & TIS about 4 miles out.
Over the top of my airport, I get three ADSB stations starting at about 800 AGL, range 25 - 30mi.

There's no way around it, if you want a real test of your rig, go vertical, or go for a drive. Hoping for random overflights is just unreliable.

skypuppy, 2016-04-16:
See update to 978 traffic in Reddit.  GPS problems persist.

peepsnet, 2016-04-18:
Which amp meter were you using.

skypuppy, 2016-04-18:
Sienoc Charger Doctor, 3.5-7V, 0-3A,  USB in and out.

closed 2016-04-19 by skypuppy