cyoung/stratux #393: RPi3: GPS cannot connect via UART

AvSquirrel, 2016-04-20:
1. Stratux version: 0.8r2 build da5f5d003d
2. Stratux config:

    SDR
      [X] single
      [ ] dual

    GPS
      [X] yes
      [ ] no
      type: MTK3339

    AHRS
      [ ] yes
      [x] no

4. Description of your issue:

On the Raspberry Pi 3, devices `ttyAMA0` and `ttyS0` are not the same device. `ttyAMA0` connects to the PL011 hardware UART (accessible via GPIO pins 8 and 10). `ttyS0` uses a differrent hardware address and interrupt (mini-UART / Bluetooth?) -- in either case it does not address hardware connected to pins 8/10.

Suggest removing [references to `/dev/ttyS0`](https://github.com/cyoung/stratux/blob/master/main/ry835ai.go#L139-L140) from ry835ai.go.

```
root@raspberrypi:/dev# dmesg | grep "tty"
...
[    0.268390] 3f215040.uart: ttyS0 at MMIO 0x3f215040 (irq = 29, base_baud = 50000000) is a 16550
[    0.927135] 3f201000.uart: ttyAMA0 at MMIO 0x3f201000 (irq = 83, base_baud = 0) is a PL011 rev2
```

```
root@raspberrypi:/dev# cat /dev/ttyAMA0
$GPGSV,3,2,12,51,36,199,,17,33,091,27,25,18,317,25,24,18,247,26*76
$GPGSV,3,3,12,03,09,034,14,05,05,187,16,28,03,147,16,23,01,068,*79
...
```

```
root@raspberrypi:/dev# cat /dev/ttyAMA0
root@raspberrypi:/dev# 
```

peepsnet, 2016-04-20:
So can we get the 835/836 to connect on the pi3? 

I am having an issue with that. I can get ttyS0 to show data with the 835 connected to pins 8 & 10. But it is unreadable. I am not able to get any baud rates to return usuable data... 

If I `screen /dev/ttyS0 9600` I see data but it's junk. If I disconnect pins 8 or 10 the data stops...

skypuppy, 2016-04-20:
http://www.briandorey.com/post/Raspberry-Pi-3-UART-Boot-Overlay-Part-Two

AvSquirrel, 2016-04-20:
Add `dtoverlay=pi3-miniuart-bt` to the end of `/boot/config.txt`. This will release the Bluetooth adapter from using the hardware UART, and make serial accessible over the GPIO pins.

AvSquirrel, 2016-04-20:
Simulpost! :-D

skypuppy, 2016-04-20:
Great minds think alike: always in the gutter.

peepsnet, 2016-04-20:
I followed the steps here: 
http://www.briandorey.com/post/Raspberry-Pi-3-UART-Overlay-Workaround

and was able to get valid NMEA data on ttyAMA0

Does this mean we have to have version checks for RPi3's and mod the `/boot/cmdline.txt` , `/boot/config.txt` and `sudo nano /lib/systemd/system/hciuart.service` on all RPi3 on initial boot??  or can we do this for all RPi's and it will still work??

AvSquirrel, 2016-04-20:
@peepsnet -- we're using a version of Jessie dated after 3/18, so the simplified method in @skypuppy 's post (and originally discussed on reddit: https://redd.it/4bh5ku) is what we've been using on our Pi3s, and what I submitted in PR #394.

In theory, this should still allow BT use on the "mini UART" - though I haven't tested. Also need to validate on older Pi's.

peepsnet, 2016-04-20:
ok. ill try that.

peepsnet, 2016-04-20:
SO the question become does this line in the /boot/config.txt break the RPi2s and RPi0s...

PR #394 (2016-04-22): GPS UART fix for RPi 3
- Resolves #393 by removing `/dev/ttyS0` from GPS initialization sequence. On the RPi 3, this port does *not* connect to GPIO pins 8 and 10; receivers connected to these pins are therefore not detected.
- Add a interactive script for configuring MTK3339 GPS receivers from the command line
commit 6e5d449a (2016-04-22): Merge pull request #394 from AvSquirrel/gps-uart-fix

GPS UART fix for RPi 3. Fixes #393.
closed 2016-04-22 by cyoung