cyoung/stratux #395: [Feature Request] - N-Number/Hex Code Database for OwnShip(rental)

peepsnet, 2016-04-20:
Since we are moving to an sqliteDB maybe we could add to the list of wants the ability to add Hex Codes to a table for recall in ownship filtering. 

Table:

ACID | N-Number | HEX
--------- | ---------- | ----------
1 | N12345 | A34D3E
2 | N54321 | A66DE2

Then you could modify this in the webui and select the current ownship you want to use. 

I am looking at a page that would list the stored A/C in the table:

ACID | N-Number | HEX | Delete/Add
--------- | ---------- | ---------- | --------
1 | N12345 | A34D3E | X
2 | N54321 | A66DE2 | X
3 | blank | blank | +

Then a dropdown list of all the a/c and you would just select your current a/c to be ownship.

This would allow people who rent a/c or flight train with multiple a/c easier to filter. 

What do you think. 
I know this would be pretty far down on the list but still possible...

JohnOCFII, 2016-04-20:
I think this would be a great idea.  I'm in a club with six aircraft, and this would be a useful, time-saving feature.

ssokol, 2016-04-20:
That would be easy and also useful if you want to do things like:

- Track your friends
- Figure out who to yell at if you have a near-miss

Anyone know how often the registry data gets updated?

skypuppy, 2016-04-20:
Start doing stuff like that and you might want to use a 64 or 128 Gb SD card.  (Can the Pi's address the 128 Gb cards?)

peepsnet, 2016-04-20:
So maybe??

ACID | N-Number | HEX | Notes
--------- | ---------- | ---------- | ----------
1 | N12345 | A34D3E | 
2 | N54321 | A66DE2 | That guy that almost hit me

heelix, 2016-04-20:
Hah.  Be nice to flag NORDO birds that share the patterns.