cyoung/stratux #400: GPS not detected when plugged in AFTER Stratux start

cyoung, 2016-04-24:
1. Stratux version: v0.8r2 (a2ca212a52)
2. Stratux config:

    SDR
      [x] single
      [ ] dual

    GPS
      [x] yes
      [ ] no
      type:

    AHRS
      [ ] yes
      [x] no

    power source: CanaKit 2.5A wall wart

    usb cable: built-in

3. EFB app and version: None

    EFB platform: N/A

    EFB hardware: N/A

4. Description of your issue:

When GPS is plugged in after Stratux start, GPS status stays in "disconnected" state (per WebUI).

GPS comes up on `/dev/ttyACM0`. Pi3.

AvSquirrel, 2016-04-24:
Can't reproduce on Pi2 / Wheezy or Pi3 / Jessie (with or without `pi3-miniuart-bt` overlay)

Is anything attached as `/dev/ttyUSB0`?

skypuppy, 2016-04-24:
I have seen this lately, too, with a USB GPS and with the serial GPS.  0.8r2 with image downloaded from stratux.me.  I don't recall it occurring when I loaded the full blown jessie, then sh'ed 8.8r1 in.

cyoung, 2016-04-27:
@AvSquirrel - there was. It seems like it does it mostly when removing one GPS device and inserting another. Probably has to do something with some baud rate hysteresis.

AvSquirrel, 2016-04-27:
> Is anything attached as `/dev/ttyUSB0`?
> there was.

That would do it. This goes back to the BU-353 support (#231) that was committed a few months ago.

If `initGPSSerial()` [sees `/dev/ttyUSB0`, it assumes that there is a BU-353 attached](https://github.com/cyoung/stratux/blob/master/main/ry835ai.go#L128-L137) and starts configuration on that port, ignoring any devices on `/dev/ttyACM0` or `/dev/ttyAMA0`.

I can address this when I rework the GPS initialization. Moving `/dev/ttyACM0` ahead of `/dev/ttyUSB0` in priority order should resolve this for the recommended use cases (VK-172 or other u-blox over USB). Not sure yet how I want to handle GPIO.

peepsnet, 2016-04-27:
Could we do some kind of detect all GPS types and allow user to select from a list if more then one??

This could be useful on the new RPi3.  Since se only use 3 of the 4 USB ports one could have a backup GPS. Or maybe a RY835/6AI and a VK-172. And the user could select from a dropdown which GPS to use.  OR an AUTO mode where the software picks.

AvSquirrel, 2016-07-15:
I'm able to repeat this on a fresh image of v1.0r1 (945c8bf). I must have disabled the serial console on a previous image, which is why I didn't see it back in April.

There's an issue with `/boot/cmdline.txt`, where `/dev/ttyAMA0` is being treated as a console device:

```
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
```

With no USB device connected, the GPS reader assumes a serial connection, but ttyAMA0 never times out. Since the device never times out, the GPS reader doesn't break out of the read loop, and never initializes the USB GPS.

This also seems to be responsible for the intermittent GPS connections via UART GPIO that some users have reported on reddit.

Manually changing this file to remove `console=ttyAMA0,115200` seems to resolve this:

```
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
```

I'm fairly weak on bash scripting and have no experience with spindle, but it looks like the problem is caused by adding ttyAMA0 as a console device in spindle [wheezy-stage2](https://github.com/cyoung/stratux/blob/8de91a55f029e94aa441e564a96d9cae7806e55b/image/spindle/wheezy-stage2#L247-L248):

```
echo "dwc_otg.lpm_enable=0 console=ttyAMA0,115200 \
console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait" > /boot/cmdline.txt
```
and not properly stripping it out in [wheezy-stage4](https://github.com/cyoung/stratux/blob/8de91a55f029e94aa441e564a96d9cae7806e55b/image/spindle/wheezy-stage4#L15-L32). The regex is valid, but it tries to operate on a file in the working path, rather than /boot.
```
 sed -i boot/cmdline.txt -e "s/console=ttyAMA0,[0-9]\+ //"
```

For updates on existing images, [this branch](https://github.com/cyoung/stratux/compare/master...AvSquirrel:boot-fixes) should resolve the issue on .sh updates and mkimg builds. I can sumbit a PR for tracking or you could test / merge it yourself. How should spindle be handled?

cyoung, 2016-09-26:
Fixed with PR #464.

closed 2016-09-26 by cyoung