cyoung/stratux #401: Update takes longer beacuse of large SQLite logs.

peepsnet, 2016-04-24:
1. Stratux version: anything after v0.8r2-32097e3b61 24-Mar-2016 21:56 
2. Stratux config:

    SDR
      [ ] single
      [ X ] dual

    GPS
      [ X ] yes
      [ ] no
      type:

    AHRS
      [ X ] yes
      [ ] no

    power source:
SHORE
    usb cable:
PORTAPOWER
3. EFB app and version: (e.g., WingX Pro7 8.6.2)
N/A
    EFB platform: (e.g., iOS 9.2)
N/A
    EFB hardware: (e.g., iPad Mini 2)
N/A
4. Description of your issue:

The last 2 SQLite files were almost 3GB each and takes a while to move/rename

That was about 100 hours(or more, at one point I left my stratux on forabout 3 days straight) of logs each but still we might look into splitting the log files at some point. 

By date, power cycle, size, or record count.

skypuppy, 2016-04-24:
I still recommend simply storing the raw data and using post-processing to put that data into a database.

peepsnet, 2016-04-24:
I like that idea

Axtel4, 2016-04-25:
Not being a software person, I find the SQL database cumbersome to work with.

peepsnet, 2016-04-25:
I like the idea of being able to query a database for info if needed but not at the expense of the project.

skypuppy, 2016-04-25:
I am a database person and *I* find SQL cumbersome, extremely so.  It's a 3 1/2 gen language cobbled together, apparently by committee.  However, it is in common usage so a DB person has to deal with it if they wish to eat, mostly because the "other" DB people don't know any better.  Of course, the bean counters and management doesn't even know that much..  More advanced methods have been introduced as far back as 1980, while SQL was still a fledgling.    Ahem.  /rant off/
Post processing, yeah, that's the ticket.  For one thing, it allows each person to import the data into the tools they prefer and rarely consumes an entire SD card when worked in a desktop or even tablet.

Axtel4, 2016-04-25:
It may just come down to what I am used to working with and maybe my SQL process needs revised.  With the previous logging method as I changed test conditions I would stop logging, change the test conditions, then restart the logging.  Each restart would create a unique file for those test conditions.  I would make a note of the logs of interest and only upload those logs for additional analysis.

 

Now I run a series of test data in various configurations and save them to the SQL log.  To post-process the data I upload the log to my desktop and use SqliteBrowser to view the log and extract the data I want to post-process.  What I find cumbersome is identifying a clear demarcation between different data runs and being able to cleanly extract the data of interest for post processing.

cyoung2, 2016-04-25:
"startup" table was added just recently and increments just as the log file
numbers did before. The timestamp rows are tagged with the ID.

Axtel4, 2016-04-25:
Thanks. I'll check it out.

Axtel4, 2016-04-25:
I've updated to Version: v0.8r2 (bd2402f0fb) and ran a series of tests while recoding the results.  I stopped and started recording between the test.  I have a few questions about the data.  

Looking in the startup table (startup.csv), I have 4 empty lines that correlate to the four times I started recording.  Is there something I need to do to record the time stamp / GMT time each time the recording was started / stopped?  

In the es_messages log (1090ES.csv) In the "data" block, what is the easiest way to extract Icao_addr, Lat, Lon, Altitude, and Signal level from the large data block?

In general, which tables / logs do I look in to correlate the "Time Received" with the GMT time recorded for the logs?  I see there is a "Timestamp" entry at the end of the data block, is this what should be used?  

Also, over the 7 minute test period the SQL Log is over 30 MB.  Is it normal to record almost 4+ MB / min into the SQL log?

Thanks.

Attached is the 1090ES log from the test.  
[1090ES.zip.txt](https://github.com/cyoung/stratux/files/235553/1090ES.zip.txt)

[startup.csv.txt](https://github.com/cyoung/stratux/files/235564/startup.csv.txt)

bradanlane, 2016-05-01:
The value of a database system is data queries - hence the "q" is SQL.

If Stratux is using it for storage only, then it doesn't add value to the product. If the goal is "data analysis" then the suggestion of a file converter/importer utility is a good solution.

It really comes down to "does SQLlite make Stratux faster, more stable, or add important features to its role as ADSB weather and traffic receiver?"

peepsnet, 2016-05-01:
I am a fan of SQL!

If we limit the size of the SQL DBs(by date, size on disk, or record count) and make new ones, then that would work. 

The option of taking RAW log files and importing them into a SQL DB for debugging, post flight, would be a great option also. 

Is it just SQLite that has the filesize issue? Would other SQL files build this fast? MySQL?

skypuppy, 2016-05-01:
By the missions of SQLLite and MySQL (and PostGresSQL) there is much more power and flexibility in the larger, more complex programs.  There are also other DB managers that can interact with SQL that may be brought into play in a post-processing scenario, some with 3D graphics capability.  IF (ant it's a big if) the Pi can accommodate even SQLLite as a real-time tool for the average pilot then the choice is obvious.  Even if used as a debugging tool for the developers the advantage is obvious.  However, due to it's resultant size and requirements overhead, it is probably not suitable for normal use; minimum requirements appear to be a -3 and 64 Gb SD card as a minimum.  Only Chris could make these a requirement for all stratux users.  Personally, I already use 64 (and on occasion the 128) Gb SD cards but that is only because Qt is SUCH a resource hog.  I now have 7 radios, 1 Pi 3, 2 Pi 2's, 9 SD cards, 3 power supplies, 2 10 port powered hubs (uSB 3.0 for future capability), 2 Beaglebone Blacks, one keyboard, one mouse, one HDMI monitor, and two bleary eyes, all dedicated to the stratux effort.  I purchased a years "subscription" to Qt at $600 and am using my zeal for the stratux effort to kinda force my hand at learning how to use Qt.  :)  :)
As the hardware progresses, the costs climb but very slowly compared to any other hobby I've ever pursued.  Hahahahaaha.  Radios at $25 each, antennas (with proper cabling) at $17 for the pair.  Batteries at $68, other miscellaneous cables at $2 to $30 each, and so on.  With nearly a thousand spent over a years time, it is trivial compared to riding motorcycles and we won't even MENTION the costs of flying.  <very big grin>  The time spent, now that's a completely different story, but we can ignore that :)
<sigh> I'm ranting again.

skypuppy, 2016-05-01:
Peepsnet, are you a fan of SQL or of the power and capabilities of databases?

peepsnet, 2016-05-01:
the later... but I know SQL most experience in MySQL( I did php,mysql programming mostly )

skypuppy, 2016-05-01:
I'm trying to learn how Qt connects and works with the various DB engines but first I have to learn Qt itself.  :)

peepsnet, 2016-05-01:
What is Qt?

skypuppy, 2016-05-01:
How do we take a discussion like this off github?  
My private email is ainut@knology.net

cyoung, 2016-09-26:
See #310.

closed 2016-09-26 by cyoung