cyoung/stratux #402: Altitude error above 32768ft

AMNegron, 2016-04-26:
1. Stratux version:  0.8r2
2. Stratux config:
    SDR      [ x ] single - type: ​NooElec Nano 2 

    GPS      [ x ] yes - type: U-blox7

    AHRS    [ x ] no

    power source: Innergie PocketCell - 3,000 mAh 2.1A
    usb cable: USB2.0 / 28AWG

3. EFB app and version: FltPlan Go 4.2.4
    EFB platform: iOS 9.3.1
    EFB hardware: iPad Mini 1

4. Description of your issue:

Altitudes above 32768 ft are not being passed properly in the data stream. Above that altitude, on FPG, the altitude value gains a "few extra bits", but on FF the altitude turns negative. GPS altitude value compared to internal iPad GPS.

Test 1:
Level flight FL360. Stratux GPS GUI showed 36521 ft. 
FltPlan Go (iPad mini) was showing 4294958387.0 m
ForeFlight (iPad 2) was showing -28985 ft (http://imgur.com/DoCmMzY)

During descent passing FL330, FPG jumped down to 9999 m and values remained fairly accurate through the remainder of the descent. On FF, the altitude descended until reaching approximately -33000 ft, then the negative sign disappeared and again, the value remained fairly accurate (and continued a proper descent) for the remainder of the descent. 

Test 2 (iPad mini)
Level flight FL330. Stratux GPS GUI showed 34015 ft (http://imgur.com/s283swH)
FltPlan Go (Stratux ublox) was showing 4294957692.0 m (http://imgur.com/Rxz7fnN)
FltPlan Go (iPad GPS) was showing 10385.31 m (http://imgur.com/gdq64rE)

I think the trigger point is 32768 ft (9989 m) [0x8000] 

Attached logs should be from Test 2.
[Stratux_Altitude_Error.zip](https://github.com/cyoung/stratux/files/235787/Stratux_Altitude_Error.zip)

=================
If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

skypuppy, 2016-04-26:
Cool.  FltPlan Go with ublox was showing you at 21,664 miles, or about where the geostationary satellites are in orbit!  What an airplane!!!  May I borrow it sometime?

AvSquirrel, 2016-04-26:
It's a unit16 overflow during creation of the ownship altitude message. Noted as issue #373, addressed as #375, and will be included in the next release.

cyoung, 2016-04-27:
Thanks for the report @AMNegron. This has been fixed, watch for v0.8r3 soon.

closed 2016-04-27 by cyoung