cyoung/stratux #404: 8r1 missing traffic

rapgood, 2016-04-28:
1. Stratux version: 8r1
2. Stratux config:

    SDR
      [ ] single
      [x] dual

    GPS
      [ ] yes
      [x] no
      type:

    AHRS
      [ ] yes
      [x] no

    power source: wall

    usb cable: no, on wifi

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2) iOS 9.2

    EFB hardware: (e.g., iPad Mini 2) iPad 4

4. Description of your issue: 8r1 doesn't see traffic that is detected by 7b1 and piaware. reverted to 7b1.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

rapgood, 2016-04-28:
correct that to 8r2

cyoung, 2016-05-13:
Have any examples of the traffic?

Any chance you could run the two side-by-side and and send the "es" logs?

rapgood, 2016-05-16:
I can probably do a side-by-side. Is there a way to designate the non-routable ip without a build so I can monitor them separately?

cyoung, 2016-05-16:
`/etc/hostapd/hostapd.conf`

rapgood, 2016-05-16:
Is there a thread or white paper discussing logging (where located, etc.). Also, uname  and pwd (I've been away from *nix systems for about 20 years, but can still muddle thru it).

rapgood, 2016-05-16:
Got past all that except log lock.
Quick question re: ip conflict potential. Can you call? 206-930-9511
Thnx
--r

Is there a thread or white paper discussing logging (where located, etc.). Also, uname  and pwd (I've been away from *nix systems for about 20 years, but can still muddle thru it).

rapgood, 2016-05-16:
Sorted out everything except where you store your logs (wow! All that UNIX stuff was still rattling around in the recesses of my memory!)

--r

Got past all that except log lock.
Quick question re: ip conflict potential. Can you call? 206-930-9511
Thnx
--r

Is there a thread or white paper discussing logging (where located, etc.). Also, uname  and pwd (I've been away from *nix systems for about 20 years, but can still muddle thru it).

rapgood, 2016-05-16:
Well, of course now I can't duplicate.  So, you might as well close the 
ticket (keeping in mind that problems that go away by themselves come 
back by themselves ;) ).

--r

ssokol, 2016-05-18:
I've had a number of FlightBox customers report missing traffic. One was on flight following an reported a number of aircraft called out by ATC and visually identified that were not visible on their EFB (FlyQ) with Stratux v0.8r2.

rapgood, 2016-05-18:
I am still seeing missed traffic on 8r2, but it is intermittent and, thus, difficult to document. I can hear the AC overhead, and see it on piaware, but FF connected to stratus shows empty.

cyoung, 2016-05-18:
Does that PiAware interface only show ADS-B out or is it getting MLAT data back from FlightAware?

Nokomis449, 2016-05-18:
Stratux will not see "ALL" traffic that Flightaware sees.  Stratux (not counting retransmissions from a tower) will only see aircraft equipped with ADSB OUT.  Flightaware uses triangulation from multiple sites in order to guesstimate th location of Mode C-only aircraft.  Click on an "MLAT" aircraft and you'll see how the track zigzags as Flightaware changes its guess.

rapgood, 2016-05-18:
MLAT is zero in piaware stats. flightaware.com/adsb/stats/user/Robflying

rapgood, 2016-05-18:
True. But see "no mlat" in my stats and traffic is commercial (read "heavy") traffic that I see in 7b2 stratus.

Nokomis449, 2016-05-18:
Looking at your site now... Interesting.  I've never noticed missing traffic in flight but I'm not sure how I'd know who was -OUT and who, like me, is -C only.

rapgood, 2016-05-18:
I happen to know the particular equipment and flights, so know what to expect and when. As my stats show, I'm in a hole and "see" few AC every day.

AvSquirrel, 2016-05-18:
@ssokol , @rapgood : Have you verified the connection of the MCX pigtails to your SDRs?

More than once, I've seen huge insertion losses (+20 dB as measured on the traffic UI page) caused by an antenna cable being slightly askew in the connector. This has affected both the "mini" and "nano" sized SDRs. Sometime this was due to side loading due to cables being bent; in one case, it was just gravity loading from an antenna plugged directly into the connector. 

In my semipermanent airplane install, I strain relieved the antenna cables to ensure that they're always square to the connector, and generally see good results, with typical message rates of 25,000 ES messages and 2000 UAT messages per minute (5-7 towers) in flight near MSP.

rapgood, 2016-05-18:
See with 7b2 but not 8r2. ASA going over right now, bur 8r2 doesn't see it

AvSquirrel, 2016-05-18:
@rapgood -- can you clarify if you are using one RPi and swapping cards, or simultaneously running two separate RPis? And are you running full .img images for both the v0.8r2 and v0.7b1 installs, or was one a .sh or git pull update?

rapgood, 2016-05-18:
One RPi, swapping cards. Full .img on both releases.

rapgood, 2016-05-18:
If you like, I think I can assign a class b routable static ip to the RPi to externalize and load 8r2 so you can square my piaware stats with stratux (understanding, of course that the piaware display will be approximately one minute behind the stratux detections due to flight aware constraints). I may not be able to because I had problems attempting to assign a class a non-routable IP to the Stratux configuration and make it work.

AvSquirrel, 2016-05-18:
Sure. PM me and/or cyoung on slack or reddit with the IP if you get it up and running; I should have some time this evening to help out.

How much traffic would you say that Stratux is missing -- is it showing any 1090 messages? Are you seeing some 1090 targets?

You mentioned that you're running PiAware. Are you running a dedicated PiAware image, or is this the add-on package on your 0.8r2 image?

rapgood, 2016-05-19:
No joy on externalization. No routable network available on wifi and RPis are all on wifi (could hardware RPi, but then lose signal to antennae).

Ergonomicmike, 2016-05-27:
FWIW, there was a Florida pilot here in the github who made his UI available over the web. (I could even change his ettings!) It's got to be in one of the closed issues now. But if you find it, perhaps you can ask how he did it.

Nokomis449, 2016-05-27:
Mike, I haven't tried it, but it should be as easy as port forwarding to the IP address that your Raspberry Pi grabs on your home network.  Easy, but not secure.

rapgood, 2016-06-03:
Upgraded to 9b1 and am seeing 3x traffic I was seeing on 8b2. But, bear in mind, problems that go away by themselves come back by themselves! ;)

emuyshondt, 2016-07-22:
I flew and drove around with a FlightBox yesterday and noticed an anomaly, most likely with ForeFlight, but I thought I'd report it:

I have a dual band FlightBox receiver (based on the Stratux open source project). It receives both ADS-B/UAT 978 traffic as well as 1090ES transponder traffic. I hadn't looked at traffic much on ForeFlight before, so I'm not sure if the problem is new (I just upgraded to the latest version of FF, 7.7.2). FF only shows the 978 traffic (both ADS-B and a few planes with 978 UAT transponders). All the traffic I see says either ADS-B 978 or TIS-B 978 when I tap on it on ForeFlight. No air-to-air 1090 ES traffic is displayed, though switching over to WingX Pro shows it all. I am running WingX Pro on the same iPad, connected to the same receiver. I have also made sure the "Hide distant traffic" setting of FF is off so it is not just a quirk of the filtering algorithm. The data is clearly being sent over the network by the receiver, since WingX sees it. The Stratux status page on my iPad also shows all the traffic. It is just FF that seems to be discarding the 1090ES TIS-B information. 

I wrote to FF yesterday and here's what they replied:

[quote]
Hi there,

ForeFlight's stance on Stratux-based products is neutral at the moment. If you want to use it, that’s great. We have no plans to block it from connecting to ForeFlight Mobile, but we also don’t have a formal agreement or partnership with the Stratux/FlightBox developers so you’ll need to look to the Stratux-based community for support resolving Stratux-related issues.
[/quote]

I understand to some extent, but I am not convinced they're not doing anything on purpose. There shouldn't be anything different in the data streams once they are decoded.

jzeevi, 2016-07-22:
Is flightbox running 8r2? There’s been a ton of work on Stratux since then (there’s a 1.0r1 that’s the latest stable release) and I know I haven’t had that problem (but haven’t looked specifically for it).

JohnOCFII, 2016-07-22:
> I hadn't looked at traffic much on ForeFlight before, so I'm not sure if the problem is new (I just upgraded to the latest version of FF, 7.7.2). FF only shows the 978 traffic (both ADS-B and a few [...]

It's a known issue with the software on Stratux.  All traffic is tagged as 978UAT traffic, even the traffic that comes in via 1090ES.  You can see where the traffic actually originates by going to the Settings page in the Web User Interface for Stratux and enabling "traffic markings"  In that way, the traffic is tagged with extra characters that show the actual source of the traffic.  

If WingX is differentiating it, they must be doing some further decoding of the streamed traffic themselves.

emuyshondt, 2016-07-22:
FF is showing less traffic than WingX Pro. It tags some traffic as 978 TIS-B and other as 978 ADS-B. Most of the missing traffic is airlines going to KAUS, so I think they are 1090ES targets. The Stratux UI does show blue and gold tags. I can't remember off the top of my head what the difference is and don't have access to my FlightBox or my Stratux right now (I have both).

emuyshondt, 2016-07-23:
These two screen captures were seconds apart. 

[image]
[image]

JohnOCFII, 2016-07-23:
Interesting.  I haven't tried with WingX, but I did run with a Stratux (v1.0r1) feeding one iPad running ForeFlight, and an Appareo Stratus 2 feeding ForeFlight on another iPad.  Every now and then, the iPad's would be different for a few seconds, but then they would show the same data.  I've seen the same thing with two Appareo Stratus units feeding ForeFlight on two iPads.

This may be different than what you are seeing, but wanted to add another data point.

AvSquirrel, 2016-07-23:
[image]

@emuyshondt - check your ForeFlight settings and make sure that "Hide Distant Traffic" is unselected. That setting is used to declutter the display by hiding any targets more than 15 NM and 3500 feet vertically from your location.

All traffic is tagged as 978 MHz in ForeFlight on Stratux due to limitations in the format of the GDL90 traffic messages.

The GDL90 was a UAT (universal access transceiver; i.e. 978 MHz in+out) developed initially by Apollo, and later by UPS AT and Garmin after a series of acquisitions in the early 2000s. Because this was used in the FAA Capstone R&D project, the protocol that it used to communicate with multifunction displays was made public.

Years later, FreeFlight Systems apparently adopted the protocol for their RANGR series of UATs, and also used to communicate over their WiFi adapter to applications like ForeFlight.

Because the protocol was developed for 978 MHz only devices, there was no need to specify the traffic source; all traffic was 978 MHz. ForeFlight assumes the same, and it was only after we added 1090 support to this project last Fall that this became a source of confusion.

emuyshondt, 2016-07-23:
I have triple checked that the switch for Hide Distant Traffic is off.

closed 2019-01-17 by Helno