cyoung/stratux #407: Support for RY836AI (and devices with MPU9250 / BMP280 modules)

AvSquirrel, 2016-04-30:
1. Stratux version: v0.8r2 (AvSquirrel@bad8390)
2. Stratux config:

    SDR
      [ ] single
      [x] dual

    GPS
      [x] yes
      [ ] no
      type: u-blox M8N (RY836AI)

    AHRS
      [X] yes - MPU9250 (RY836AI)
      [ ] no

    power source: 2A wallwart

This issue is being submitted to track support for the MPU9250 9-axis AHRS  and BMP280 pressure sensor found in the RY836AI and other modules.

[Background information on reddit](https://redd.it/4f1wvo)

Proof of concept 6-axis IMU has been added a new [mpu9250 branch](https://github.com/AvSquirrel/stratux/tree/mpu9250). This requires modification to the linux-mpu9150 middleware since module type is a compile-time option, and to ry835ai.go to suppress error handling related to the BMP180 pressure sensor... which also has changed formats.

skypuppy, 2016-05-01:
AvSquirrel, what to do you think of us developing a back-end library for IMU's, similar in function to gpsd?  It would allow multiple types of IMU's and not lock anyone in to a specific model.  Adafruit has  good start at it, as do several other groups.  If you've read the book by Professor whats-his-name about the Beaglebone Black (BB), he gives the basis for one as well, in plain English but not much depth.  Prof. Derek Molloy.  His book focuses on the BBB specifically, but 95% of the writing is applicable to SOC's that run Linux in general.  The BBB was my platform of choice for the equivalent to stratux until the RPi2 came out.  There are several IMU's MEM's devices on the market and more coning in the future.

peepsnet, 2016-05-01:
@AvSquirrel and I were discussing the same thing. 

I was pitching the idea of having a library for each chip.

This is an excerpt from our conversation:
>ME: OK. So we(by we I mean someone else with the knowledge) could write functions to handle each chip then have a library of chip configuration or a method of detection and just turn on each chip type. like an array of hardware?
>
>so. the ry835ai.go file would kinda be replaced by... lets say...  
MPU9250.go and
BMP280.go and 
MPU6050.go 
BMP180.go and 
ublox8.go
bu353.go

>each a library for each chip?

> >This will require a fork / rewrite of the existing code.

>Now I see what you mean!!

skypuppy, 2016-05-01:
It makes the most sense to me, as well as incorporating the (already proven) gpsd.  IMO, we gotta get away from writing for specific ships as the market changes so rapidly else we'll write ourselves into oblivion.

tino-247, 2016-11-18:
I would like to participate in the development as well.
Libraries are definitely the best approach. One for each chip and then one doing the calculations.

Here is a page that offers some algorithms for the calculations:
http://x-io.co.uk/open-source-imu-and-ahrs-algorithms/

Is anyone working on it at the moment?
I have an RY835AI and could dedicate some time.

cyoung, 2016-11-18:
Now is a good time.

https://github.com/cyoung/stratux/tree/gpsattitude

I'm working on removing RY835AI support since it has some pretty bad hardware limitations.

There is no certain contender for AHRS hardware, so if you have any research findings then it would be helpful.

tino-247, 2016-11-18:
After some reading I'm leaning towards the BNO055. It seems to have some bugs and the BNO070 is supposed to be better, but that doesn't help since it isn't available.
Another candidate could be the FIS1100, but it's not so well adapted by the community.

Adafruit has a breakout board for the BNO055 and software and instructions for the rPI available.
Here are some links I collected:
https://learn.adafruit.com/bno055-absolute-orientation-sensor-with-raspberry-pi-and-beaglebone-black
https://learn.adafruit.com/adafruit-bno055-absolute-orientation-sensor
https://forums.adafruit.com/viewtopic.php?f=19&t=81384&p=533456

I ordered a BNO055 and when I got it I will wrap my head around it how to get the data we need.

andofrjando, 2016-11-20:
I like the look of the BNO055. I've just ordered one too, so happy to join the party with development. Seems like a good reason to learn GoLang.

tino-247, 2016-11-20:
My BNO055 arrived today. I'll first play around with it on my HW platforms. When it looks good I will implement it into stratux. I'm programming in C/C++ for embedded and C# for desktop.
@cyoung can you point me to the hardware limitations of the RY835AI? I found a couple of projects working with the MPU-6050 and HMC5883L.

tino-247, 2016-11-21:
The BNO055 is up and running. I implemented the initialization and calibration and checked which data is most reliable and I must say that so far the internal sensor fusion works really well.
I tried the Madgwick and Mahony algorithms with the raw sensor data but they're quite unreliable.
The Euler data of the BNO055 has unreliable yaw at over 45° roll.
The Quaternion data converted to Euler works great.
Du to the hardware I2C clock stretching bug in the Raspberry Pi we have to connect the BNO055 to the serial port. For now I will just use an USB-serial adapter that also supplies the power (http://www.ebay.com/itm/162141314370) and later we can switch to the port that the RY835AI used so far.
I'll create a new file BNO055.go and implement what's needed in there. Then we can check how to integrate it.
I also want to add some features to the website to start calibration and display status/error messages.
Do you use any remote debugging or do you work locally on the rPI?

tino-247, 2016-11-22:
Somehow I got along with GO and ported my code to it.
The BNO055 is now functional on the rPI with the USB-serial adapter.
Calibration works and it's giving me nice data.
Next step is the integration into stratux which should be fairly easy.
Just use "sendMsg(prepareMessage(msg), NETWORK_AHRS_GDL90, false)".

tino-247, 2016-11-23:
Working

tino-247, 2016-11-26:
I also started integrating the settings into the web interface. But compiling takes about 6 minutes and I'm a bit rusty with the angular framework, so I better don't change too much.
Whenever you have time we can work on that.
The BNO055 is already integrated and working. I also prepared the functions to be called from the web interface.
So we could let some people test it.

cyoung, 2016-11-26:
`go install github.com/mattn/go-sqlite3`

speeds up the compiles quite a bit.

cyoung, 2016-11-26:
Do you have the code on GitHub? Can't find the implementation you're referring to.

tino-247, 2016-11-26:
I couldn't commit to your branch, so I forked it and committed there:
https://github.com/tino-247/stratux/tree/gpsattitude
In the Git desktop tool it shows committed but it's not online. I'm not really used to working with Git and it's not my favorite (yet?).
I'm new at Go and the code is probably not overly optimized since I don't know all classes and functions yet. But it runs stable and does what it should.

How did you know about "go install github.com/mattn/go-sqlite3"? It brought down compiling to about 10s. That makes it definitely easier to add some more features.

cyoung, 2017-08-04:
Has been completed as of v1.3 betas.

closed 2017-08-04 by cyoung