cyoung/stratux #409: How long for cached NEXRAD to populate?

Ergonomicmike, 2016-05-01:
1. Stratux version: v0.8r2 + April 28 .sh
2. Stratux config:

    SDR
      [ ] single
      [ ] dual

    GPS
      [ ] yes
      [ ] no
      type:

    AHRS
      [ ] yes
      [ ] no

    power source:

    usb cable:

3. EFB app and version:  IFly v9.7.1b

    EFB platform: Android KK

    EFB hardware: Asus Zenpad S

4. Description of your issue:

I'm sitting at an airport with an ADS-B tower nearby. Can get 978 here on the ground. (Seems like the FAA has been installing more towers lately. I wasn't able to get ADS-B here before.) I left the Stratux running while eating dinner. EFB off. There are TRW's around. I come back to the plane and power up my EFB. NEXRAD can take 5 minutes before an image appears. Is that cached FIS-B or over the air?

Axtel4, 2016-05-01:
I've seen the NEXRAD image take as long as 15 minutes to  fill in.

peepsnet, 2016-05-01:
https://bruceair.wordpress.com/2014/09/09/flight-information-service-fis-b-weather-and-info-in-the-cockpit/

Ergonomicmike, 2016-05-02:
Thanks guys. I knew that the NEXRAD data can take a while to come over the air. Since it was taking a while for a new image to arrive,  it appears to me that the Stratux is not buffering NEXRAD data to iFly. (I've edited the issue title accordingly.) That is, I expected the latest and greatest NEXRAD image to appear immediately when I restarted iFly, since the Stratux was left on in the plane receiving stuff from the nearby tower.

So, if Stratux stores NEXRAD data as part of the FIS-B stream, then it appears that Stratux does not know when iFly has reconnected? (Or maybe Stratux doesn't store the NEXRAD? Too big?)

peepsnet, 2016-05-02:
I do not know who long the data is stored or what the work flow is for the process. 

@cyoung has said that you have to have connected at least once before in a **stratux session** (I am calling the **time from power on to power off** of the stratux a **stratux session**) then upon reconnect the latest nexrad should come through. 

What it not clear is the internal logic(only because I have not looked at the code or asked this detailed of a question about the process) behind the NEXRAD data. 

Is the latest data stored until new data is received??

Is there a max time to store NEXRAD data, say you leave the startux on and get a bite to eat and that takes 1+30min on the ground away from FIS-B data, will the NEXRAD data be there on reconnect??

I'll think of more questions and get back...

Ergonomicmike, 2016-05-02:
It turns out that iFly displays the time stamp (and calculates the age) of the NEXRAD image. (When you zoom out far enough to get the boundary box for the NEXRAD picture.) So, for example, when I woke iFly back up, I zoomed out and saw an old (45 minute) NEXRAD image. Whereas, if Stratux were storing the image, I expected that the last real time image would paint from the buffer in a minute or two after wake up.

cyoung, 2016-05-03:
@Ergonomicmike - this sounds like it could be an issue with the stratux Sleep Mode detection on either your tablet or the iFly app.

Added extra debug information to work on this. See `http://192.168.10.1/getClients`. There will be clients that are not active in this printout. When you expect stratux to be buffering FIS-B data, the `MessageQueueLen` corresponding to your iFly tablet should be constantly increasing.

If you look at /getClients on a second device while the iFly tablet is sleeping, `MessageQueueLen` will likely be increasing. If you bring the iFly tablet with the iFly app in the background, it should be still increasing. Finally, when you bring the iFly app into context is when `MessageQueueLen` should decrease.

cyoung, 2016-05-03:
`MessageQueueLen` added in update-stratux-v0.8r2-0a62fd2d9a.sh.

cyoung, 2016-05-03:
@peepsnet 

>Is the latest data stored until new data is received??

All FIS-B data is buffered and is sent to the EFB in the order that it was received. FIFO.
The only limitation on this is the 25,000 message cap. Once it reaches 25,000 FIS-B messages it will start dropping the oldest. It does not discriminate between types of data (radar, text reports, etc). Traffic messages aren't buffered and don't affect this system.

Ergonomicmike, 2016-05-03:
@cyoung Cool! I remember using /getClients months ago when you were working on the sleep detection and it seemed to work okay with iFly at the time. In any event, now that there's a new ADS-B Tower near KPAN that I can receive on the ground there, and since there's a fine restaurant there (with fantastic chocolate cream pie), I'll try to fly there again Friday or Saturday evening for a test of these commits.

Axtel4, 2016-05-04:
@cyoung an observation. . .

> If you look at /getClients on a second device while the iFly tablet is sleeping,  MessageQueueLen  will likely be increasing. If you bring the iFly tablet with the iFly app in the background, it [...]

Stratux connected to DroidEFB.  DroidEFB put to sleep.  MessageQueueLen increases.  DroidEFB MessageQueueLen brought into context.  The MessageQueueLen continues to increase.  Only 1090ES and 978 UAT traffic information passed to Stratux as no towers are visible.

from /getClients. . .
`{"Conn":{},"Ip":"192.168.10.38","Port":43211,"Capability":5,"MessageQueueLen":1865,"LastUnreachable":"0001-01-01T00:00:00Z","SleepFlag":true,"FFCrippled":false}}`

Ergonomicmike, 2016-05-04:
Kinda like Axtel4's post, a Sanity Check of getClients.
This is with iFly v9.7.2b only talking to the Stratux. Connected. No ADS-B towers. Am getting a lot of lines, an something about FF Crippled. Just making sure this is normal before I get around to flying it.

[image]

cyoung, 2016-05-04:
You guys are checking this from a browser on a different device from the EFB?

Axtel4, 2016-05-04:
I am checking on a browser from my desktop.

cyoung, 2016-05-04:
@Ergonomicmike - looks like your tablet is 192.168.10.10 and the `MessageQueueLen` is 0. It won't increase unless you're getting FIS-B data.

@Axtel4 - seems different. Have you had issues with DroidEFB not coming back into context and receiving FIS-B data?

cyoung, 2016-05-04:
@Axtel4 - disregard, I had only read the email which didn't include your edit. Thanks.

Ergonomicmike, 2016-05-04:
@cyoung I didn't have iFly running. (Oops. Corrected comment above.) Just had one tablet talking to the Stratux. Don't know why I'm seeing DHCP's 11 thru 16. There's only one neighbor's Router on Ch 1.

Ergonomicmike, 2016-05-08:
Was hoping to fly this evening to test this,. Unfortunately, the wx is too nice. Will plan to fly next time RW are forecast for Phx.

Ergonomicmike, 2016-05-15:
Okay, just got back from some flying/testing today. The wx was thunderstormy today so I flew to KPAN again to try to do some testing of the buffering/sleep client thingie in Stratux. Unfortunately, two problems today:

1) The Stratux totally disconnected in flight about 16 minutes in. Wi-Fi signal still there, so something above the Physical Layer. I tried resetting my tablet's Wi-Fi in flight. No joy. Had to manually power down/reset the Stratux. I could post the relevant part of the Stratux.log if Chris wants it. (I'll check the SD card later to see if it's full. I've had logging off, and I ran the build that was supposed to stop sqlite from growing. Anker was at 4 bars.)

Related to the failure and perhaps a clue for other disconnect problems, when the Stratux came back up, my other tablet was connected. But my Asus tablet would not connect. I had to restart iFly. 

2) When I got on the ground, I was all set to start making vidoes of getClients. But even tho I was receiving the ADS-B signal from the tower nearby (sometimes at -11 dB), no weather data came over the signal. (Although I was getting UAT messages.)

Perhaps that's why the Stratux barfed in flight? I had left two towers, one of which was in Phoenix. When I dropped to just one tower, with no FIS-B, the Stratux quit?

OR - during the flight, just before the Stratux quit, I saw lightning strike the Mogollon Rim, somewhat near where the ADS-B tower is. It could be that lightning took out the box at the tower that sends FIS-B. (I never did get weather until coming back to Phx and picking up a Phx tower.)

Last, on the way home, when I picked up the Phx tower, I recorded a little bit of video of getClients. It looks to me like it's not seeing the iFly sleeping anymore. (It used to many versions ago.) I'll have to review the video to see if it's "legible" and worth posting.

Sigh. While on the ground at KPAN I sh'd up to 1b4c build. The wx is supposed to be interesting again later this week. Maybe I'll try again then.

Ergonomicmike, 2016-05-16:
Okay, I made two videos. (The first was too big in file size to upload at once, so I had to break it up into Parts I & II. Total of three links then.)

The first is about 11 minutes long, and kind of boring. Only a geek could like it.

It was taken at home, on the ground, no ADS-B Tower in range. It basically shows that the only time that the Stratux sets the Sleepflag to true for iFly on my Asus Android tablet (KK) is when the WiFi connection proper on the tablet is turned off. (IIRC, it didn't used to be this way with earlier versions of iFly and/or Stratux.) Nevertheless, iFly sometimes has a problem reconnecting with the Stratux after the tablet wakes up. (See Time Index 0.30 and also 2:30 in Part II.) Stratux issue or an iFly issue?

The second video is somewhat like it. I only had a minute in flight to make this. (Solo - at night, descending fast for arrival in 5 minutes.) I had just started receiving FIS-B weather. I put my iFly tablet to sleep. Consistent with the first video, the Stratux did NOT set the sleepflag = true. So the MessageQueueLen did not change from zero (i.e., did not increase).

(Remember, you can always pause the video to read the screen.)

But I noticed that the sleepflag got set for my other Android tablet, running the same version of iFly. 

Also, the buffer size for it is different than the buffer size for non-connected IP's. Is this expected behavior? (I suppose it would be expected. Different buffers for different devices. But no devices on these addresses.)

First video, part 1 of 2: https://www.sendspace.com/file/dwfg1t
First video, part 2 of 2: https://www.sendspace.com/file/weyzqu

Second video: https://vid.me/aMWb

skypuppy, 2016-05-16:
Mike,
1. There is a setting on Androids that sets wifi on or off while it's asleep.    The default is to not turn it off while sleeping.
2. One can put a tablet to sleep by pressing the power button briefly.  Wake it up by pressing the power button briefly again.
3. Where in the physical world are you?  I couldn't read the fine print on the sectional.  :)

Skypuppy

Ergonomicmike, 2016-05-16:
1. I have my tablets set so that WiFi is off when the tablets sleep. (Saves power. I don't use 'em for email notifications, etc.)
2. Yes. However, on mine, there are also alternative ways of waking them up. 
3. Phoenix area
4. Yes, I do that.

Ergonomicmike, 2016-05-16:
Dumb question: The FIS-B buffering: RAM, right? (As opposed to writing to the SD card?) Thinking about RO in the future.

Ergonomicmike, 2016-05-16:
Skypuppy might have hit on something above. My tablets are set so that the WiFi goes to sleep when the tablets go to sleep. So as a first question: If I can get the sleep flag to set when I manually turn the WiFi off on my tablet, why doesn't the sleep flag get set when my tablet goes to sleep? Is iFly somehow overriding my WiFi settings? 

And as for iFly not reconnecting when my tablet comes out of sleep - does the Stratux expect WiFi to be on while the tablet is sleeping?

cyoung, 2016-05-16:
>does the Stratux expect WiFi to be on while the tablet is sleeping

No

skypuppy, 2016-05-16:
I normally see several seconds of delay between stratux and iFly when going to sleep or waking.  It is not instantaneous.  But it has happened every time.

Ergonomicmike, 2016-05-17:
@skypuppy So then you're saying that the sleep flag works reliably for you? And that iFly always reconnects when you wake your tablet up?

Of course, if you have your WiFi always on when sleeping, I would expect better behavior.

What happens if you set your WiFi to sleep? And what version of iFly are you on? I'm on v9.7.2b.

skypuppy, 2016-05-17:
Mike:  Like you, I set the wifi to be off while sleeping.  There was one occasion when iFly did not reconnect after re-awakening but the Pi3 had zero wifi output -- still haven't figured that one out.  After powering down the Pi, wifi was on when it came back up.
When I set wifi to sleep, when I turn it back on the Pi connects to the (stronger) home wifi rather than stratux.
v9.6.11

Ergonomicmike, 2016-05-17:
Okay, thanks. I'm thinking that this might be an iFly problem. 'Cause 1) it turns out that I had my tablet set for WiFi to stay on when power plugged in, and I was testing that way for while today and still couldn't reconnect. 2) I dropped down to Stratux ab76 and still have the iFly reconnect problem.  I'll ask iFly if anything changed from 9.6.11 to 9.7.2b.

BTW, as a workaround, if I open the Statux UI on the tablet with iFly, then I stay connected after waking up.

skypuppy, 2016-05-17:
I've not played with the sleep flag.

Ergonomicmike, 2016-05-17:
Okay, more sanity checks for the Sleep Flag.

I decided to eliminate iFly from the mix. I used Avare's I/O plug-in instead.

I used two stock Android tablets. (Neither of which was my sometimes problematic Nook HD+ with its custom ROM.) I set up getClients on one. I/O plug-in on the other. (And I switched this around later for yet another Sanity check. Results are the same.)

No matter what I tried - as far as putting tablets to sleep, either by power button or letting them time out - the Sleep Flag did not go "true" on the tablet running I/O plug-in. The only time the Flag when true, as far as WiFi was concerned, was when I manually turned off WiFi on the tablet.

I don't know know why that is, since purportedly, WiFi turns off when the tablet goes to sleep. But I confirmed that WiFi says on by looking at the battery app in Android.  Even with the screen sleeping, WiFi was solidly awake. (I dunno - maybe I have to wait a half hour? Because I know that my HD+ and my old Samsung tablet take a short pause for the Wi-Fi to reconnect when I'm gone for a while.)

I could also get the Sleep Flag to toggle true when I unchecked the "Listen" box on I/O Plug-in. When I do that, the Sleep Flag toggles between false and true, mostly staying true. (But it's not constantly staying true, which might confuse the FIS-B buffering? And/or might confuse iFly on reconnect?)

Unlike iFly, when I would toggle "Listen" on/off, the connection always reconnected.

I also noticed that the Sleep Flag toggles between true and false on the tablet that is only running a browser with the /getClients connection only. I would expect that to stay "false" all the time?

IIRC, part of the reason for implementing the Sleep Flag was to control BW on the Stratux network? Instead of a Sleep Flag, perhaps the number of DHCP clients should simply be limited to four? (Or whatever overhead calculates, with margin.)

Also, IIRC, a secondary reason for the Sleep Flag was to buffer FIS-B data. Which I think is a cool idea. From what I'm seeing in this instant sanity test, the buffering feature is only going to work for me if I turn my tablet off after I've landed, while leaving the Stratux on (if an ADS-B tower nearby). That appears to be the only way (at present) to set the Flag always true for my EFB. More lousy weather is expected this week. I'll try to do some more testing in flight.

Ergonomicmike, 2016-05-18:
iFly informally says that they've duplicated the disconnect on reconnect problem at their end for the beta version I have and are looking into it.

Ergonomicmike, 2016-05-25:
As a side note, iFly also says that they've been able to reproduce Jeff's problem with Flight Plan sharing with the Stratux (https://github.com/cyoung/stratux/issues/331), which only occurs with the Pi2. And only w/ Stratux.

Ergonomicmike, 2016-05-27:
Now that Stratux v0.9r1 and iFly v0.7.19b are out, I'll plan to make a test flight Monday to see if NEXRAD is buffering when my tablet is "off-line."

Ergonomicmike, 2016-05-30:
Am at KPAN. Receiving tower on ground. Everything is working acceptability well with Stratux v0.9b1 and iFly v9.7.19b. 

The only time that the sleepflag gets set is when I manually disable WiFi on my tablet. Or if I shutdown iFly.

Curiously, I show 3 clients on /getClients, even tho the number of clients on the Status page shows 2. This is with a brand new image. AFAIK, only two tablets have ever talked to this image. The third client appears to have been connected for a while, 'cause the message length was in the 15000's when I got to the reataurant. There's a WiFi router on Ch 1 here. Could that be the third client?

I forget - is NEXRAD data supposed to be in the FIS-B buffer?

closed 2016-05-31 by Ergonomicmike