cyoung/stratux #413: 1090 SDR Startup Log error

Axtel4, 2016-05-09:
1. Stratux version:
2. Stratux config:

    SDR
      [ ] single
      [ X ] dual

    GPS
      [ X ] yes
      [ ] no
      type: VK-172

    AHRS
      [ ] yes
      [ X ] no

    power source: Ship power via mygoflight USB adaptor

    usb cable: N/A

3. EFB app and version: (e.g., WingX Pro7 8.6.2) N/A

    EFB platform: (e.g., iOS 9.2) N/A

    EFB hardware: (e.g., iPad Mini 2) N/A

4. Description of your issue:
The ES Device data as reported in stratux.log appears to be corrupted.  This is the snippet from the log:

2016/05/08 21:25:38 ===== ES Device Serial: stx:1090:40������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������ PPM 40 =====

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2016-05-13:
Seems like a fairly innocuous error. Looks like we're parsing the string and assuming it is null terminated but it's not.

What are those characters? If you "cat -e" the logfile it'll show you character codes.

Axtel4, 2016-05-13:
I'll take a look at it.

Ergonomicmike, 2016-05-13:
FWIW, I got confused by these characters in the logs a few months ago when we were troubleshooting the failure of the Edimax. I saw the characters and posted a snippet for Chris, thinking that was when the failure occured. https://github.com/cyoung/stratux/issues/267

cyoung, 2016-05-13:
Yeah, I'll just have to add in some extra string handling. It should not affect function.

Axtel4, 2016-05-14:
@cyoung 
Here is the result for the 1090 SDR:
> 2016/05/13 19:09:17 ===== ES Device Serial: stx:1090:40^@^@^@^@^@^@^@^@^@^@^@^@^ @^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^   @^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@ PPM 40 =====       $

I also found this for the 978 device:
> 2016/05/13 19:09:14 ===== UAT Device Serial: stx:978:34^@^@^@^@^@^@^@^@^@^@^@^@^       @^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@=====$

Axtel4, 2016-05-14:
@cyoung 

Update: I cleared my old log and created a new one so you could see the whole log with the interesting character.  What I was aiming for was to see the 1090 SDR gain setting like used to be in the stratux.log, but I think it may have been removed.

[stratux_char.log.txt](https://github.com/cyoung/stratux/files/264255/stratux_char.log.txt)

commit 2231ee49 (2016-05-14): Remove null characters from serial string.

Fixes #413.
closed 2016-05-14 by cyoung