cyoung/stratux #415: SQL Log Questions

Axtel4, 2016-05-14:
1. Stratux version:  v0.8r2 (fba162a4e9)
2. Stratux config:

    SDR
      [ ] single
      [ X ] dual

    GPS
      [ X ] yes
      [ ] no
      type:

    AHRS
      [ ] yes
      [ X ] no

    power source:  Ship

    usb cable: Heavy Duty

3. EFB app and version: (e.g., WingX Pro7 8.6.2) NA

    EFB platform: (e.g., iOS 9.2) NA

    EFB hardware: (e.g., iPad Mini 2) NA

4. Description of your issue:
I am trying to get a feel for the SQL logs and I have a few questions.

UAT Traffic Signal Strength:
Where can one find the signal strength for UAT traffic?  The 1090ES traffic signal strength is embedded in the Data Field of the es_messages table.  I thought I might find it in the “traffic” table, but signal strength isn’t one of the fields.

“Traffic” table:
In the “traffic” table how is the 1090ES and 978UAT direct received traffic identified?  

What unit of measure is the “distance” field?  I assume it is meters, but I am not sure.

“Messages” table:
What is the source of the “Signal_strength” field?  If the “ADSBTowerid” field is empty, is the “Signal_strength” direct UAT target data?  If so, how can this be correlated with a specific target?  ADS-B address?

I may have more questions as I delve into this further.

Thanks.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

AvSquirrel, 2016-05-15:
> Where can one find the signal strength for UAT traffic?
> What is the source of the “Signal_strength” field?

The `SignalLevel` field  in the `traffic` table reports signal strength for 1090ES and UAT traffic in dB. If you're missing it, disable "Record Replay Logs, delete the `stratux.sqlite` file, and re-enable logs.

[image]

[image]

The raw (linear) signal level, normalized to 0 = no signal and 1 = max signal (I or Q axis saturated), is reported as part of the decoded dump1090 Stratux message, and logged to `es_messages` as you've noted.

The `messages` table contains the raw 978 UAT messages, along with message metadata. `Signal_Amplitude` is raw amplitude from dump978, scaled such that 0 = no signal and 1000 = max signal. `Signal_Strength` is the dB value, normalized to 0 dB = 1000 amplitude.

> If the “ADSBTowerid” field is empty, is the “Signal_strength” direct UAT target data?

Yes.

> If so, how can this be correlated with a specific target?

See the notes above about the `SignalLevel` field in `traffic`.

> In the “traffic” table how is the 1090ES and 978UAT direct received traffic identified?

1090 and 978 traffic can be differentiated by the `Last_Source` field.
```
	TRAFFIC_SOURCE_1090ES = 1
	TRAFFIC_SOURCE_UAT    = 2
```
The type of target (air-to-air or ground-to-air) can be determined by the `TargetType` field, using the following type definitions from traffic.go:
```
	TARGET_TYPE_MODE_S    = 0
	TARGET_TYPE_ADSB      = 1
	TARGET_TYPE_ADSR      = 2
	// Assign next type to UAT messages with address qualifier == 2
	// (code corresponds to any UAT GBT targets with Mode S addresses.
	// These will be displayed with the airplane icon on the traffic UI page.
	// If we see a proper emitter category and NIC > 7, they'll be reassigned to TYPE_ADSR.
	TARGET_TYPE_TISB_S = 3
	TARGET_TYPE_TISB   = 4
```
Direct transmissions from aircraft will be either TargetType == 0 (Mode S messages, but no ADS-B position received yet for that ICAO code) or TargetType == 1 (ADS-B position has been received). Ground-based transmissions will be assigned codes 2-4. TargetType == 2 are ground-to-air ADS-R rebroadcasts of positions received on the other frequency. TargetType == 4 are TIS-B ground-air broadcasts of Mode C radar targets. TargetType == 3 are indeterminate -- they have Mode S addresses, and can either be TIS-B or ADS-R.

 > What unit of measure is the “distance” field? I assume it is meters, but I am not sure.

Correct. Meters, great circle distance, based on the last valid GPS position.

cyoung, 2017-04-19:
@Axtel4 - please reopen/add to this is you have further questions. Closing it for now.

closed 2017-04-19 by cyoung