cyoung/stratux #417: f514 reports as version 1b4c

Ergonomicmike, 2016-05-16:
Just .sh'd an in flux version of v0.8r2 with the f514.sh. (Twice.) It reports still as the previous, 1b4c. (Cleared cache in browser. NC.) Noticed that the .ah file was only 5 MB, in contrast to the usual 8.3 MB.

Will wait to see if a new .sh pops up after the 186th commit just made.

cyoung, 2016-05-16:
What about the latest?

Looks like you caught it while it was being uploaded from the build machine, maybe

Ergonomicmike, 2016-05-16:
@cyoung Yeah, that fixed it. 

So now that brings up a side issue: kinda scary that the Stratux will accept a partial .sh update and act like everything is okay. Should there be a rudimentary End Of File or some kind of simple sanity check that the downloaded .sh is more less intact? Not talking about a hash or CRC. Just something simple so that a partial file won't be accepted as a valid update.

Please close this issue after you've read it.

cyoung, 2016-05-16:
The way it's built sort of works that way - if the file is truncated then it "mostly" will fail silently and actually do nothing (unless the cut happens in the last ~1K of the update file). We'll have to build some of the front-end features to throw the error, after that an MD5 check vs end-of-file are about equal work. Thanks for the suggestion.

closed 2016-05-16 by cyoung

Ergonomicmike, 2016-05-17:
I don't know anything about EOF markers. I asked my airplane partner who designed FMC's for Hwell. She thought it was simply a special character. I was thinking something like the file has to end with ffffff or something like that for Stratux to know it got the whole file. Not perfect but would catch most truncations. (I'm surprised the build process uploads a partial file while in process.)

skypuppy, 2016-05-17:
One can add an md5 checksum to files and them compare them after downloading to the pi.  Advantage: surety of file loads.  Disadvantages: 1. longer effective download times, 2. PIA to maintain.

How are the files transferred?  FTP?  Sftp?