cyoung/stratux #421: Suppress GDL90 broadcast of ownship ADS-B messages to prevent ghosting

AvSquirrel, 2016-05-20:
1. Stratux version: v0.8r2
2. Stratux config:

    SDR
      [ ] single
      [X] dual

    GPS
      [ ] yes
      [x] no
      type:

3. EFB app and version: ForeFlight 7.6.1

    EFB platform: iOS 9.3.1

    EFB hardware: iPad Air

4. Description of your issue:

A Reddit user [reported ghosting of actual ownship ADS-B broadcasts](https://redd.it/4jwu16) when using dual SDRs and no GPS. Without a GPS position, Stratux does not broadcast a GDL90 ownship message. Without an ownship message, clients (ForeFlight) are unaware of the ownship ICAO code, and therefore do not suppress the target.

Transmitting unknown / invalid position as NACp = 0, Lat = 0, Lng = 0 in accordance with the GDL90 protocol is unreliable due to certain clients ignoring the NACp flag and treating 0° , 0° as a valid position.

Instead, a prototype for ownship GDL90 traffic suppression (401c372ef) based on the most recent commits has been pushed to https://github.com/AvSquirrel/stratux/tree/ownship-testing and a .sh update with these changes is attached to this message. Note that this build is two months of development commits ahead of the v0.8r2 release

[update-stratux-v0.8r2-401c372ef4.sh.zip](https://github.com/cyoung/stratux/files/273861/update-stratux-v0.8r2-401c372ef4.sh.zip)

Axtel4, 2016-05-20:
Also usefull when sending Mode S Datalink DF-17 broudcasts for the older generation Mode S equipped aircraft?

cyoung, 2016-05-31:
@AvSquirrel - I'm going to start tagging "GDL90" on the "policy decisions" we make that deviate from GDL90 spec or where GDL90 doesn't provide guidance. This is probably the best way to handle ownship with no GPS IMO, but keeping a label on these will help in the future when compliance questions come up.

commit 94d2467e (2016-05-31): Merge pull request #435 from AvSquirrel/ownship-testing

Intial prototype for ownship traffic suppression. Fixes #421.
closed 2016-05-31 by cyoung