cyoung/stratux #422: Timeouts on AHRS?

skypuppy, 2016-05-20:
1. Stratux version:
2. Stratux config:

    SDR
      [ ] single
      [x ] dual

    GPS
      [x ] yes
      [ ] no
      type:

    AHRS
      [x ] yes
      [ ] no

    power source:  Anker battery

    usb cable: Amazon Basics

3. EFB app and version: (e.g., WingX Pro7 8.6.2)  iFlyGPS

    EFB platform: (e.g., iOS 9.2)  Android

    EFB hardware: (e.g., iPad Mini 2)  Samsung Galaxy Tab S, 10.5

4. Description of your issue:
Strangely, the invensense IMU, mounted vertically in my test platform, is defaulting to flat, level, and north pointing within a short time after stratux starts.  Then, it passes that reading on to iFly via wifi as well as setting those readings for the webui.  This is entirely wrong and totally inappropriate.  Further, once that happens, no motions at all are transferred to the external app or webui.
Very bad bug.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Nokomis449, 2016-05-21:
I had a similar issue a few months ago with my RY835AI.  Didn't matter the orientation on bootup, it just didn't work (except for the GPS).  Wiggled wires and connections, no change.  Rebooted, no change.  Rebooted and rebooted and rebooted via web interface with Stratux/RY on the table untouched, and once every occasional glorious reboot it would work correctly.  At that point I could pick it up and shake the dog out of it, and like a Timex it just kept on ticking and indicating correctly so I knew it wasn't a loose connection or bad solder joint.  Same behavior in the car - Most every boot had no AHRS (GPS would work fine; all was connected via pins) but occasionally the glorious boot would appear and it would work until shutdown.  This was several versions ago, and I've since shelved the RY in favor of the VK until such time as someone claims the $4,000.00.

Ergonomicmike, 2016-05-24:
https://github.com/cyoung/stratux/issues/131

cyoung, 2016-06-13:
Closing for deferral as I'm certain that the AHRS implementation in the future won't be the same as this one.

closed 2016-06-13 by cyoung