cyoung/stratux #424: After update DEV build may 16 ... GPS Disconnected

mmaitre74, 2016-05-23:
1. Stratux version:
2. Stratux config:

    SDR
      [ ] single
      [X ] dual

    GPS
      [X ] yes
      [ ] no
      type:

    AHRS
      [X ] yes
      [ ] no

    power source: 5V adaptor

    usb cable: GPS micro-USB connection

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue: THe Gps is alway disconnected .  Version build b9951241a5 and up to aae8e925ef.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2016-05-23:
Which GPS unit?

mmaitre74, 2016-05-23:
Sorry : RY835AI

 

Which GPS unit?

cyoung2, 2016-05-24:
Got it. We'll have an updated version later tonight with a fix.

AvSquirrel, 2016-05-24:
#418 ?

closed 2016-05-24 by cyoung

cyoung, 2016-05-24:
@AvSquirrel 's PR integrated which fixes it here: http://updates.stratux.me/builds/update-stratux-v0.8r2-47271df764.sh

@mmaitre74 - give it a try and let us know if it fixes it, reopen if any problem.

mmaitre74, 2016-05-24:
Working now. Issue Fix.

commit 47271df7 (2016-05-24): Merge pull request #418 from AvSquirrel/udev-updates

Udev updates. Fixes #424.