cyoung/stratux #425: Upgraded to RaspPI3 gps shows disconnected, AHRS works

dcecchino, 2016-05-26:
Stratux version:Stratux v0.8r2
Stratux config:

SDR
[ ] single
[ X] dual

GPS
[ X] yes
[ ] no
type: RY835AI

AHRS
[X ] yes
[ ] no

power source: Anker PowerCore External Battery 20100mAh

usb cable: supplied with Anker

EFB app and version: (e.g., WingX Pro7 8.6.2) WingX Pro 7 (8.6.3.1)

EFB platform: (e.g., iOS 9.2)

EFB hardware: (e.g., iPad Mini 2) IPAD PRO 9.7

Description of your issue: Hello, I have upgraded to PI 3 and GPS shows not detected. I'll send the logs later today. I did a clean install as well and still the same result, GPS does not acquire a signal when turned on in the settings. 

Thanks for your support and development!
If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

AvSquirrel, 2016-05-26:
Known issue (https://github.com/cyoung/stratux/issues/393). Solved with #394.

Fix is included on manual updates since April 20, and involves adding the following line to /boot/config.txt:

`dtoverlay=pi3-miniuart-bt`

Reboot and the serial connection over GPIO to /dev/ttyAMA0 will be usable.

closed 2016-05-26 by AvSquirrel

dcecchino, 2016-05-26:
Awesome, thanks for the fast response AvSquirrel!