cyoung/stratux #426: Remove bloat from interface pages

cyoung, 2016-05-27:
1. Stratux version: 47271df7647b4efa49d86409d98897400559b018
2. Stratux config:

    SDR
      [x] single
      [ ] dual

    GPS
      [ ] yes
      [x] no
      type:

    AHRS
      [ ] yes
      [x] no

    power source: wall

    usb cable: a good one

3. EFB app and version: N/A

    EFB platform: N/A

    EFB hardware: N/A

4. Description of your issue:

a. I have GPS disabled but it's showing GPS-enabled options on the traffic page.

[image]

b. What is the clock difference used for now? We've abstracted away from RTC dependence.

[image]

Ergonomicmike, 2016-05-27:
On the clock difference: I was kind of shocked to see my clock dif be some huge negative number the other day when I started the Stratux. Turns out that it had been more than a few days since I had powered up the BU-353 and it took a minute to get GPS time. So the clock dif could be considered a Diagnostic value if something goes wrong with the GPS.

As for abstracting away from Pi time, if one is not using GPS on the Stratux, then one is stuck on Pi time. Does that matter for anything else? It makes the time stamp in the logs goofy. Any implications for traffic or FIS-B? Should there be a way for those users to capture their device time and paste it to the Pi?

AMNegron, 2016-05-30:
Do UAT messages send a time stamp? If not, should there be a way to manually update the clock? And as Ergonomicmike asked, is it required?

AvSquirrel, 2016-06-08:
@cyoung - 

For a while, some power users ran into problems with certificate issues if they tried to pull and build on a Pi with incorrect clock settings. The move to .sh updates mitigates that. Developers will just have to know to use a GPS, set the clock manually, or enable ntp.

With all of the user-facing functions using the monotonic ticker, about the only reason I can think to keep user-facing time around is log correlation, as @Ergonomicmike notes. I wouldn't call it a strong reason to keep it.

@AMNegron -- traffic messages are flagged with the timeslice within the second they are sent, but are not timestamped. Some of the FIS-B (weather) messages have timestamps / datestamps for message applicability (e.g. NOTAMs), but not the actual message send time.

closed 2016-12-01 by cyoung