cyoung/stratux #429: SQL Logs

Axtel4, 2016-05-28:
1. Stratux version: v0.9b1 (89107818b8)

2. Stratux config:

    SDR
      [ ] single
      [ X ] dual

    GPS
      [ X ] yes
      [ ] no
      type: VK-172

    AHRS
      [ ] yes
      [ X ] no

    power source: NA

    usb cable: NA

3. EFB app and version: (e.g., WingX Pro7 8.6.2) NA

    EFB platform: (e.g., iOS 9.2) NA

    EFB hardware: (e.g., iPad Mini 2) NA 

4. Description of your issue:
Is there any discussion of having the SQL logs downloadable through the Web UI like the older text log files (XXXX-YYYY.log) were?  Current the only things listed on the SDR, AHRS and GPS log page are LOGINDEX and wifi_watch.log.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2016-05-31:
They can be downloaded all at once since it's all added to a single file now, http://192.168.10.1/logs/stratux.sqlite. Does that answer the question?

Axtel4, 2016-06-01:
Thanks. I was able to download the file.  However, I had to SSH in to figure out which file to download based the date stamp.

closed 2016-06-13 by cyoung