cyoung/stratux #431: Text overwriting on WebUI traffic page

JohnOCFII, 2016-05-30:
1. Stratux version: v0.9b1 (89107818b8)
2. Stratux config:

    SDR
      [ ] single
      [X] dual

    GPS
      [X] yes
      [ ] no
      type: RY835AI

    AHRS
      [X] yes
      [ ] no

    power source: panel

    usb cable: PortaPow

3. EFB app and version: (e.g., WingX Pro7 8.6.2) N/A

    EFB platform: (e.g., iOS 9.2) iOS 9.3.2.

    EFB hardware: (e.g., iPad Mini 2) iPad Air 2

4. Description of your issue:

Using the standard mobile Safari browser, looking at the Traffic page, the Power (dB) and Age fields overwrite each other, and also extend past the graphic "box" area.  Limiting the display to fewer characters is probably all that is required to resolve.
[image]

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

AvSquirrel, 2016-05-30:
This looks like a client-side cache problem. Your screenshot shows the page structure and headers from v0.8r2. Can you hit refresh a couple times to see if it clears up?

Improvements to the UI, including reflow improvements on devices of different sizes, were incorporated at #398. Among other things, rounding of power and age were move from traffic.js to Angular expressions in traffic.html. If the .js file updated and the .html didn't, it would explain what you're seeing.

AvSquirrel, 2016-05-30:
It should look like this:

[image]

JohnOCFII, 2016-05-31:
You are right.  It is a client-side problem.  I'll close this out.  (Separately, I use the option to create a Home Screen short-cut, and even after I clear cache, and delete and re-make the shortcut, it still pulls in the old format.  Going to 192.168.10.1 from the browser directly pulls in the new format.)

closed 2016-05-31 by JohnOCFII

JohnOCFII, 2016-05-31:
@AvSquirrel  It looks like this is a pretty common problem with web sites saved to the home screen.  There is no user provision to clear cache.  Either the developer needs to make a change, such as updating the version of the manifest, or the user needs to connect their iOS device to the computer and run some debugging tools from the desktop.

https://coderwall.com/p/knft5a/3-ios-homescreen-web-app-tips