cyoung/stratux #433: Should Stratux buffer FIS-B anytime it sees FIS-B?

Ergonomicmike, 2016-05-31:
1. Stratux version: 0.v9b1

4. Description of your issue:

A request for Enhancement after v1.0 comes out.

If technologically possible, I would like Stratux to buffer FIS-B anytime it's getting FIS-B, regardless of whether a client has connected to it or not. Here's the scenario:

You're getting ready for a flight into thunderstormy weather. ADS-B coverage is getting better, such that you can often receive a tower on the ground.

You power up your Stratux before preflight.  I suppose you should power up your EFB too. But you don't just yet. (Or perhaps you're using your tablet on the Net for one last look of weather from there.)

When you turn your EFB on, you would like it to populate with the latest NEXRAD immediately if it has the data stored, regardless of whether you had established a connection with the Stratux prior to this time. Otherwise, you might have to wait 15 minutes in flight for the next cycle of NEXRAD to come in live.

skypuppy, 2016-05-31:
iFly gathers weather if it has a web connection to the Internet.  i don't know if other EFB's do this as well.  
It would be great if stratux can gather weather from an ADS-B tower OR from the Internet as well.  The airports I've seen lately (except one) all have open wifi just for situations like this.  Call  it an enhancement, an add-on, anything you like.  I would love it if the stratux wifi could reach out and get weather info from places besides towers, too.  I don't even know if the message formats are the same between ADS-B streams and Internet sources.  I also don't know if all EFB's use the same web sources for their weather info.  Yes, that almost makes the stratux a "proxy" vpn but the tools are there.
Might as well ask for it all: iFly updates sectionals, etc. via the web as well, so let's allow the stratux to feed that raw info to the EFB as well when asked.  
I have no clue what the programming overhead would be for this but for the web guys, I bet it isn't much.  Just to enable a virtual pass-through when asked to do so.  It would save some fumble-fingered pilots some gnashing of teeth.  :)

cyoung, 2016-05-31:
Good idea, and this is the same as #358.

closed 2016-05-31 by cyoung

Ergonomicmike, 2016-05-31:
Oops, sorry. I can't keep up. (Especially embarrassing since I commented on #358.