cyoung/stratux #434: FIS-B buffering stops when all Sleepflags = true ?

Ergonomicmike, 2016-05-31:
1. Stratux version: v0.9b1
2. Stratux config:

    SDR
      [ ] single
      [X ] dual

    GPS
      [ X] yes
      [ ] no

3. EFB app and version: iFly v9.7.19b

    EFB platform: Android KK

    EFB hardware: Asus ZenPad S

4. Description of your issue:

I might have tripped across a bug, of sorts, today with my testing at KPAN, which now has an ADS-B tower in range on the ground. If found that when there are no clients connected to the Stratux, it seems to stop buffering FIS-B. 

For example, I initially had two tablets connected to the Stratux, one on /getClients and the other on iFly. With that set up,  the sleep flag and FIS-B buffer worked as expected. Whenever I shut down iFly or turned off the WiFi on that tablet, the sleep flag wen True and the Message Length buffer started growing from zero. When I brought iFly back on line, the buffer dropped to zero. (The sleep flag for the /getClients tablet toggles quickly between false and true, mostly staying false.)

Then I went for a long walk around the airport.

I left the Stratux running, gathering data from the nearby ADS-B tower. The /getClients tablet put itself to sleep. I shutdown iFly and that tablet went to sleep. Presumably the WiFi connections on both tablets went cold sometime after that.

When I got back from my walk and woke up the /getClients tablet, I expected that the FIS-B buffer for the iFly client would be full (dithering around 24950, like mysterious client 3, did earlier in my testing at KPAN). But to my surprise, the buffer length for the iFly tablet was only up to 5000. (This was after two hours. The client 3 buffer had filled up in a half hour. So surely the iFly buffer should have been full by this time.)

Message Length dropped to zero as soon as I brought iFly back on line. I zoomed out to see if I had a NEXRAD image. No joy.  I had to wait for NEXRAD to come in on a cycle.

So it seems that the Stratux stopped buffering when it had nothing hooked up to it. That is, when all the sleep flags went solidly True. Is this the desired behavior?

cyoung, 2016-05-31:
Is the tower (34.43111,-111.50389)? This is the closest I see from KPAN, maybe there's a brand new one? If so it's not on my map.

iFly sleep behavior is a little strange. Are you certain that you closed the app? You said that before leaving the tablet unattended that you "shut down" iFly, this means pressing the Overview Button (bottom right) and swiping iFly off the screen? The Home Button (bottom middle) only puts it into the background.

Ergonomicmike, 2016-05-31:
Yes, those are the coordinates for the tower. It's on a mountain top and can be received on the ground at KPAN. (About -11 dB.)

I closed iFly from within iFly. (IFly Menu button > Exit application > Shtdown application.)

Ergonomicmike, 2016-06-08:
Flew for about 5 hours yesterday. First leg with with iFly v9.7.23b on two tablets and Stratux v0.9b1 feeding them both. No problems at all.

Then, I updated the Stratux to 18b8.sh. After I did that, I noticed that the AIRMET/SIGMET was grayed out in iFly on my first tablet. But I had a bunch of AIRMETS/SIGMETS on my second tablet. (The weather was fine but a local ATIS mentioned Hazardous weather and to listen to HIWAS for more info. I was surprised, since I wasn't showing any AIRMETS/SIGMETS.)

I tried a bunch of different things (reboot Stratux), but in the end, it was when I turned off the WiFi on my second tablet that AIRMETS/SIGMETS suddenly came alive on my first tablet.

Ergonomicmike, 2016-07-05:
I don't know if this is related. But flew to Leadville, CO yesterday. Coming home and watching NEXRAD via FIS-B,  two tablets running the same version  of iFly, and getting their same data from the same Stratux showed two different time stamps for the NEXRAD composite. (I thought I had taken screen shots. If I did, I can't find them.)

JohnOCFII, 2016-07-05:
> I don't know if this is related. But flew to Leadville, CO yesterday. Coming home and watching NEXRAD via FIS-B, two tablets running the same version of iFly, and getting their same data from the [...]

Just FYI - I've seen this same behavior a few times over the last couple of years with the Appareo Stratus and ForeFlight.  I don't think that behavior is specific to Stratux.

Ergonomicmike, 2016-07-06:
Tnx. I can't imagine why the same data would give different time stamps. I assume that all the NEXRAD data comes in at the same time. So that would leave buffering to the individual tablets as the difference? (Although they were about 4 minutes apart.) But if you've seen this before, then I guess that's the way it is. I'll try to do some more testing when our monsoon kicks into gear.

JohnOCFII, 2016-07-06:
> Tnx. I can't imagine why the same data would give different time stamps. I assume that all the NEXRAD data comes in at the same time. So that would leave buffering to the individual tablets as the [...]

My understanding is that the NEXRAD images themselves take a few minutes to assemble before they are pushed to the providers (such as the ADS-B system, or XM weather), then they pushes take a few minutes to assemble.  Depending on the particular tower you are at, you may receive things that are a few minutes separated from what was pushed to another tower.  It's all a bit fuzzy to me, but that is what I remember from what I've been told.

Ergonomicmike, 2016-07-06:
Yeah, I understand that part. What I don't understand is why two tablets hooked up to the same Stratux display two different things. It's like hooking up two monitors to the same computer and getting different images. (You know, strangely, I'll see slightly different GS's and GPS altitudes on the two tablets. Am back to thinking it's an OSI layer thing in the Stratux and the two tablets are not identical clients to the Stratux server.)

skypuppy, 2016-07-06:
It's also quite possible that the two EFB's use different algorithms.

bglaplante, 2016-07-23:
I observed a similar problem I think, and maybe know why it happens.  Setup is Stratux 0.8r2, dual radio, gps.  I have my tablet connected (Android Avaire), I also have my laptop connected with my own software which is reading the traffic data via the webservice.  Tablet is Ubuntu 14.04.4.  All works fine.  When I go away, the laptop locks, and the tablet loses GPS (set to use Stratux GPS).  https://ubuntuforums.org/showthread.php?t=2004690 describes a problem with wifi on Ubuntu.  Like the link, I have to disable/enable wifi on Ubuntu to get it connected to Stratux again.  I understand the Ubuntu issue is my problem.  But, note that the Ubuntu networking stopping apparently also jammed the Stratux webservice and caused it to stop responding to Avaire as well.  Seems like there is a missing check for a jammed webservice client.

Ergonomicmike, 2016-07-23:
I'm going to close this issue for now, based on my latest findings. (Perhaps bglaplante should open a new issue focusing on the jammed webservice issue?) 

I just got back from KPAN, where there was NEXRAD weather around. This time, sleep buffering appears to work fine. (Although I was getting banding in the NEXRAD picture on the ground. I wonder if the buffer needs to be bigger for NEXRAD data?) This was version v1.0r1.

One differences between this test and my last (failed) test was Stratux addressed the issue where AIRMETS were causing the Stratux to go offline after a few minutes. And iFly is up to v9.7.23b. (There seems to be a small bug in iFly, where the moniker for NEXRAD doesn't drop the ADS-B tag when the unit isn't getting new NEXRAD data. I'll report that to iFly.)

I recorded some video of my tablet coming back on-line, but I don't think there's much point in posting a video showing something working.

closed 2016-07-23 by Ergonomicmike

cyoung, 2016-07-23:
@bglaplante - please upgrade to v1.0r1. @Ergonomicmike - thanks for the update.

Ergonomicmike, 2016-08-26:
I was able to do more testing on the ground at KPAN recently with weather in the area. IFly EFB and ADS-B reception on the ground.

I put the tablet to sleep and left the Stratux running. Periodically I woke the tablet up. At first, iFly would show the last NEXRAD image with a time delay (typically 30 minutes to 60). And then, within about 2 minutes, the NEXRAD image would update to current.

The 2 minutes to update appears consistent. If that is the Stratux sending buffered data, that seems slow. (iFly updates NEXRAD images within seconds on my home Wi-Fi.) OTOH, it doesn't seem like I could be catching FIS-B data updating every 2 minutes. Or could it? (There's virtually no traffic at 2 am near KPAN - so very little traffic data being snet by the tower.)

Any high level test that I can do to see if I'm getting buffered data or direct updates from the tower?